ros2/rclpy #1047: Can not setup qos override option with QoSPolicyKind.DEADLINE

maghauke, 2022-11-28:
## Bug report

#### Expected behavior

I try to set up a qos override option for my subscriber using:
```
qos_override_opts = QosOverride.QoSOverridingOptions(
            policy_kinds=(
                Qos.QoSPolicyKind.HISTORY,
                Qos.QoSPolicyKind.DEPTH,
                Qos.QoSPolicyKind.RELIABILITY,
                Qos.QoSPolicyKind.DEADLINE,
                )
        )
```

#### Actual behavior

When launching the relevant nodes subscriber I get the following error message:

```
File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos_overriding_options.py", line 165, in _get_qos_policy_parameter
value = value.nanoseconds()
TypeError: 'int' object is not callable

```

To me it seems like: [this](https://github.com/ros2/rclpy/blob/c7feb4b70c33aaf22e8ad467605c0a58451299c9/rclpy/rclpy/qos_overriding_options.py#L154)
does not return a Duration object but only an int...

fujitatomoya, 2022-12-02:
@maghauke thanks for posting issue. this can be fixed with https://github.com/ros2/rclpy/pull/1050.

fujitatomoya, 2022-12-22:
https://github.com/ros2/rclpy/pull/1050 has been completed including backport of humble, closing this issue.

closed 2022-12-22 by fujitatomoya