ros2/rclpy #1056: rclcpp and rclpy `spin*()` functions seem to have inconsisntent behvaior for timeout=0

EricCousineau-TRI, 2022-12-10:
## Feature request

#### Feature description

Comparing based on `spin_some`:

`rclcpp`, 
https://github.com/ros2/rclcpp/blob/338eed0c060ab58608fc6e44c6936bde2c4e061e/rclcpp/include/rclcpp/executor.hpp#L308-L309
```
   * \param[in] max_duration The maximum amount of time to spend executing work, must be >= 0.
   *   `0` is potentially block forever until no more work is available.
```
`rclpy`:
https://github.com/ros2/rclpy/blob/fdb04566e20e9ccbaceeeffc51916b65b1f86a19/rclpy/rclpy/executors.py#L310-L311

I'm not sure if I understand why they're different. Ideally, they shouldn't be?
I was unable to figure out the analogous functionality in `rcl` C itself.

#### Implementation considerations

See above.

\cc @sloretz re: Anzu PR 9961

sloretz, 2022-12-13:
It looks like this issue is comparing `rclcpp`'s `spin_all` to `rclpy`'s `spin_once`.

I don't see documentation in [`rclcpp` for `spin_once`](https://github.com/ros2/rclcpp/blob/432bf21f0261bab209b37ccfe6da550f02751a22/rclcpp/include/rclcpp/executor.hpp#L317-L320), but I'd assume `spin_node_once` is similar just restricted to callbacks from one node. It looks like it has the same behavior as `rcply` [where it blocks forever when negative, and doesn't block at all when 0](https://github.com/ros2/rclcpp/blob/432bf21f0261bab209b37ccfe6da550f02751a22/rclcpp/include/rclcpp/executor.hpp#L239-L256).

EricCousineau-TRI, 2023-03-22:
Gotcha, thanks! In that case, I will recater this issue to say rclpy should also offer the same methods, e.g. `spin_al`, and that documentation in both should make it clear the behaviors are similar.