ros2/rclpy #1064: Problems with importing pybind11 when ROS2 is built with Clang 12

sebastian-freitag-pylot, 2023-01-04:
## Bug report
Ubuntu 22.04
Installation from source
humble(6b6617af3ebf8caf03b969c71074dcbc6a725ee7)
Default DDS
rclpy

#### Steps to reproduce issue
* Follow the documentation for building from source but compile with Clang 12 (llvmorg-12.0.1).
* Use pytest.

#### Expected behavior
Everything works.

#### Actual behavior
pytest reports `ImportError` with the following error message. Repeated re-building of workspace etc. does NOT solve the problem.

```E   ImportError: /opt/ros/humble/rclpy/local/lib/python3.10/dist-packages/rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so: undefined symbol: __atomic_is_lock_free```
```E   The C extension '/opt/ros/humble/rclpy/local/lib/python3.10/dist-packages/rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://docs.ros.org/en/{distro}/Guides/Installation-Troubleshooting.html#import-failing-even-with-library-present-on-the-system' for possible solutions```

#### Additional information
When checking for `__atomic_is_lock_free` on the entire system (with `nm` etc.), it is not defined anywhere. There is https://github.com/ros2/rcutils/blob/humble/include/rcutils/stdatomic_helper/gcc/stdatomic.h but it is not used apparently. Also I am not sure if it even works, since `__has_feature(c_atomic)` does not work when the file is compiled with a c++ compiler (it should be `__has_feature(cxx_atomic)`). But this is not about rcutils, it is about rclpy. The symbol `__atomic_is_lock_free` is provided by a compiler builtin, but someone needs to tell clang to use it, maybe.

mjcarroll, 2023-01-04:
I believe that clang explicitly requires the `-latomic` flag.  I ran into this recently on a personal project, and there are other reports of this: https://github.com/nodejs/node/issues/2823

sebastian-freitag-pylot, 2023-01-04:
I created [a PR, ](https://github.com/ros2/rclpy/pull/1065). This fixes it for me.

sebastian-freitag-pylot, 2023-01-04:
Yeah my mistake

> this should be open until #1065 is merged?

Yes, my mistake I guess. Sorry.

commit 9584697c (2023-01-27): Explicitly link atomic when building with Clang (#1065)

This fixes https://github.com/ros2/rclpy/issues/1064.
When building with Clang, tell cmake to link atomic to the _rclpy_pybind11 target.

Closes: #1064
closed 2023-01-27 by mjcarroll