ros2/rclpy #1069: TF listener works only when running from `ros2 run`, not using launch file

songanz, 2023-01-14:
## Bug report

**Required Info:**

- Operating System:
  - ROS1 melodic and Gazebo running in docker with Ubuntu 18.04
  - ROS2 foxy running in docker with Ubuntu 20.04
- Installation type:
  - Build from source
- Version or commit hash:
  - foxy
- DDS implementation:
  - NAN
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

1. Start any gazebo simulator you have which contains at least two frames in a ROS1 docker
  I am particularly using this environment from CMU: [cmu_env](https://github.com/HongbiaoZ/autonomous_exploration_development_environment) with 
  `roslaunch vehicle_simulator system_garage.launch`
2. Start the ros1_bridge in a ROS2 docker
```bash
export ROS_MASTER_URI=http://localhost:11311
ros2 run ros1_bridge dynamic_bridge --bridge-all-topics --ros-args -p use_sim_time:=true
```
3. Set up tf_listener in ROS2 using rclpy
Create a tf_listener package following: https://docs.ros.org/en/foxy/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Listener-Py.html
Specifically my python file:
```python
import rclpy
from rclpy.node import Node

from tf2_ros import TransformException
from tf2_ros.buffer import Buffer
from tf2_ros.transform_listener import TransformListener

class FrameListener(Node):

    def __init__(self):
        super().__init__('tf_frame_listener_py')

        # Declare and acquire `target_frame` parameter
        self.target_frame = self.declare_parameter(
          'target_frame', 'camera').get_parameter_value().string_value

        self.tf_buffer = Buffer()
        self.tf_listener = TransformListener(self.tf_buffer, self)

        # Call on_timer function every second
        self.timer = self.create_timer(1.0, self.on_timer)

    def on_timer(self):
        # Store frame names in variables that will be used to
        # compute transformations
        from_frame_rel = self.target_frame
        to_frame_rel = 'map'

        try:
            t = self.tf_buffer.lookup_transform(
                to_frame_rel,
                from_frame_rel,
                rclpy.time.Time())
            self.get_logger().info("TF success!")
            self.get_logger().info(f'self.get_clock().now(): {self.get_clock().now()}')
            self.get_logger().info(f'rclpy.time.Time(): {rclpy.time.Time()}')
        except TransformException as ex:
            self.get_logger().info(
                f'Could not transform {to_frame_rel} to {from_frame_rel}: {ex}')
            self.get_logger().info(f'self.get_clock().now(): {self.get_clock().now()}')
            self.get_logger().info(f'rclpy.time.Time(): {rclpy.time.Time()}')
            return

def main():
    rclpy.init()
    node = FrameListener()
    try:
        rclpy.spin(node)
    except KeyboardInterrupt:
        pass

    rclpy.shutdown()
```
5. Setup launch file
```python
import os

from launch import LaunchDescription
from launch_ros.actions import Node
from launch.substitutions import LaunchConfiguration
from ament_index_python.packages import get_package_share_directory

def generate_launch_description():
    package_name='tf_py'
    tf_listener_node = Node(
        package= package_name,
        namespace= 'tf_py_node',
        executable= 'tf_listener_py',
        name= package_name,
        output='screen',
        emulate_tty=False
    )
    return LaunchDescription([tf_listener_node])
```
6. Run the TF listener using both `ros2 run tf_py tf_listener_py` and `ros2 launch tf_py tf_listener.launch.py `

#### Expected behavior
The listener can listen to the TF correctly.
#### Actual behavior
[image]

#### Additional information
I was not sure whether this is the rclpy lib's problem. So I create a C++ package too using rclcpp. And as you can see from the figure above, the listener written using the rclcpp lib can listen to the TF correctly. 
For your reference, the C++ listener's code is here:
```cpp
#include <chrono>
#include <functional>
#include <memory>
#include <string>

#include "geometry_msgs/msg/transform_stamped.hpp"
#include "rclcpp/rclcpp.hpp"
#include "tf2/exceptions.h"
#include "tf2_ros/transform_listener.h"
#include "tf2_ros/buffer.h"

using namespace std::chrono_literals;

class FrameListener : public rclcpp::Node
{
public:
  FrameListener()
  : Node("tf_frame_listener_cpp")
  {
    // Declare and acquire `target_frame` parameter
    target_frame_ = this->declare_parameter<std::string>("target_frame", "map");

    tf_buffer_ =
      std::make_unique<tf2_ros::Buffer>(this->get_clock());
    tf_listener_ =
      std::make_shared<tf2_ros::TransformListener>(*tf_buffer_);

    // Call on_timer function every second
    timer_ = this->create_wall_timer(
      1s, std::bind(&FrameListener::on_timer, this));
  }

private:
  void on_timer()
  {
    // Store frame names in variables that will be used to
    // compute transformations
    std::string fromFrameRel = target_frame_.c_str();
    std::string toFrameRel = "camera";

    geometry_msgs::msg::TransformStamped t;

    // Look up for the transformation between target_frame and turtle2 frames
    // and send velocity commands for turtle2 to reach target_frame
    try {
        t = tf_buffer_->lookupTransform(
                        toFrameRel, fromFrameRel,
                        tf2::TimePointZero);
        RCLCPP_INFO(this->get_logger(), "TF success!");

    } catch (const tf2::TransformException & ex) {
        RCLCPP_INFO(
        this->get_logger(), "Could not transform %s to %s: %s",
        toFrameRel.c_str(), fromFrameRel.c_str(), ex.what());
        return;
    }

  }

  // Boolean values to store the information
  rclcpp::TimerBase::SharedPtr timer_{nullptr};
  std::shared_ptr<tf2_ros::TransformListener> tf_listener_{nullptr};
  std::unique_ptr<tf2_ros::Buffer> tf_buffer_;
  std::string target_frame_;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<FrameListener>());
  rclcpp::shutdown();
  return 0;
}
```

fujitatomoya, 2023-01-17:
@songanz can you check topics are bridged as expected from ROS 1 to ROS 2 to issue `ros2 topic list`, `ros2 topic info -v` before start running or launching the ROS 2 node? I believe that is the 1st step we need to confirm.

songanz, 2023-01-18:
Let me include those info:
[image]
[image]

And I have also checked the frame using `ros2 run tf2_tools view_frames.py`
[image]

songanz, 2023-01-25:
Dear all, any thoughts about this issue?

fujitatomoya, 2023-01-25:
So after all it bridges the topic w/o any problems, but tf listener does not subscribe on that topic. btw, do we need `namespace= 'tf_py_node` in launch file?

songanz, 2023-01-26:
OMG, remove the namespace solved my problem! Thank you so much!

closed 2023-01-26 by songanz