ros2/rclpy #1080: 404 for API Docs Link on ROS Index

ottojo, 2023-02-07:
The [ros index page for rclpy](https://index.ros.org/p/rclpy/github-ros2-rclpy/#humble) contains a "API Docs" link `http://docs.ros.org/en/humble/p/rclpy`, but the docs are not found at that URL.

Using google, i could only find docs for foxy (https://docs.ros2.org/foxy/api/rclpy/index.html) and galactic (https://docs.ros2.org/galactic/api/rclpy/index.html)

The CI job for generating (humble) docs seems to run, but docs are nowhere to be found: https://build.ros2.org/view/Hdoc/job/Hdoc__rclpy__ubuntu_jammy_amd64/

clalancette, 2023-02-07:
You did a good job tracking down how all of that works.  In short, you are correct that we aren't currently generating anything for rclpy properly.

In order to get there, we need to land https://github.com/ros-infrastructure/rosdoc2/pull/28 (which still needs some work), as well as do some custom configuration in `rclpy` itself.  I'll leave this open as the issue to track the addition of that configuration.

Dario-Mantegazza, 2023-03-23:
Hi,
it is probably too early to ask since the pull request cited above was just merged yesterday, but, when can we expect that the doc (for humble at least) will be published?
Thanks

clalancette, 2023-03-23:
> it is probably too early to ask since the pull request cited above was just merged yesterday, but, when can we expect that the doc (for humble at least) will be published?

While we did get the PR landed, there are still additional follow-up fixes that are needed to rosdoc2.  And we did not get the custom configuration in rclpy landed.  And then we need to backport it to humble.  We are slowly getting there, but it is still going to be some time to get it all working end-to-end.

ottojo, 2023-05-30:
Since the iron release is now out and the api-docs are built and correctly linked in the index, i'm happy with that resolution so far! I won't close this yet since backporting to humble is mentioned above, but if that should be moved to another issue, feel free to close this :+1: