ros2/rclpy #1082: rclpy fails to build with error about Python3 version

ckurtz22, 2023-02-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 Jetson
- Installation type:
  - from source
- Version or commit hash:
  - 7ceb3791f42d4a98d5211988fdd4fcbec98e7de8
- DDS implementation:
  - default
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
Building ROS Humble from source as per the instructions here https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html

#### Expected behavior
Builds properly

#### Actual behavior
rclpy fails to build with a Python error

#### Additional information
colcon output:
```Starting >>> rclpy
--- stderr: rclpy                                   
CMake Error at /usr/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Development Development.Module
  Development.Embed) (found version "3.9.5")
Call Stack (most recent call first):
  /usr/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.25/Modules/FindPython/Support.cmake:3245 (find_package_handle_standard_args)
  /usr/share/cmake-3.25/Modules/FindPython3.cmake:490 (include)
  CMakeLists.txt:41 (find_package)

---
Failed   <<< rclpy [9.98s, exited with code 1]
```

python3 and python3-dev are installed through apt, python3 version is 3.8.10

fujitatomoya, 2023-02-11:
@ckurtz22 

can you try Ubuntu 22.04 instead? Ubuntu 20.04 is only for community support. 

- https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html#system-requirements
- https://www.ros.org/reps/rep-2000.html#support-tiers

ckurtz22, 2023-02-11:
@fujitatomoya unfortunately not. As mentioned in the first post, this is on an NVidia Jetson (AGX Xavier to be specific). This can only go up to Ubuntu 20.04. The only version of ROS 2 that is officially supported on this OS is Foxy, but I do not want to use this as it will be eol in 3 months.

fujitatomoya, 2023-02-11:
yeah i understand, but unfortunately that is only community support. i am not really sure how to fix this right away either.
we can keep this open to get feedback and information from community.

something else i would try is, how about using container like docker?

cottsay, 2023-02-23:
If present, `FindPython` will always prefer a newer version of the interpreter and libraries, even if your system is using a different version as the "default" one.

You can specify CMake variables to give CMake a hint as to which interpreter you'd like to use. Here's the official documentation on CMake's site: https://cmake.org/cmake/help/latest/module/FindPython.html#artifacts-specification

I set all four of these on my Fedora system, which would otherwise use the wrong Python interpreter as well:
* `Python_EXECUTABLE`
* `Python3_EXECUTABLE`
* `PYTHON_EXECUTABLE`
* `PYTHON3_EXECUTABLE`

Be sure that CMake's cache is purged when you rebuild with these new variables set by using the proper colcon option or deleting your `build` and `install` directories.