ros2/rclpy #1092: Setting string parameter with empty content or space only content fails from ros2cli

guihomework, 2023-03-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.03
- Installation type:
  - binaries
- Version or commit hash:
  - rolling
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

* in one terminal start a node (ex using python3)
```
import rclpy.node
import rclpy
rclpy.init()
node = rclpy.node.Node("mynode", allow_undeclared_parameters=True)
rclpy.spin(node)
```

* in a second terminal run one of the ros2cli command
```
ros2 param set /mynode myparam " "
ros2 param set /mynode myparam "  "
ros2 param set /mynode myparam ''
ros2 param set /mynode myparam ' '
ros2 param load param.yaml
```

#### Expected behavior
Set parameter successful
At minimum for the quoted space, the value should be a quoted space, which is a valid string character, but I expect a quoted empty string to be a valid string as well.

#### Actual behavior
```
Traceback (most recent call last):
  File "/opt/ros/rolling/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.21.0', 'console_scripts', 'ros2')())
  File "/opt/ros/rolling/lib/python3.10/site-packages/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/rolling/lib/python3.10/site-packages/ros2param/command/param.py", line 39, in main
    return extension.main(args=args)
  File "/opt/ros/rolling/lib/python3.10/site-packages/ros2param/verb/set.py", line 61, in main
    parameter.value = get_parameter_value(string_value=args.value)
  File "/opt/ros/rolling/local/lib/python3.10/dist-packages/rclpy/parameter.py", line 230, in get_parameter_value
    value.string_value = yaml_value
  File "/opt/ros/rolling/local/lib/python3.10/dist-packages/rcl_interfaces/msg/_parameter_value.py", line 239, in string_value
    assert \
AssertionError: The 'string_value' field must be of type 'str'
```

OR for the load
```
Traceback (most recent call last):
  File "/opt/ros/rolling/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.21.0', 'console_scripts', 'ros2')())
  File "/opt/ros/rolling/lib/python3.10/site-packages/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/rolling/lib/python3.10/site-packages/ros2param/command/param.py", line 39, in main
    return extension.main(args=args)
  File "/opt/ros/rolling/lib/python3.10/site-packages/ros2param/verb/load.py", line 53, in main
    load_parameter_file(node=node, node_name=node_name, parameter_file=args.parameter_file,
  File "/opt/ros/rolling/lib/python3.10/site-packages/ros2param/api/__init__.py", line 49, in load_parameter_file
    future = client.load_parameter_file(parameter_file, use_wildcard)
  File "/opt/ros/rolling/local/lib/python3.10/dist-packages/rclpy/parameter_client.py", line 307, in load_parameter_file
    param_dict = parameter_dict_from_yaml_file(parameter_file, use_wildcard)
  File "/opt/ros/rolling/local/lib/python3.10/dist-packages/rclpy/parameter.py", line 319, in parameter_dict_from_yaml_file
    return _unpack_parameter_dict(namespace, param_dict)
  File "/opt/ros/rolling/local/lib/python3.10/dist-packages/rclpy/parameter.py", line 341, in _unpack_parameter_dict
    parameter.value = get_parameter_value(str(param_value))
  File "/opt/ros/rolling/local/lib/python3.10/dist-packages/rclpy/parameter.py", line 230, in get_parameter_value
    value.string_value = yaml_value
  File "/opt/ros/rolling/local/lib/python3.10/dist-packages/rcl_interfaces/msg/_parameter_value.py", line 239, in string_value
    assert \
AssertionError: The 'string_value' field must be of type 'str'
```
#### Additional information

The culprit of this behavior is yaml.safe_load() that will generate a NoneType for "", '', " ", ' ' 
```
type(yaml.safe_load(" "))
<class 'NoneType'>
```
I can understand that strings must be passed through `yaml.safe_load`, but if the result is _NoneType_, it means that the `safe_load` did not generate a _ParseError_ or any other exception, and thus the _string_value_ should be used as is.
I tried that solution
`_value.string_value = yaml_value if yaml_value is not None else string_value`

which sets the value correctly, when the content is empty or space(s) 

( I must mention that setting it once does not set the type, setting it twice does (for any param set call on allow_undeclared_parameters=True)  but this is not related to that issue)

Barry-Xu-2018, 2023-03-13:
> I can understand that strings must be passed through yaml.safe_load, but if the result is NoneType, it means that the safe_load did not generate a ParseError or any other exception, and thus the string_value should be used as is.
> I tried that solution
_value.string_value = yaml_value if yaml_value is not None else string_value

I think you are right.  Current code didn't consider None return from yaml.safe_load().   
Maybe you can create a PR to fix this.

guihomework, 2023-03-13:
I can and will, but this won't fix the secondary issue of needing to set the param twice to make it "typed" and will therefore introduce a failure for describe/list --param-type

fujitatomoya, 2023-03-13:
> At minimum for the quoted space, the value should be a quoted space, which is a valid string character, but I expect a quoted empty string to be a valid string as well.

So do I.

@clalancette @wjwwood what do you think?

guihomework, 2023-03-13:
Just to inform, that although the fix is simple, I wanted to create a PR including a test that fails without the fix, and this proves to be lot harder than expected. 
First because I could not build rclpy on rolling, even with multiple other packages, there was always something missing (due to the new service introspection). So I went for building 3.7.1 which is my current version on binary rolling. 
Second, I somehow do not manage to call a single pytest even using the documented command
`colcon test --packages-select rclpy  --pytest-args -k test_create_string_parameter`
still runs 900 tests or more.

So my progress is rather slow. 
sorry for failing on such basic things.

guihomework, 2023-03-14:
This was quite an adventure. It took me a while to realize that 3.7.1 has a test_parameter_client.py but the test was not run at all due to a missing declaration in `CMakeLists.txt` in that version (fine in rolling)
Then I noticed that through the service client call, as `SetParameter` with space or empty string does not fail.
This comes from the fact that there are 2 functions `get_parameter_value`:
 * one is a method of `Parameter` 
 * one is a stand-alone function with a different implementation. 

Only the second one involves yaml and has issues with empty strings.

wjwwood, 2023-03-14:
> > At minimum for the quoted space, the value should be a quoted space, which is a valid string character, but I expect a quoted empty string to be a valid string as well.
> 
> So do I.
> 
> @clalancette @wjwwood what do you think?

I don't quite understand, maybe you could give a list of example inputs and outputs and I could comment on each?

It's possible we could restructure the input string in a certain way so that `yaml.safe_load()` returns what we would expect.

guihomework, 2023-03-14:
@wjwwood thanks for looking at this issue. I updated my initial bug report to specify what I await when setting and getting a space or an empty string. Currently doing so raises an exception.

and for confirmation, this is only happening with `ros2 param set` but works with the service call
```
> ros2 service call /mynode/set_parameters rcl_interfaces/srv/SetParameters "parameters: [{name: 'myparam', value: {type: 4, string_value: ' '}}]" 
requester: making request: rcl_interfaces.srv.SetParameters_Request(parameters=[rcl_interfaces.msg.Parameter(name='myparam', value=rcl_interfaces.msg.ParameterValue(type=4, bool_value=False, integer_value=0, double_value=0.0, string_value=' ', byte_array_value=[], bool_array_value=[], integer_array_value=[], double_array_value=[], string_array_value=[]))])

response:
rcl_interfaces.srv.SetParameters_Response(results=[rcl_interfaces.msg.SetParametersResult(successful=True, reason='')])

> ros2 param get /mynode myparam
String value is:(   )
```

guihomework, 2023-03-14:
> It's possible we could restructure the input string in a certain way so that `yaml.safe_load()` returns what we would expect.

this led me to think of testing something else : handling bash automatic removing of the double quotes when args are passed to `set`
```
>ros2 param set /mynode myparam '""'
Set parameter successful
> ros2 param get /mynode myparam
String value is:()
>ros2 param set /mynode myparam '" "'
Set parameter successful
> ros2 param get /mynode myparam
String value is:( )
```
this all works with single-quotes around double-quoted strings (and for double-quotes around single-quoted strings)

I still find that not intuitive that for a quoted space one needs to single-quote this double-quote so that it works as expected.  The work-around should be in code not at the user side.
I am not sure one can avoid the loss of the double-quotes from bash.

PR #1093 (2023-03-16): Allow space or empty strings when using ros2 param set
Should fix #1092
closed 2023-03-16 by clalancette