ros2/rclpy #1097: Feature request: Get all declared parameters (and print them)

zkytony, 2023-03-27:
## Feature Request

#### Feature description

I liked that in ROS1, when running roslaunch, it prints out all the parameters and their values. It is great for logging and debugging. I hope to do something like this more easily with ROS 2.

I wish there is a 'get_parameter_names' that returns all declared (and other important) parameters when I don't pass in anything. Or, when running 'ros2 launch', it prints out all parameters of each node being launched.

Currently (with ROS2 Humber), there is [get_parameters](https://docs.ros2.org/latest/api/rclpy/api/node.html?highlight=parameter#rclpy.node.Node.get_parameters) but it requires a list of names as input.

clalancette, 2023-03-27:
> I wish there is a 'get_parameter_names' that returns all declared (and other important) parameters when I don't pass in anything. Or, when running 'ros2 launch', it prints out all parameters of each node being launched.

The underlying functionality is all there; we have the call for `list_parameters` in rclcpp.  So this just needs an rclpy implementation.  If you are willing to take a shot, then you can look at https://github.com/ros2/rclcpp/blob/rolling/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp for some inspiration on how to implement it.

fujitatomoya, 2023-04-19:
@leeminju531 AFAIK, nobody is working on this. if you are willing to contribute, i am happy to do review 👍

leeminju531, 2023-04-30:
>  The underlying functionality is all there; we have the call for list_parameters in rclcpp. So this just needs an rclpy implementation. 

I believe that the functionality for listing all declared parameters is already present in rclpy. The implementation can be found in https://github.com/ros2/rclpy/blob/rolling/rclpy/rclpy/parameter_service.py#L100#L140 .
From my understanding, this implementation looks for parameters based on a given prefix or returns all parameters if no prefix is provided. therefore, it may not be necessary to implement this feature.

If not, please let me know your thoughts on how to proceed.

fujitatomoya, 2023-05-01:
@leeminju531 

> I believe that the functionality for listing all declared parameters is already present in rclpy. The implementation can be found in https://github.com/ros2/rclpy/blob/rolling/rclpy/rclpy/parameter_service.py#L100#L140 .

this is parameter service callback. so that it can prints all parameters via service.

i think the feature request here is to implement the method that belongs to `Node`? to print the all parameters that the node owns.