ros2/rclpy #1099: Cancelling asynchronous tasks has no effect?

haudren-woven, 2023-03-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Binary
- Version or commit hash:
  - Foxy
- DDS implementation:
  - Cyclone
- Client library (if applicable):
  - rclpy
  - 
#### Steps to reproduce issue

```python
import rclpy

async def test_cancel():
    future = rclpy.task.Future()
    future.cancel()

    print(future, future.cancelled(), future.done())

    return await future

if __name__ == "__main__":
    rclpy.init()

    node = rclpy.create_node("test")
    node.create_timer(0.1, lambda: None)

    executor = rclpy.get_global_executor()
    executor.add_node(node)
    task = executor.create_task(test_cancel)

    executor.spin_until_future_complete(task)
```

#### Expected behavior
This should terminate immediately, raising a `CancelledError` or equivalent

#### Actual behavior
This runs forever

#### Additional information
This is related to #1098 , as I was experimenting with exceptions and asynchronous tasks. I was wondering what happens when you `await` a task that's cancelled. since in `asyncio` it raises a `CancelledError`. It turns out that in `rclpy` it does call the done callbacks, but the task is not marked as done / doesn't raise.

I'm guessing that we need to adjust this logic:
https://github.com/ros2/rclpy/blob/f6d652fa2d5d3a8164cc69fb543ae0693e958262/rclpy/rclpy/task.py#L56

To be something like:
```python
        while not self._done and not self._cancelled:
            yield
        if self._cancelled:
            raise CancelledError()
        return self.result()
```

But I'd like to hear the maintainers' opinion first!

weber-niklas, 2024-11-03:
Is there any updates here or plans to adjust the code to the suggested version?

fujitatomoya, 2024-11-03:
@weber-niklas i think noone is working on this at this moment, besides Foxy is already E.O.L, we probably need to check if the same issue can happen with rolling and available distros. i will try to take a look in this week.

tonygoldman, 2024-11-03:
It's not fixed in rolling...
In `_wait_for_ready_callbacks` only done tasks are filtered, cancelled tasks remain executed.
https://github.com/ros2/rclpy/blob/rolling/rclpy/rclpy/executors.py#L612
Also awaiting a cancelled task blocks forever, as the `__await__` method returns only when the task is done.
https://github.com/ros2/rclpy/blob/f6d652fa2d5d3a8164cc69fb543ae0693e958262/rclpy/rclpy/task.py#L56-L57

> @weber-niklas i think noone is working on this at this moment, besides Foxy is already E.O.L, we probably need to check if the same issue can happen with rolling and available distros. i will try to take a look in this week.

weber-niklas, 2024-11-04:
I have created another a little more complex example illustrating the problem

```python

import rclpy
from rclpy.node import Node
from rclpy.task import Future
from rclpy.callback_groups import MutuallyExclusiveCallbackGroup

def sleep_async(node: Node, time: float) -> Future:
    """
    Sleeps for a given amount of time.

    Creates a timer with the sleep time as frequency and destroys it on the first callback.

    args:
        node: The node to sleep on. Future will be created with the nodes executor.
        time: The time to sleep in seconds
    
    returns:
        A future that will be done after the given amount of time.
    """
    future = Future(executor=node.executor)
    timer = node.create_timer(timer_period_sec=time, callback=lambda: _timer_callback(future), callback_group=MutuallyExclusiveCallbackGroup())

    def _timer_callback(future: Future):
        future.set_result(None)
        node.destroy_timer(timer)

    return future

async def sleep(self, seconds):
        for i in range(seconds):
            await sleep_async(self, 1)
            print(f"Slept for {i} seconds")

if __name__ == "__main__":
    rclpy.init()

    node = rclpy.create_node("test")
    executor = rclpy.get_global_executor()
    executor.add_node(node)
    task = executor.create_task(sleep, node, 5)

    task.cancel()
    print(f"{task.cancelled()=}")
    executor.spin_until_future_complete(task)

    print(f"{task.done()=}")
    print(f"{task.cancelled()=}")
```
Expected output:

```
task.cancelled()=True
task.done()=False
task.cancelled()=True
```
Actual output:

```
task.cancelled()=True
Slept for 0 seconds
Slept for 1 seconds
Slept for 2 seconds
Slept for 3 seconds
Slept for 4 seconds
task.done()=True
task.cancelled()=False
```

fujitatomoya, 2024-11-08:
@tonygoldman @weber-niklas if you can share the feedback to https://github.com/ros2/rclpy/pull/1377, that would be appreciated.

closed 2025-01-15 by fujitatomoya