ros2/rclpy #1105: Add ParamEventHandler to rclpy

EsipovPA, 2023-04-07:
## Add ParamEventHandler to rclpy

#### Feature description

Add ParameterEventHandler class to the rclpy lib, as an analogy to the same class in the rclcpp lib.
Might improve python ros2 users experience. May make python code more compact in some cases.

#### Implementation considerations

Not really a game-changing improvement. The expected behavior may be achieved with available tools.
On the conceptual level, it looks like the implementation from rclcpp might be ported to python to do the trick. Or maybe there are some other approaches

#### Problem at hand
I've encountered a need to monitor a parameter changes in a number of nodes in my python-api-node.
As far as I understand, there is the [ParameterClient class](https://github.com/ros2/rclpy/blob/rolling/rclpy/rclpy/parameter_client.py), that has the [on_parameter_event](https://github.com/ros2/rclpy/blob/rolling/rclpy/rclpy/parameter_client.py#L338) method, that gives me the ability to setup callbacks to parameters change. Also, there is a direct approach, which is to subscribe to parameter_events topic and monitor changes "manually". The last one looks like the best option if using ros-humble or ros-galactic (my case). In my opinion, code for the second approach looks something like this:

    class MyNode(rclpy.node.Node):
        def __init__(self, *args, **kwargs):
            super().__init__(...)
            ...

            self._param_change_subscription(
                rcl_interfaces.msg.ParameterEvent,
                "/parameter_events",
                ...,
                self.my_param_event_callback
            )

        def my_param_event_callback(
            self,
            node_name: str,
            param_name: str,
            msg: rcl_interfaces.msg.ParameterEvent
        ):
            # Suppose I want to log a parameter value change
            if msg.node == node_name:
                for param in msg.changed_parameters:
                    if param.name == param_name:
                        self.logger.debug(f"I see a parameter change: {param.name}: {param.value}")

And if the ParameterEventHandler, the code might be a little shorter:

    class MyNode(rclpy.node.Node):
        def __init__(self, *args, **kwargs):
            super().__init__(...)
            ...
            self._parameter_event_handler = rclpy.ParameterEventHandler
            self._param_handle = self._parameter_event_handler.add_parameter_callback(
                 some_param_name,
                 some_node_name,
                 self.my_param_event_callback
            )

        def my_param_event_callback(self, param: rcl_interfaces.msg.Parameter):
            self.logger.debug(f"I see a parameter change: {param.name}: {param.value}")

The first solution is perfectly filne, but the second one looks more compact and supporter-friendly.
Maybe someone is already working on this feature, but I did not find the discussion on this topic in the issues here or at [discourse](https://discourse.ros.org/search?q=rclpy)

EsipovPA, 2023-05-22:
Trying to add this feature myself in this [repo](https://github.com/EsipovPA/rclpy). Struggling with building ros from sources, to run build and run tests for the new class.

fujitatomoya, 2023-05-22:
> Maybe someone is already working on this feature, but I did not find the discussion on this topic in the issues here or at [discourse](https://discourse.ros.org/search?q=rclpy)

AFAIK, nobody is working on this.

> But the second one looks more compact and supporter-friendly.

besides, `ParameterEventHandler` (based on `rclcpp` implementation) could provide more useful things,

- only creates single subscription endpoint.
- multiple callbacks can be added and removed.
- parameter specific and multiple callbacks can be added and removed.
- method to get the parameter from the event, such as `get_parameter_from_event`

after all, i think this is reasonable.

cyborgdennett, 2023-06-25:
any news on this? I support this feature

EsipovPA, 2023-06-25:
Completed the class itself. Added 2 tests. 6 tests to go

EsipovPA, 2023-06-27:
Opened the PR: https://github.com/ros2/rclpy/pull/1135
Not sure about the Callbacks class declared in the ParameterEventHandler class. It contains all the callback control logic and a mutex to synchronize this control. Although it seems that this class takes on too much. Maybe it's better to reduce this class or get rid of it. I think this is debatable. Will add a comment on this to the PR dicussion.

EsipovPA, 2023-07-14:
It seems like the the PR is stuck somewhere in review. I was wondering if there is anything I can do to make that process move forward somehow?

EsipovPA, 2023-10-04:
Merged here: https://github.com/ros2/rclpy/pull/1135

EsipovPA, 2023-10-04:
Merged here:

closed 2023-10-04 by EsipovPA

gordon-n-stevenson, 2024-06-03:
Come into this feature a bit late, we work on humble at the moment - @EsipovPA would this be possible to backport to humble IYHO? I'd be happy to try and do a PR on this if you are flat out!

EsipovPA, 2024-06-04:
Hello @gordon-n-stevenson !

Thank you for the interest in this feature! :blush:

As far as I understand the contribution practices in this repo, it is possible to backport a feature to a release that is already out. Though it seems to me that mostly bugfixes and critical improvements are being backported, and this feature is more of a quality of life improvement.

If it is possible, I would like to ask what are the maintainers thoughts on this. @fujitatomoya what do you think about backporting this feature to humble? Anyway, I'll be glad to make a backport PR, if it's OK.

fujitatomoya, 2024-06-04:
https://github.com/ros2/rclpy/pull/1135 is rclpy, and purely added new class, so i think it is backport-able.
but i would like to confirm this because usually we do not backport new feature like this to existed distro? @clalancette what do you think?

btw, this should be added to https://docs.ros.org/en/rolling/Releases/Release-Jazzy-Jalisco.html#new-features-in-this-ros-2-release, this feature is enabled from Jazzy.

fujitatomoya, 2024-06-04:
> btw, this should be added to https://docs.ros.org/en/rolling/Releases/Release-Jazzy-Jalisco.html#new-features-in-this-ros-2-release, this feature is enabled from Jazzy.

https://github.com/ros2/ros2_documentation/pull/4510 addresses this.

gordon-n-stevenson, 2024-06-05:
Understood! I think better for us to think about a step to Jazzy rather than backporting new features