ros2/rclpy #1106: Make rclpy.context.Context be a context manager

felixdivo, 2023-04-09:
## Feature request

#### Feature description
`rclpy.context.Context` would be useful to have as a [context manager](https://docs.python.org/3/reference/datamodel.html#context-managers). This would allow for:
```python
with Context() as context:
    executor = MultiThreadedExecutor(context=context)
    do_something_cool(executor)
```

Instead of
```python
context = Context()
try:
    context.init()
    executor = MultiThreadedExecutor(context=context)
    do_something_cool(executor)
finally:
    context.try_shutdown()
```
The first approach is shorter and (for some) conceptually easier. In both cases `context` can be referenced afterwards although it is invalid. However, in the first case it is much more clear that any sane use is within the `with context: ...` construct. And, one cannot forget to close it in the end. It is arguably more Pythonic.

#### Implementation considerations
One could use `shutdown` or `try_shutdown` at the end of the context block. The latter makes more sense, since it allows for more complex scenarios where the context is already in some of the cases.

clalancette, 2023-04-20:
This is a good idea, thanks.  We probably aren't going to work on this, but we would definitely review pull requests that implemented it.

felixdivo, 2023-04-21:
Okay nice. Is wrapping around `context.init()`/`context.try_shutdown()` even the right thing to do? Then maybe I can have a look. 😄

mjcarroll, 2023-04-21:
Yep, I think it maps the ROS concept cleanly into a more pythonic pattern.  Happy to take a look if you open a PR.

felixdivo, 2023-04-29:
There is also #1118, for `rclpy.executor.Executor`

clalancette, 2024-12-30:
We merged the PR implementing this a long time ago, so closing this out.

closed 2024-12-30 by clalancette