ros2/rclpy #1119: Provide a simpler construct for running nodes

felixdivo, 2023-04-25:
## Feature request

#### Feature description
Running a node should be easy. Currently, you need something like:

```python
import rclpy
from local_stuff import MyNodeClass

if __name__ == "__main__":
    rclpy.init()
    try:
        rclpy.spin(MyNodeClass())
    except KeyboardInterrupt:
        pass
    finally:
        rclpy.shutdown()
```

Instead, the user could write:

```python
import rclpy
from local_stuff import MyNodeClass

if __name__ == "__main__":
    with rclpy.console_runner() as runner:
        runner.spin(MyNodeClass())
```

Obviously, this should never replace the other options, but rather provide a very simple helper for a scenario that occurs frequently.

#### Implementation considerations

The key question would be the "returned" object of `rclpy.console_runner()`. An executor makes sense, although it does not support passing a `Node` to `spin()`. That meant users had to do `runner.add_node(MyNodeClass()); runner.spin()`. Also, one would not need to return anything at all, and users would just use the normal `rclpy.spin(MyNodeClass())`. This needs some discussions before implementation can start.

#### Further notes
The name of `rclpy.console_runner()` is just a first idea. Maybe, something else makes it clearer what it is and how to use it.

Related to #1117 & #1118.

clalancette, 2023-05-04:
With #1117 and #1118 merged, we can now do a bunch of this.  @felixdivo does this satisfy your request?  If so, can you please close this out?  If not, can you explain what else you'd like to see here?

felixdivo, 2023-05-04:
Well, since I authored both PRs, I am aware of them and but think they are not enough. One cannot achive the above simple example with them, right? The code does not become this simple:

```python
import rclpy
from local_stuff import MyNodeClass

if __name__ == "__main__":
    with rclpy.console_runner() as runner:
        runner.spin(MyNodeClass())
```

Please correct me if I a wrong, one does not want to bloat rclpy with redundant features!

But having such a thing would be very nice since people would not be able to forget handling of `Ctrl+C` and would not need to think about shutting down ROS.

hotung1027, 2025-04-02:
I guess if rclpy.node could provide an attach alike function, so that we could use the active node and manage it.