ros2/rclpy #1123: Race condition in  `ActionClient.*_goal_async` functions which result in a futures that never complete

apockill, 2023-05-02:
## Bug report

Hi all, thank you for maintaining this wonderful library! 

I've been encountering an intermittent bug that has left my head scratching. Unfortunately recently it started occurring more often, so I sat down and spent some time tracking down what was going on. Now, I've got a good idea of the problem- but I need some more expert eyes to decide on a good solution. I'll lay out all my findings and some perceived paths forward in this bug report.

**Required Info:**

- Operating System:
  - Ubuntu 22.04, within a docker image that is also running Ubuntu 22.04
- Installation type:
  - Binaries as installed from ros2 apt.
- Version or commit hash:
  - rclpy==3.3.7, python3.10
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclpy==3.3.7, python3.10

#### Steps to reproduce issue

There are multiple places where this race condition manifests itself. I will demonstrate how to reproduce it when calling `send_goal_async`. Take the `ActionClient.send_goal_async` and add a `sleep(#)`  directly below the line `sequence_number = self._client_handle.send_goal_request(request)`. For example:

```python3
    def send_goal_async(self, goal, feedback_callback=None, goal_uuid=None):
        """
        Send a goal and asynchronously get the result.

        The result of the returned Future is set to a ClientGoalHandle when receipt of the goal
        is acknowledged by an action server.

        :param goal: The goal request.
        :type goal: action_type.Goal
        :param feedback_callback: Callback function for feedback associated with the goal.
        :type feedback_callback: function
        :param goal_uuid: Universally unique identifier for the goal.
            If None, then a random UUID is generated.
        :type: unique_identifier_msgs.UUID
        :return: a Future instance to a goal handle that completes when the goal request
            has been accepted or rejected.
        :rtype: :class:`rclpy.task.Future` instance
        :raises: TypeError if the type of the passed goal isn't an instance of
          the Goal type of the provided action when the service was
          constructed.
        """

        if not isinstance(goal, self._action_type.Goal):
            raise TypeError()

        request = self._action_type.Impl.SendGoalService.Request()
        request.goal_id = (
            self._generate_random_uuid() if goal_uuid is None else goal_uuid
        )
        request.goal = goal
        sequence_number = self._client_handle.send_goal_request(request)
        sleep(5)   # <--- this line causes this race condition nearly 100% of the time

        if sequence_number in self._pending_goal_requests:
            raise RuntimeError(
                "Sequence ({}) conflicts with pending goal request".format(
                    sequence_number
                )
            )

        if feedback_callback is not None:
            # TODO(jacobperron): Move conversion function to a general-use package
            goal_uuid = bytes(request.goal_id.uuid)
            self._feedback_callbacks[goal_uuid] = feedback_callback

        future = Future()
        self._pending_goal_requests[sequence_number] = future
        self._goal_sequence_number_to_goal_id[sequence_number] = request.goal_id
        future.add_done_callback(self._remove_pending_goal_request)
        # Add future so executor is aware
        self.add_future(future)

        return future
```

#### Expected behavior
I would expect 'send_goal_async' to run a bit slower than usual,  then return a future. That future might complete quite quickly (or already be complete).

#### Actual behavior
The future never completes, and I see the following line in the logs
```bash
Ignoring unexpected goal response. There may be more than one action server for the action '{ACTION_NAME}'
```

#### Additional information

By my sleuthing, it appears that this error is caused by the following important lines being called:

**client:Thread1**
1. `send_goal_async` is called, and within it:
2. `sequence_number = self._client_handle.send_goal_request(request)`
3. `self._goal_sequence_number_to_goal_id[sequence_number] = request.goal_id`

**client:Thread2**
1. A response to a goal request is received in the client C bindings, causing it to return True to `is_ready(...)`
2. `execute(...)` is called
3. `if sequence_number in self._goal_sequence_number_to_goal_id:` evaluates to False, because `client:Thread1` has not reached its step 3
4. It ignores the `taken_data`, and thus the future will never have `set_result` called.

**server:Thread1**
1. A goal request is received, and responded to 

---
## Cause Summary
There is a race condition when sending goal requests, because the goal request can complete before the `sequence_number` is tracked in the `self._goal_sequence_number_to_goal_id` dictionary. Thus `execute` is called before it is tracked, and the `taken_data` is ignored, assumed to be a response from a duplicate action server.

I believe this same race condition affects all usages of the client API, such as `_cancel_goal_async`, or `_get_result_async`.

## Recommendations

1. Put a lock around every usage of `self._client_handle` so that python-side objects can be edited and synced up. That way if `execute` is called it will block until the `self._client_handle` isn't being used, and the `_goal_sequence_number_to_goal_id` will be filled in.
2. Alternatively, create locks for the various in-memory dictionaries that are used in `execute`: 
  - _goal_handles
  - _pending_goal_requests
  - _goal_sequence_number_to_goal_id
  - _pending_cancel_requests
  - _pending_result_requests
  - _result_sequence_number_to_goal_id

aditya2592, 2023-05-04:
Hi @apockill, I ran into the same issue. I was wondering if adding locks around dictionary `_goal_sequence_number_to_goal_id` will solve the problem or not. If there is a lock, it prevents data being written or read simultaneously but it doesnt synchronize the fact that `self._goal_sequence_number_to_goal_id[sequence_number] = request.goal_id` in `send_goal_async` needs to happen before `if sequence_number in self._goal_sequence_number_to_goal_id` check in `execute` ?

I believe to synchronize that, we would need some an event that is set only in `send_goal_async` and waited on in `execute`.

apockill, 2023-05-10:
Any thoughts on the best way to resolve this bug? At the moment this affects all users of ActionClients. I can contribute better once there's guidance from maintainers :grin:

aditya2592, 2023-05-10:
It seems like the change I attempted didn't fully fix the problem, still see the following on some compute platforms
```
Ignoring unexpected goal response. There may be more than one action server for the action
Ignoring unexpected result response. There may be more than one action server for the action
```

aditya2592, 2023-05-11:
I tried looking deeper in `client.py` and it seems a simple condition/lock wont solve the problem because it doesn't guarantee that the dictionaries are updated *before* they are checked in `async def execute`. So I attempted to use a `threading.Event` to synchronize dictionary write and read. However each goal, result, cancel cases in `async def execute` need their own events to avoid write of one wrongly triggering the read of another. Trying this now in the existing PR https://github.com/ros2/rclpy/pull/1125

apockill, 2023-05-11:
I don't fully understand how a lock doesn't fix the issue in this case. Are you certain you were running the code with the lock when you observed this issue? 😅. It guarantees the dictionaries are updated at least in the send_goal_async case, because the execute function can't receive a sequence number until the client has been used in send_goal_async (which is inside a lock). Unless I am misunderstanding this race condition?

I have a similar fix on my end (I subclass ActionClient and lock the entirety of send_goal_async and execute) and it has solved the problem for me.

adityapande-1995, 2023-05-11:
Hello @apockill ! Other than the ``sleep()`` example, what is your use case where you're seeing this bug ? Is it a flaky one ?

apockill, 2023-05-11:
Hi @adityapande-1995! I gave the sleep example just to show how this race condition is purely caused by flaky timing, and luck currently prevents most users from seeing it. Also, a much much smaller wait can also trigger it. 

I'm seeing this flakiness happen on a regular basis on my stack. We happen to call actions on a regular basis (maybe 1-3 actions per second, on average). This bug  will reveal itself sometimes minutes after startup, other times within a few hours of runtime. 

What mainly concerns me is that it's purely timing luck that this works at all, and the error message is confusing enough that other users might run into this but not know what happened. After all, the symptom is a confusing log message and a future that will never complete.

If we could find a way to protect the relevant python state tied to sequence_number's, I believe that will solve the problem.

aditya2592, 2023-05-12:
> I don't fully understand how a lock doesn't fix the issue in this case. Are you certain you were running the code with the lock when you observed this issue? 😅. It guarantees the dictionaries are [...]

You are right, lock does seem to solve the issue. Once I fixed the issue caused by the race condition I had another unrelated bug in my setup which was misleading me to believe that the issue is not fixed. 

@adityapande-1995. Adding another viewpoint here: the dictionaries like `_goal_sequence_number_to_goal_id` are being edited in two threads: the thread that user is sending the goal request in and the thread in which the node spins which calls `async def execute`. So the race condition here is evident from the code itself.

apockill, 2023-05-17:
On another note- @adityapande-1995:
Thoughts on a solution where there's some decorator called `uses_cpp_client` which can wrap all of the functions in the `ActionClient` which use the underlying C++ actionclient?
For example:

```python3
def uses_cpp_client(fn):
    def wrapper(self, *args: Any, **kwargs: Any) -> Any:
        with self._cpp_client_handle_lock:
            return fn()
        
    return wrapper

class ActionClient(Waitable):
    ...
   
    @uses_cpp_client
    def send_goal_async(...):
        ....

    @uses_cpp_client
    def _cancel_goal_async(...):
        ....

    @uses_cpp_client
    def _get_result_async(...):
        ...

    @async_uses_cpp_client
    async def execute(...):
        ...
```

apockill, 2023-06-14:
I was looking at the code for service clients and it also uses a lock to sync the C++ state (`self.__client` calls) with the python state (`self.__pending_requests`). I think this highlights that the issue on the action client side is just an oversight, and that using locks is the way to go.

I'll submit a PR!

fujitatomoya, 2023-06-14:
@apockill thanks for being patient on this.

> I think this highlights that the issue on the action client side is just an oversight, and that using locks is the way to go.

i think that is correct assumption we have here. `_pending_requests` is protected by `_lock` in the service client.

apockill, 2023-08-03:
Hi @clalancette, thank you for looking into this! I've completely forgotten to submit a PR for this. 

Do you think that the difference might be in the executor? Come to think of it, I'm guessing this would only occur in a MultiThreadedExecutor case, since a SingleThreadedExecutor wouldn't be able to have a race condition like this.

clalancette, 2023-08-03:
> Do you think that the difference might be in the executor? Come to think of it, I'm guessing this would only occur in a MultiThreadedExecutor case, since a SingleThreadedExecutor wouldn't be able to have a race condition like this.

Yeah, that very well may be the case.  If there is a simple example I can use to try to reproduce it, I'm certainly happy to take a look and review a PR.

apockill, 2023-08-04:
Absolutely! Here's an example that replicated it on my end. You might have to switch out `custom_msgs` for some other action that takes no parameters. 

`client.py`
```python3
from time import sleep

import rclpy
from rclpy import Future
from rclpy.action import ActionClient
from rclpy.callback_groups import MutuallyExclusiveCallbackGroup
from rclpy.executors import MultiThreadedExecutor
from rclpy.node import Node

from custom_msgs.action import Home as SomeMsg

class RaceyActionClient(ActionClient):
    def send_goal_async(self, goal, feedback_callback=None, goal_uuid=None):
        """
        Send a goal and asynchronously get the result.

        The result of the returned Future is set to a ClientGoalHandle when receipt of the goal
        is acknowledged by an action server.

        :param goal: The goal request.
        :type goal: action_type.Goal
        :param feedback_callback: Callback function for feedback associated with the goal.
        :type feedback_callback: function
        :param goal_uuid: Universally unique identifier for the goal.
            If None, then a random UUID is generated.
        :type: unique_identifier_msgs.UUID
        :return: a Future instance to a goal handle that completes when the goal request
            has been accepted or rejected.
        :rtype: :class:`rclpy.task.Future` instance
        :raises: TypeError if the type of the passed goal isn't an instance of
          the Goal type of the provided action when the service was
          constructed.
        """
        if not isinstance(goal, self._action_type.Goal):
            raise TypeError()

        request = self._action_type.Impl.SendGoalService.Request()
        request.goal_id = (
            self._generate_random_uuid() if goal_uuid is None else goal_uuid
        )
        request.goal = goal
        sequence_number = self._client_handle.send_goal_request(request)

        sleep(5)  # <--- this line causes this race condition nearly 100% of the time

        if sequence_number in self._pending_goal_requests:
            raise RuntimeError(
                "Sequence ({}) conflicts with pending goal request".format(
                    sequence_number
                )
            )

        if feedback_callback is not None:
            # TODO(jacobperron): Move conversion function to a general-use package
            goal_uuid = bytes(request.goal_id.uuid)
            self._feedback_callbacks[goal_uuid] = feedback_callback

        future = Future()
        self._pending_goal_requests[sequence_number] = future
        self._goal_sequence_number_to_goal_id[sequence_number] = request.goal_id
        future.add_done_callback(self._remove_pending_goal_request)
        # Add future so executor is aware
        self.add_future(future)

        return future

class SomeActionClient(Node):
    def __init__(self):
        super().__init__("action_client")
        self._action_client = RaceyActionClient(
            self,
            SomeMsg,
            "some_action",
            callback_group=MutuallyExclusiveCallbackGroup(),
        )
        self.call_action_timer = self.create_timer(
            timer_period_sec=10,
            callback=self.send_goal,
            callback_group=MutuallyExclusiveCallbackGroup(),
        )

    def send_goal(self):
        goal_msg = SomeMsg.Goal()
        # self._action_client.wait_for_server()
        result = self._action_client.send_goal(goal_msg)
        print(result)

def main(args=None):
    rclpy.init(args=args)

    action_client = SomeActionClient()

    executor = MultiThreadedExecutor()
    rclpy.spin(action_client, executor)

if __name__ == "__main__":
    main()
```

`server.py`
```python3
import rclpy
from rclpy.action import ActionServer
from rclpy.node import Node

from custom_msgs.action import Home as SomeMsg

class SomeActionServer(Node):
    def __init__(self):
        super().__init__("fibonacci_action_server")
        self._action_server = ActionServer(
            self, SomeMsg, "some_action", self.execute_callback
        )

    def execute_callback(self, goal_handle):
        self.get_logger().info("Executing goal...")
        result = SomeMsg.Result()
        return result

def main(args=None):
    rclpy.init(args=args)

    action_server = SomeActionServer()
    rclpy.spin(action_server)

if __name__ == "__main__":
    main()
```

Replication instructions:
1) Run `server.py` on one terminal
2) Run `client.py` on one terminal

Observe `client.py` calls failing. It will call every 10 seconds, however it will only actually end up getting called once since the race condition occurs and locks up a thread:
```
[WARN] [1691123997.104652764] [action_client]: Ignoring unexpected goal response. There may be more than one action server for the action 'some_action'
```

mpollayil, 2023-10-12:
Are there any updates on this? At our company, we are also experiencing the same issue.

tompe17, 2023-12-18:
I am also seeing this when loading the system.  Using the same callback group (Reentrant) did not solve the issue. I will try the the other suggested workaround.

Is it supposed to be fixed in the current humble or not or what is the progress?  This is a show stopper for the application I have...

Fingerling42, 2024-06-15:
Still experiencing this issue on the current Humble release.

In my case, I encountered this in NavigateToPose action from nav2. If I call this action sequentially (for several poses) and insert `time.sleep` between them to make some routine after one goal is succeeded, then sometimes this error (`Ignoring unexpected goal response`) occurs.

firesurfer, 2024-06-25:
Looks like the issue also still existent in iron. 
I agree with the other posters that this is an absolute showstopper for the usage of actions in any industrial system and should finally be prioritized by the ROS2 core team. 

@clalancette if there is anything that can be done from a user side to help fixing this issue please let us know!

The rclcpp side btw. seems to be finally fixed by https://github.com/ros2/rclcpp/pull/2531 (at least in iron)

brta-jc, 2024-07-09:
+1 we're also encountering this error. We're on Humble.

akupferb, 2024-08-20:
Also still encountering this error on Humble...