ros2/rclpy #1127: Allow ActionClient  to get_status() of Goal from Server side

dcconner, 2023-05-17:
## Feature request

Seeing #1072, comments on https://answers.ros.org/question/406766/ros2-actions-publish-status-messages-that-get-exponentially-more-expensive/, and issue on  https://github.com/FlexBE/flexbe_behavior_engine/issues/3 

I'd like the ActionClient to be able to report the GoalStatus from the server side.

In ROS 1, we were able to monitor the goal status separately from the "feedback" topic.   

Is there a reason this was dropped for ROS 2?

Should the server publish the status topic as a regular topic?

I'm happy to develop a proposal and PR, but I'd like to understand the larger history and any known issues.

fujitatomoya, 2024-02-27:
> I'd like the ActionClient to be able to report the GoalStatus from the server side.

https://github.com/ros2/rclpy/blob/220d714b2b6da81a4abd6a804e3ed4ee8cfd7c3f/rclpy/rclpy/action/client.py#L68-L70

this does not work???

> I see there is a status_pub_qos_profile, but it does not appear to be used in server code.

rcl_action_server_init takes it and used?

https://github.com/ros2/rclpy/blob/220d714b2b6da81a4abd6a804e3ed4ee8cfd7c3f/rclpy/src/rclpy/action_server.cpp#L101

dcconner, 2024-05-02:
@fujitatomoya the status propery belongs to the GoalHandle.  I've been using an async call that returns a future 

`
        future = client.send_goal_async(new_goal,
                                        feedback_callback=lambda f: ProxyActionClient._feedback_callback(topic, f)
                                        )
        ProxyActionClient._result_status[topic] = GoalStatus.STATUS_ACCEPTED
        future.add_done_callback(partial(ProxyActionClient._done_callback, topic=topic))

    @classmethod
    def _done_callback(cls, future, topic):
        ProxyActionClient._current_goal[topic] = future
        result = future.result().get_result_async()
        result.add_done_callback(partial(ProxyActionClient._result_callback, topic=topic))

    @classmethod
    def _result_callback(cls, future, topic):
        result = future.result().result
        result_status = future.result().status

`
Which lets me get the final status properly. 

Can I query "STATUS_ACCEPTED", "EXECUTING", and "CANCELING" while waiting on the future using the Python client?