ros2/rclpy #1141: Client successfully calls service, but does not return result

Schteve-Earl, 2023-07-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binary
- Client library (if applicable):
  - rclpy

#### Problem code
```
class ImageCaptureControlNode(Node):
    def __init__(self):
        super().__init__("image_capture_control_node")
        self.initialize_parameters()
        self.load_parameters()
        self.initialize_publishers()
        self.initialize_clients()
        self.initialize_services()

    def initialize_parameters(self):
        self.declare_parameter("camera_names", ["camera_1"])
        self.declare_parameter("trigger_topic", "trigger_cameras")
        self.declare_parameter("packaged_topic", "captured_images")

    def load_parameters(self):
        self.camera_names = (
            self.get_parameter("camera_names").get_parameter_value()
            .string_array_value
        )
        self.trigger_topic = (
            self.get_parameter("trigger_topic").get_parameter_value()
            .string_value
        )
        self.packaged_topic = (
            self.get_parameter("packaged_topic").get_parameter_value()
            .string_value
        )

    def initialize_publishers(self):
        """Initializes all the required publishers
        """
        self.image_pair_publisher = self.create_publisher(
            ImagePairs, self.packaged_topic, 10
        )

    def initialize_clients(self):
        """Initiailizes all the required service clients
        """
        self.camera_clients = dict()

        for name in self.camera_names:
            client = self.create_client(CameraTrigger, f'{name}/trigger')
            if not client.wait_for_service(timeout_sec=1.0):
                self.get_logger().error(f"Service {name}/trigger not available")
                rclpy.shutdown()
            self.camera_clients[name] = client

    def initialize_services(self):
        """Initializes all the required services
        """
        self.image_capture_service = self.create_service(
            Trigger, self.trigger_topic, self.capture_callback
        )

    def capture_callback(self, request, response):

        # Prepare to store all captured images
        image_pairs = ImagePairs()

        # Call all camera services
        for camera_name, client in self.camera_clients.items():
            # Construct a request
            camera_request = CameraTrigger.Request()
            self.get_logger().info(f"calling {camera_name}/trigger")
            future = client.call_async(camera_request)
            self.get_logger().info(f"spinning until {camera_name}/trigger returns")
            rclpy.spin_until_future_complete(self, future)
            self.get_logger().info(f"{camera_name}/trigger returned")
            image_pairs.image_pairs.append(future.result().image_pair)
                    
        # Package all images into a single message
        response.success = True
        response.message = "Successfully captured all images"

        self.image_pair_publisher.publish(image_pairs)

        return response

def main(args=None):
    rclpy.init(args=args)
    node = ImageCaptureControlNode()

    rclpy.spin(node)

    node.destroy_node()
    rclpy.shutdown()

if __name__ == "__main__":
    main()

```

#### Expected behavior
The above code is a file that creates a class that contains a service and multiple clients. The callback for the service should call all available clients, and then add all the responses to a single message which it then publishes over a topic

#### Actual behavior
The code works up until these lines:
```
camera_request = CameraTrigger.Request()
self.get_logger().info(f"calling {camera_name}/trigger")
future = client.call_async(camera_request)
self.get_logger().info(f"spinning until {camera_name}/trigger returns")
rclpy.spin_until_future_complete(self, future)
self.get_logger().info(f"{camera_name}/trigger returned")
image_pairs.image_pairs.append(future.result().image_pair)
```
Instead, the client does call the service, and the service successfully returns a result. The code then hangs, and is unable to retrieve the result. I have tried using a synchronous call, like so:
```
camera_result = client.call(camera_request)
```
but the same issue occurs. The service is called and returns correctly, but the client hangs and will not progress.
#### Additional information
I have verified that the service being called is not the issue, I can call it from the command line and it reliably returns the correct results.

clalancette, 2023-07-20:
In short, this is due to the asynchronous nature of ROS 2.  The problem is that in order for a future to complete, the executor must be spinning.  But because you are blocking within a callback, the executor cannot spin.

The best way to fix this is to not block from within the callback.  Doing that mean that you are slowing down all other callbacks that want to run.  So probably the best thing in this code is to have the `capture_callback` kick off a thread that will make the service call and wait.  Then the capture callback can return immediately, letting other processing happen.

All of that said, this doesn't look like a bug so I'm going to close this out.  If you still need help with this, I'll suggest opening a question on https://answers.ros.org.

closed 2023-07-20 by clalancette

alvgaona, 2024-05-06:
I have experienced the same issue in C++. I had subscribed to the `camera/image` topic and then within the associated callback, I was calling a service and blocking that callback which caused the `future_response` to timeout. So I then used the `async_send_request(request, &callback)`, and waiting for the future within the second callback. It fixed my issue and the future was instantly ready to be consumed.