ros2/rclpy #1144: _rclpy_pybind11.cpython-310-x86_64-linux-gnu.so failed to be imported while being present on the system

squizz617, 2023-07-20:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04
- Installation type: Tried both binary and source installation
- Version or commit hash:
  - bin: `ros-humble-rclpy/jammy,now 3.3.8-2jammy.20230623.051015 amd64 [installed,automatic]`
  - src: `76212669`
- DDS implementation: Fast-DDS (default)
- Client library (if applicable): `rclpy`

#### Steps to reproduce issue
* Binary installation:
```
$ source /opt/ros/humble/setup.sh
$ python3 -c "import rclpy"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 49, in <module>
    from rclpy.signals import install_signal_handlers
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/signals.py", line 15, in <module>
    from rclpy.exceptions import InvalidHandle
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/exceptions.py", line 15, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/impl/implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
  File "/opt/ros/humble/lib/python3.10/site-packages/rpyutils/import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: librcl_action.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://docs.ros.org/en/{distro}/Guides/Installation-Troubleshooting.html#import-failing-even-with-library-present-on-the-system' for possible solutions

$ file /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so
/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=7a13e46bccf18deb10a3cfdb76b436aa3856fdda, stripped
```

* Source installation:
```
$ source ~/ros2_humble/install/setup.sh
$ python3 -c "import rclpy"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/seulbae/ros2_humble/install/rclpy/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 49, in <module>
    from rclpy.signals import install_signal_handlers
  File "/home/seulbae/ros2_humble/install/rclpy/local/lib/python3.10/dist-packages/rclpy/signals.py", line 15, in <module>
    from rclpy.exceptions import InvalidHandle
  File "/home/seulbae/ros2_humble/install/rclpy/local/lib/python3.10/dist-packages/rclpy/exceptions.py", line 15, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "/home/seulbae/ros2_humble/install/rclpy/local/lib/python3.10/dist-packages/rclpy/impl/implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
  File "/home/seulbae/ros2_humble/build/rpyutils/rpyutils/import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: librcl_action.so: cannot open shared object file: No such file or directory
The C extension '/home/seulbae/ros2_humble/install/rclpy/local/lib/python3.10/dist-packages/rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://docs.ros.org/en/{distro}/Guides/Installation-Troubleshooting.html#import-failing-even-with-library-present-on-the-system' for possible solutions

$ file /home/seulbae/ros2_humble/build/rclpy/test_rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so
/home/seulbae/ros2_humble/build/rclpy/test_rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=521156ac1b623401f2b012c6dcece989048e6273, not stripped
```

#### Expected behavior
`rclpy` successfully imported.

#### Actual behavior
Import failure.

#### Additional information
I'm not sure what's causing this issue. A ROS 2 scripts on my machine that worked before suddenly stopped working, and it turned out to be a import failure. Also, the trobleshooting guide in the error message seems to be only for Windows.
Thank you.

mjcarroll, 2023-07-20:
What is the output of `python3 -VV` ?

squizz617, 2023-07-20:
```
$ python3 -VV
Python 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]
```

mjcarroll, 2023-07-20:
Ah, I didn't read closely enough, this is likely the culprit: 

```
ImportError: librcl_action.so: cannot open shared object file: No such file or directory
```

Do you have `ros-humble-rcl-action` installed?  You should also have the library located at `/opt/ros/humble/librcl_action.so` from the binaries.

clalancette, 2023-07-20:
Further, what does your environment look like?

```
$ printenv | grep -i ros
```

squizz617, 2023-07-20:
I have `ros-humble-rcl-action` installed.
```
$ apt list | grep humble-rcl-action                                                                                           

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-humble-rcl-action-dbgsym/jammy 5.3.3-1jammy.20230623.045652 amd64
ros-humble-rcl-action/jammy,now 5.3.3-1jammy.20230623.045652 amd64 [installed,automatic]
```

Env when using the binary installation:
```
$ printenv | grep -i ros          
PATH=/opt/ros/humble/bin:/usr/local/texlive/2022/bin/x86_64-linux:/home/seulbae/.local/bin:/usr/local/texlive/2022/bin/x86_64-linux:/home/seulbae/.local/bin:/opt/gcc-arm-none-eabi-9-2020-q2-update/bin:/home/seulbae/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
AMENT_PREFIX_PATH=/opt/ros/humble
PYTHONPATH=/opt/ros/humble/lib/python3.10/site-packages:/opt/ros/humble/local/lib/python3.10/dist-packages
LD_LIBRARY_PATH=/opt/ros/humble/opt/rviz_ogre_vendor/lib:/opt/ros/humble/lib/x86_64-linux-gnu:/opt/ros/humble/lib
ROS_DISTRO=humble
ROS_LOCALHOST_ONLY=0
ROS_PYTHON_VERSION=3
ROS_VERSION=2
```

squizz617, 2023-07-20:
```
$ file /opt/ros/humble/lib/librcl_action.so 
/opt/ros/humble/lib/librcl_action.so: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=e1d3f18a4b279ed1c5c6aca4dd4332e5fd3f06e5, stripped
```

mjcarroll, 2023-07-27:
I'm not seeing anything obvious. Can you also run `ldd /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so`

squizz617, 2023-07-27:
```
seulbae@xps13 ~   [23-07-27 13:37:31]
$ source /opt/ros/humble/setup.zsh                                                                                
seulbae@xps13 ~   [23-07-27 13:37:32]
$ ldd /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so 
	linux-vdso.so.1 (0x00007fff79b99000)
	librcl_action.so => /opt/ros/humble/lib/librcl_action.so (0x00007f7f3459f000)
	librcl_lifecycle.so => /opt/ros/humble/lib/librcl_lifecycle.so (0x00007f7f34595000)
	libpython3.10.so.1.0 => /lib/x86_64-linux-gnu/libpython3.10.so.1.0 (0x00007f7f33e00000)
	librcl.so => /opt/ros/humble/lib/librcl.so (0x00007f7f34526000)
	librcl_yaml_param_parser.so => /opt/ros/humble/lib/librcl_yaml_param_parser.so (0x00007f7f3451a000)
	librmw_implementation.so => /opt/ros/humble/lib/librmw_implementation.so (0x00007f7f3450b000)
	librcl_logging_interface.so => /opt/ros/humble/lib/librcl_logging_interface.so (0x00007f7f34506000)
	librmw.so => /opt/ros/humble/lib/librmw.so (0x00007f7f344fa000)
	liblifecycle_msgs__rosidl_typesupport_c.so => /opt/ros/humble/lib/liblifecycle_msgs__rosidl_typesupport_c.so (0x00007f7f344f3000)
	librcutils.so => /opt/ros/humble/lib/librcutils.so (0x00007f7f344db000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f7f33a00000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f7f344b9000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f7f33600000)
	libtracetools.so => /opt/ros/humble/lib/libtracetools.so (0x00007f7f344b4000)
	liblifecycle_msgs__rosidl_generator_c.so => /opt/ros/humble/lib/liblifecycle_msgs__rosidl_generator_c.so (0x00007f7f344a4000)
	librosidl_runtime_c.so => /opt/ros/humble/lib/librosidl_runtime_c.so (0x00007f7f34499000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f7f34468000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f7f3444a000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f7f33d19000)
	librcl_logging_spdlog.so => /opt/ros/humble/lib/librcl_logging_spdlog.so (0x00007f7f34443000)
	librcl_interfaces__rosidl_typesupport_c.so => /opt/ros/humble/lib/librcl_interfaces__rosidl_typesupport_c.so (0x00007f7f3443a000)
	librcl_interfaces__rosidl_generator_c.so => /opt/ros/humble/lib/librcl_interfaces__rosidl_generator_c.so (0x00007f7f3441e000)
	libyaml.so => /opt/ros/humble/lib/libyaml.so (0x00007f7f343fc000)
	libament_index_cpp.so => /opt/ros/humble/lib/libament_index_cpp.so (0x00007f7f343ef000)
	librcpputils.so => /opt/ros/humble/lib/librcpputils.so (0x00007f7f343e1000)
	librosidl_typesupport_c.so => /opt/ros/humble/lib/librosidl_typesupport_c.so (0x00007f7f343db000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f7f34696000)
	libspdlog.so.1 => /lib/x86_64-linux-gnu/libspdlog.so.1 (0x00007f7f33c9e000)
	libbuiltin_interfaces__rosidl_generator_c.so => /opt/ros/humble/lib/libbuiltin_interfaces__rosidl_generator_c.so (0x00007f7f33c99000)
	libfmt.so.8 => /lib/x86_64-linux-gnu/libfmt.so.8 (0x00007f7f33c76000)
```

squizz617, 2023-07-27:
The issue persists after upgrading ros-humble apt package to `ros-humble-rcl-action/jammy,now 5.3.4-3jammy.20230721.204515 amd64 [installed,automatic]`..

mjcarroll, 2023-07-27:
I'm pretty stumped here.  I suppose as a next step, you could look at each of the dependent shared libraries and make sure that they can correctly resolve their dependencies via `ldd`?

iuhilnehc-ynos, 2023-07-28:
@squizz617 

I'm interested in this issue because it's almost impossible as the `ldd` shows that the `rcl_action.so` is found.

Could you run the command `strace python3 -c "import rclpy" > /tmp/failed-to-import-rclpy.txt 2>&1`, and then upload the log `/tmp/failed-to-import-rclpy.txt`?

iuhilnehc-ynos, 2023-08-10:
I believe that you change the `/usr/bin/python3.10` with '+s' that will erase the `LD_LIBRARY_PATH`.

squizz617, 2023-08-21:
>  I believe that you changed the /usr/bin/python3.10 with '+s' 

@iuhilnehc-ynos Thanks for the suggestion but I haven't set the setuid bit.. 

```
$ ls -alh /usr/bin/python3.10 
-rwxr-xr-x 1 root root 5.7M Jun 11 01:26 /usr/bin/python3.10
```

iuhilnehc-ynos, 2023-08-22:
Please check the `/usr/bin/python3` again.

iuhilnehc-ynos, 2023-08-22:
Wow, so there are other cases can make the [__libc_enable_secure](https://github.com/bminor/glibc/blob/glibc-2.35/elf/rtld.c#L2843) get true and then call `unsetenv` the `LD_LIBRARY_PATH`.

Besides the manually setting suid for the `/usr/bin/python3.10`, another case I can think of is the mount option.

@squizz617  Could you show the mount options for the device?

DRAGON-BOOSTERR, 2023-08-30:
I also have the same problem, for me the command import rclpy was not working in pycharm and after some messing around I found that we need to add the interpreter paths in pycharm. after that I ended up this error and by following this beautiful thread, Im looking for a way forward. but when I run the code with just the python command `python3 ros_test.py`, the program runs.

PS: I was following a udemy tutorial  on ROS2

chenyenru, 2023-08-31:
@DRAGON-BOOSTERR May you elaborate on how you solved the problem? I'm having the same problem too 😭

DRAGON-BOOSTERR, 2023-08-31:
@chenyenru yeah here is the stackover flow page that I raised the issue and answered myself 
[https://stackoverflow.com/questions/76983995/python-package-is-installed-but-it-doesnt-get-detected-in-a-ide-ros2/77009205#77009205](url)

chenyenru, 2023-09-01:
Thank you! I was able to solve it by deactivating my anaconda environment (don't know why but that way my Python interpreter finds `rclpy` package).

DRAGON-BOOSTERR, 2023-09-01:
@chenyenru @squizz617 Hey guys I think i found a short term fix for this issue! As said before I was having problems with IDEs not recognising the `rclpy` library of ROS2. As by the comment yesterday I mentioned how I solved the problem by adding the interpreter paths in Pycharm IDE. The Pycharm IDE tries to run the file inside the IDE or something like that and received the same error as the title of this thread. Today I tried the code with visual studio code, surprisingly VS code recognised that `rclpy` library and showed all the auto complete and other things of that library. Since VS code runs every file in terminal inside the IDE, so the code works without any issues. I think this might fix your problem with that error?

chenyenru, 2023-09-03:
I was able to include extra paths in the `settings.json` file, where the interpreter would successfully autosuggest functions from the `rclpy` library. 

However, when I tried to compile it on the IDE, it still showed that `rclpy` was not found. I was able to resolve this problem by not running it on `conda` environment and instead the base file while installing required packages with `pip`. Hope this result would help you too!

fferri, 2023-09-18:
I got the exact issue reported by OP.
Turned out I was overriding `LD_LIBRARY_PATH` for the process responsible for spawning `python3`.
Appending/prepending paths to `LD_LIBRARY_PATH` instead of overriding fixed the issue for me.

asukiaaa, 2023-10-22:
I could resolve this problem by removing all packages of using ros2 distribution and install again.
```sh
sudo apt remove ros-rolling-*
sudo apt install ros-rolling-desktop-full
```

2019lelexia, 2023-11-09:
According to this,
https://docs.ros.org/en/rolling/How-To-Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system
I think we cannot use different version of Python because the Python of ROS2 build different binary file of C extension from the Python of another

yerkgb, 2023-11-23:
I had a same problem. These two lines are helped me out:
`$sudo apt install --reinstall ros-humble-rclpy`
`$source /opt/ros/humble/setup.bash
`
Try your luck!

sungchan1, 2024-01-14:
I finally managed to solve the problem and wrote a blog post about it in Korean.

https://du-sungchan-24k.tistory.com/37

In summary, you need to directly inject this into your pycharm.sh script.

. /opt/ros/humble/setup.sh' 
If installed via Jetbrain Toolbox, your pycharm.sh will be located at

'/home/{USER_NAME}/.local/share/JetBrains/Toolbox/apps/{pycharm_version}/bin/pycharm.sh'. 
For Jetbrain Remote Development, pycharm.sh is in

'/home/{USER_NAME}/.cache/JetBrains/RemoteDev/dist/{cached_pycharm_version}/bin/pycharm.sh'. 
The same applies to CLion and IntelliJ.

nixrunner, 2024-03-04:
`nixrunner@fedora:/usr/lib64/ros2-iron$ ros2
Traceback (most recent call last):
  File "/usr/lib64/ros2-iron/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.25.4', 'console_scripts', 'ros2')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/ros2-iron/bin/ros2", line 25, in importlib_load_entry_point
    return next(matches).load()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/importlib/metadata/__init__.py", line 205, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/usr/lib64/ros2-iron/lib/python3.12/site-packages/ros2cli/cli.py", line 22, in <module>
    from rclpy.executors import ExternalShutdownException
  File "/usr/lib64/ros2-iron/lib/python3.12/site-packages/rclpy/__init__.py", line 49, in <module>
    from rclpy.signals import install_signal_handlers
  File "/usr/lib64/ros2-iron/lib/python3.12/site-packages/rclpy/signals.py", line 15, in <module>
    from rclpy.exceptions import InvalidHandle
  File "/usr/lib64/ros2-iron/lib/python3.12/site-packages/rclpy/exceptions.py", line 15, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "/usr/lib64/ros2-iron/lib/python3.12/site-packages/rclpy/impl/implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/ros2-iron/lib/python3.12/site-packages/rpyutils/import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: librosidl_typesupport_c.so: cannot open shared object file: No such file or directory
The C extension '/usr/lib64/ros2-iron/lib/python3.12/site-packages/rclpy/_rclpy_pybind11.cpython-312-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://docs.ros.org/en/{distro}/Guides/Installation-Troubleshooting.html#import-failing-even-with-library-present-on-the-system' for possible solutions
`

I am having a similar error. I faced with same error on Ubuntu 24.04

dheera, 2024-06-04:
Bump, same error or Ubuntu 24.04 + rolling
```
Python 3.11.5 (main, Sep 11 2023, 13:54:46) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import rclpy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/rolling/lib/python3.12/site-packages/rclpy/__init__.py", line 49, in <module>
    from rclpy.signals import install_signal_handlers
  File "/opt/ros/rolling/lib/python3.12/site-packages/rclpy/signals.py", line 15, in <module>
    from rclpy.exceptions import InvalidHandle
  File "/opt/ros/rolling/lib/python3.12/site-packages/rclpy/exceptions.py", line 16, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "/opt/ros/rolling/lib/python3.12/site-packages/rclpy/impl/implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/rolling/lib/python3.12/site-packages/rpyutils/import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dheera/miniconda3/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'
The C extension '/opt/ros/rolling/lib/python3.12/site-packages/_rclpy_pybind11.cpython-311-x86_64-linux-gnu.so' isn't present on the system. Please refer to 'https://docs.ros.org/en/rolling/How-To-Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system' for possible solutions
```

NirvanaDragon, 2024-06-19:
> Bump, same error or Ubuntu 24.04 + rolling
>
> ```
> Python 3.11.5 (main, Sep 11 2023, 13:54:46) [GCC 11.2.0] on linux
> Type "help", "copyright", "credits" or "license" for more information.
> Traceback [...]

You should check if the Python environment for ROS matches the current terminal's Python environment. I noticed your terminal is using Python 3.11, but ROS is looking for Python 3.12, which could be why it can't find the required library. This is just my guess.

Rizo11, 2024-06-21:
SOLUTION:
I had the same problem. I was running it in conda env. I truned off conda env, and it worked!

dholukeval, 2024-07-02:
`setcap` solved the problem for me. [here](https://answers.ros.org/question/366380/ubuntu1804ros2-dashing-failed-to-load-entry-point-issue/)

    sudo setcap -r /usr/bin/python3.10
    
This will remove extended capabilities form the executable. Change the python version according to your setup.

dhruvmsheth, 2024-07-14:
I had the same issue. I made a dumb error of changing PATH in the `bashrc` file to something else while making another installation and forgot to change it back. Removed those and it worked. As mentioned in the above comments, make sure to source the setup file only at the end of the `bashrc` file. Use `gedit ~/.bashrc` to check what's wrong with the file.

AndyZe, 2024-10-04:
I'm here because I was using IsaacSim, without ROS2 installed natively. Remove this line from `.bashrc` to get ROS2 working again natively:

`# export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$isaac_sim_package_path/exts/omni.isaac.ros2_bridge/humble/lib`

AnarchistKnight, 2024-10-30:
> ```shell
> sudo apt install ros-rolling-desktop-full
> ```

May I ask what is the python version of ROS2?

AnarchistKnight, 2024-10-30:
anyone kindly show the complete way to fix this issue, please?

sungchan1, 2024-10-30:
> anyone kindly show the complete way to fix this issue, please?

It’s been a while, but I solved this issue. Here’s the link to my blog. I hope it helps!

https://du-sungchan-24k.tistory.com/37

AnarchistKnight, 2024-10-31:
> > anyone kindly show the complete way to fix this issue, please?
> 
> It’s been a while, but I solved this issue. Here’s the link to my blog. I hope it helps!
> 
> https://du-sungchan-24k.tistory.com/37

The problem I encountered might be a bit different from yours. I am setting up ROS under WSL2, specific version info is `Linux version 5.15.153.1-microsoft-standard-WSL2 (root@941d701f84f1) (gcc (GCC) 11.2.0, GNU ld (GNU Binutils) 2.37) #1 SMP Fri Mar 29 23:14:13 UTC 2024`. 
Python version is 3.8.2. I would also like to paste pip list here. `Package                              Version
------------------------------------ -------
action-msgs                          2.0.2
action-tutorials-interfaces          0.33.5
action-tutorials-py                  0.33.5
actionlib-msgs                       5.3.5
ament-clang-format                   0.17.1
ament-clang-tidy                     0.17.1
ament-cmake-google-benchmark         2.5.2
ament-cmake-test                     2.5.2
ament-copyright                      0.17.1
ament-cppcheck                       0.17.1
ament-cpplint                        0.17.1
ament-flake8                         0.17.1
ament-index-python                   1.8.1
ament-lint                           0.17.1
ament-lint-cmake                     0.17.1
ament-mypy                           0.17.1
ament-package                        0.16.3
ament-pclint                         0.17.1
ament-pep257                         0.17.1
ament-pycodestyle                    0.17.1
ament-pyflakes                       0.17.1
ament-uncrustify                     0.17.1
ament-xmllint                        0.17.1
argcomplete                          3.5.1
builtin-interfaces                   2.0.2
composition-interfaces               2.0.2
demo-nodes-py                        0.33.5
diagnostic-msgs                      5.3.5
domain-coordinator                   0.12.0
example-interfaces                   0.12.0
examples-rclpy-executors             0.19.4
examples-rclpy-guard-conditions      0.19.4
examples-rclpy-minimal-action-client 0.19.4
examples-rclpy-minimal-action-server 0.19.4
examples-rclpy-minimal-client        0.19.4
examples-rclpy-minimal-publisher     0.19.4
examples-rclpy-minimal-service       0.19.4
examples-rclpy-minimal-subscriber    0.19.4
examples-rclpy-pointcloud-publisher  0.19.4
examples-tf2-py                      0.36.4
geometry-msgs                        5.3.5
interactive-markers                  2.5.4
laser-geometry                       2.7.0
launch                               3.4.2
launch-pytest                        3.4.2
launch-ros                           0.26.5
launch-testing                       3.4.2
launch-testing-examples              0.19.4
launch-testing-ros                   0.26.5
launch-xml                           3.4.2
launch-yaml                          3.4.2
lifecycle-msgs                       2.0.2
lifecycle-py                         0.33.5
logging-demo                         0.33.5
lttngpy                              8.2.2
map-msgs                             2.4.1
message-filters                      4.11.2
nav-msgs                             5.3.5
osrf-pycommon                        2.1.4
pendulum-msgs                        0.33.5
pip                                  24.3.1
[... 115 lines cut ...]
    module = import_module(match.group('module'))
  File "/usr/local/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/guozhengyang/ros2-linux/lib/python3.12/site-packages/ros2cli/cli.py", line 22, in <module>
    from rclpy.executors import ExternalShutdownException
  File "/home/guozhengyang/ros2-linux/lib/python3.12/site-packages/rclpy/__init__.py", line 49, in <module>
    from rclpy.signals import install_signal_handlers
  File "/home/guozhengyang/ros2-linux/lib/python3.12/site-packages/rclpy/signals.py", line 15, in <module>
    from rclpy.exceptions import InvalidHandle
  File "/home/guozhengyang/ros2-linux/lib/python3.12/site-packages/rclpy/exceptions.py", line 16, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "/home/guozhengyang/ros2-linux/lib/python3.12/site-packages/rclpy/impl/implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
  File "/home/guozhengyang/ros2-linux/lib/python3.12/site-packages/rpyutils/import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
  File "/usr/local/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'
The C extension '/home/guozhengyang/ros2-linux/lib/python3.12/site-packages/_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so' isn't present on the system. Please refer to 'https://docs.ros.org/en/jazzy/How-To-Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system' for possible solutions`

AnarchistKnight, 2024-10-31:
> You should check if the Python environment for ROS matches the current terminal's Python environment. I noticed your terminal is using Python 3.11, but ROS is looking for Python 3.12, which could be [...]

would you like to kindly show your ubuntu and python version?

lhx0217, 2024-11-26:
I have solved this problem, i copy the setup.sh in /opt/ros/humble to my project and add
import os
os.system("setup.sh")
it works!!
[image]

[image]

Harland666, 2025-07-05:
same issue