ros2/rclpy #1145: [node.py] Error in check parameter type

mikekondr, 2023-07-25:
Hi!

Seems to be, there is a bug:
https://github.com/ros2/rclpy/blob/cc8687852039cdd82d8ffb6cd544945cf3146a60/rclpy/rclpy/node.py#L1172

There must be 
`parameter.type_.value != descriptor.type.value`

Steps to reproduce:
1. create new simple node
2. declare new parameter with type STRING
3. try to set this parameter via command line, like 'ros2 param set /node param test' - and got error "Wrong parameter type,
expected 'Type. STRING' got 'Type. STRING'"

Please, forgive me, if I am wrong - I am newbee

clalancette, 2023-07-25:
I can't seem to reproduce this on the latest code on Rolling.  Can you give us more information about the platform and ROS 2 distribution you are using, and a full example when you see the problem?

mikekondr, 2023-07-25:
Ubuntu 22.04 Desktop ARM64
`Linux ubuntu-linux-22-04-desktop 5.15.0-76-generic #83-Ubuntu SMP Thu Jun 15 19:21:56 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux`

ROS2 Iron from repo:
`ros-iron-desktop-full 0.10.0-3.jammy.20230714.021808`
[http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-iron-desktop-full/](url)

Python 3.10.6

Code example:
[https://github.com/mikekondr/ros2_bugtest/](url)

Error example:
[image]

clalancette, 2023-07-25:
Thanks for putting the example together; that is really helpful.  With that, I can see the bug in your code.

The problem is in [this](https://github.com/mikekondr/ros2_bugtest/blob/eece44b841ee1265e832f8af226b732207ce0fd2/bugtest/bugtest/bugtest.py#L17) line:

```
                type = Parameter.Type.STRING
```

That should instead be:

```
            type = ParameterType.PARAMETER_STRING
```

That said, we should probably have some additional checking when setting a parameter that the type being passed there is what we expect.  I'll see what I can do about improving that error checking.

mikekondr, 2023-07-26:
@clalancette Thank you for your help!

Now I see my mistake

closed 2023-07-26 by mikekondr