ros2/rclpy #1147: lost subscription after repeated create/destroy cycles

dcconner, 2023-08-03:
## Bug report

I have a use case that requires temporary subscriptions.  I want to destroy_subscription when not in use, but this causes some crashes similar to what's reported in #1142.

I tried to create a simple demo in hopes of recreating the crash, but uncovered a different bug where the new subscription does not receive data.

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binary (Rolling) 
- Version or commit hash:
  - binary - ros-rolling-rclpy                                  5.0.0-1jammy.20230711.215529 
- DDS implementation:
  - Fast-RTPS (default)
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Start with basic [pub/sub tutorial](https://docs.ros.org/en/rolling/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html) 
Modify the publisher to publisher faster (I've reproduced with rate of 0.05 and 0.01 seconds)

Use this MinimalSubscriber code that turns subscription on and off

```python
# Copyright 2016 Open Source Robotics Foundation, Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

import rclpy
from rclpy.node import Node

from std_msgs.msg import String

class MinimalSubscriber(Node):

    def __init__(self):
        super().__init__('minimal_subscriber')
        self.subscription = self.create_subscription(
            String,
            'topic',
            self.listener_callback,
            10)
        self.subscription  # prevent unused variable warning

    def toggle_listening(self):
        if self.subscription is not None:
            self.destroy_subscription(self.subscription)
            self.subscription = None
        else:
            self.subscription = self.create_subscription(
                String,
                'topic',
                self.listener_callback,
                10)

    def listener_callback(self, msg):
        self.get_logger().info('I heard: "%s"' % msg.data)
[... 96 lines cut ...]
[INFO] [1691072616.659094660] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072617.168184955] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072617.674967913] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072618.184749013] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072618.694865075] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072619.205123416] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072619.713618794] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072620.224821481] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072620.735706665] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072621.244897479] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072621.755333433] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072622.266069073] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072622.774237412] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072623.285798480] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072623.797201730] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072624.309329566] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072624.820561969] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072625.331116627] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072625.837969617] [minimal_subscriber]:  toggle listener ...
[INFO] [1691072626.349654745] [minimal_subscriber]:  toggle listener ...
```

I have repeatedly observed this behavior after modifications and restarts, but have not tracked to a specific interval before the loss occurs.

#### Additional information

clalancette, 2023-08-10:
Out of curiousity: does this happen with CycloneDDS as well?

dcconner, 2023-08-10:
Just confirmed same issue with rmw-cyclonedds-cpp 

Until it failed, it seemed that cyclone was more predictable in the ordering of destroy/callback as sometimes we'd get a callback after requesting destroy with fast-rtps.   But in the end, both of them jus started toggling without invoking call back.

Using a 0.01 timing on publisher, and 0.5 second toggle, it dropped call backs after about 156 seconds with cyclone

dcconner, 2023-08-10:
ran again, and dropped after about 30 seconds with cyclone

dcconner, 2023-08-10:
ran again and dropped after 33 s

clalancette, 2023-08-10:
Hm, interesting.  I'm using the latest Rolling from sources (rclpy commit 540b809b1b4d6fc064cfd392834123f713593be4), with Fast-DDS.  I took your exact code above, and ran it in one terminal.  Then in a second terminal I ran:

```
ros2 topic pub -r10 /topic std_msgs/msg/String '{data: hello}'
```

And with that, I've been able to run this for up to 10 minutes at a time without problems.  A few questions:
1.  How exactly are you doing the publication, and at what rate?
2.  Can you try with the latest Rolling sources and see if you can still reproduce it?
3. What environment are you running this in?  A Docker container?  Host system?  What does the network look like?

dcconner, 2023-08-10:
1) I'm using the 'minimal publisher" example with rate changed to 100 hz.  

   I'll try with your `ros2 topic pub` and see what happens

2) Unfortunately, I don't have bandwidth to build and test with rolling binaries right now.

3) I'm running from rolling binaries on desktop Ubuntu host system.  Wired network port, but running all localhost

dcconner, 2023-08-10:
It happened with ros2 topic pub after 264 seconds

dcconner, 2023-08-10:
Tomorrow I'll try to reproduce on a different machine in our lab

clalancette, 2023-08-10:
Ah, there we go.  OK, with a -r100, I was eventually able to reproduce it here locally (eventually).

dcconner, 2023-08-10:
I also note this exception when I finally killed the process
(Note: this last test version moved the toggle listener INFO output into the toggle method to note which toggle was on vs off)

<pre>
INFO] [1691696994.784342628] [minimal_subscriber]:  toggle listener : renew subscription ...
[INFO] [1691696995.294053477] [minimal_subscriber]:  toggle listener : destroy existing subscription ...
[INFO] [1691696995.803947177] [minimal_subscriber]:  toggle listener : renew subscription ...
[INFO] [1691696996.314541920] [minimal_subscriber]:  toggle listener : destroy existing subscription ...
[INFO] [1691696996.824285524] [minimal_subscriber]:  toggle listener : renew subscription ...
[INFO] [1691696997.334216577] [minimal_subscriber]:  toggle listener : destroy existing subscription ...
[INFO] [1691696997.843316997] [minimal_subscriber]:  toggle listener : renew subscription ...
[INFO] [1691696998.353403879] [minimal_subscriber]:  toggle listener : destroy existing subscription ...
[INFO] [1691696998.860105559] [minimal_subscriber]:  toggle listener : renew subscription ...
[INFO] [1691696999.365411732] [minimal_subscriber]:  toggle listener : destroy existing subscription ...
[INFO] [1691696999.873281148] [minimal_subscriber]:  toggle listener : renew subscription ...
[INFO] [1691697000.381207783] [minimal_subscriber]:  toggle listener : destroy existing subscription ...
^CTraceback (most recent call last):
  File "/home/conner/test/debug_destroyable/install/py_pubsub/lib/py_pubsub/listener", line 33, in <module>
    sys.exit(load_entry_point('py-pubsub==0.0.0', 'console_scripts', 'listener')())
  File "/home/conner/test/debug_destroyable/install/py_pubsub/lib/python3.10/site-packages/py_pubsub/subscriber_member_function.py", line 67, in main
    executor.spin_once(timeout_sec=0.01)
  File "/opt/ros/rolling/lib/python3.10/site-packages/rclpy/executors.py", line 829, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/rolling/lib/python3.10/site-packages/rclpy/executors.py", line 808, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(
  File "/opt/ros/rolling/lib/python3.10/site-packages/rclpy/executors.py", line 724, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "/opt/ros/rolling/lib/python3.10/site-packages/rclpy/executors.py", line 621, in _wait_for_ready_callbacks
    wait_set.wait(timeout_nsec)
KeyboardInterrupt
The following exception was never retrieved: cannot use Destroyable because destruction was requested
[ros2run]: Interrupt

</pre>

clalancette, 2023-08-11:
All right, I think I see what is happening.  Essentially it is a race between the executor trying to execute the subscription, and the subscription being destroyed.  When you lose that race, then what happens is that you start executing the subscription but sometime halfway in-between the subscription is destroyed.  When that happens at just the right time, then the next thing we'll do is attempt to `__enter__` the subscription.  But because it has already been destroyed, it throws an exception saying that destruction has been already been requested.  For some reason that exception is being swallowed by the async machinery (which is a separate problem, but made it harder to debug).

The easiest solution here is to catch the exception that causes it to fail to start the context manager, and that should let us drop the subscription on the next round.  I have a prototype patch at #1150.  That seems to fix it in my testing, but if you could give it a whirl that would be helpful.

dcconner, 2023-08-11:
ran with your fix and Cyclone for over 30 minutes without issue.

Rebuilt with default RMW, and running that now.  

It's a separate issue but what I notice is that Cyclone seemed very predictable in the timing of output messages and consistent with expectations.  

The Fast implementation seems to be very "choppy" in the timing and number of messages received during a given interval.
I'll leave Fast running for a while to verify , and then plan to post example of this "choppiness".

dcconner, 2023-08-11:
ran without losing subscriptions for over 30 minutes on Fast, but "choppy"

<pre>
[INFO] [1691767102.591812995] [minimal_subscriber]: I heard: "Hello World: 225341"
[INFO] [1691767102.610655027] [minimal_subscriber]: I heard: "Hello World: 225342"
[INFO] [1691767102.612631335] [minimal_subscriber]: I heard: "Hello World: 225343"
[INFO] [1691767102.630193321] [minimal_subscriber]: I heard: "Hello World: 225344"
[INFO] [1691767102.632402719] [minimal_subscriber]: I heard: "Hello World: 225345"
[INFO] [1691767102.640839854] [minimal_subscriber]:  toggle listener : destroy existing subscription ...
[INFO] [1691767102.641856866] [minimal_subscriber]: I heard: "Hello World: 225346"
[INFO] [1691767103.152329727] [minimal_subscriber]:  toggle listener : renew subscription ...
[INFO] [1691767103.205494449] [minimal_subscriber]: I heard: "Hello World: 225398"
[INFO] [1691767103.207923730] [minimal_subscriber]: I heard: "Hello World: 225399"
[INFO] [1691767103.210004426] [minimal_subscriber]: I heard: "Hello World: 225400"
[INFO] [1691767103.214089069] [minimal_subscriber]: I heard: "Hello World: 225401"
[INFO] [1691767103.216397478] [minimal_subscriber]: I heard: "Hello World: 225402"
[INFO] [1691767103.218405546] [minimal_subscriber]: I heard: "Hello World: 225403"
[INFO] [1691767103.220939442] [minimal_subscriber]: I heard: "Hello World: 225404"
[INFO] [1691767103.244863957] [minimal_subscriber]: I heard: "Hello World: 225405"
[INFO] [1691767103.246982868] [minimal_subscriber]: I heard: "Hello World: 225406"
[INFO] [1691767103.251187143] [minimal_subscriber]: I heard: "Hello World: 225407"
[INFO] [1691767103.264155425] [minimal_subscriber]: I heard: "Hello World: 225408"
[INFO] [1691767103.271562157] [minimal_subscriber]: I heard: "Hello World: 225409"
[INFO] [1691767103.281507014] [minimal_subscriber]: I heard: "Hello World: 225410"
[INFO] [1691767103.290975965] [minimal_subscriber]: I heard: "Hello World: 225411"
[INFO] [1691767103.301744607] [minimal_subscriber]: I heard: "Hello World: 225412"
[INFO] [1691767103.311234187] [minimal_subscriber]: I heard: "Hello World: 225413"
[INFO] [1691767103.321359715] [minimal_subscriber]: I heard: "Hello World: 225414"
[INFO] [1691767103.332827787] [minimal_subscriber]: I heard: "Hello World: 225415"
[INFO] [1691767103.341451379] [minimal_subscriber]: I heard: "Hello World: 225416"
[INFO] [1691767103.351292818] [minimal_subscriber]: I heard: "Hello World: 225417"
[INFO] [1691767103.361112922] [minimal_subscriber]: I heard: "Hello World: 225418"
[INFO] [1691767103.371116394] [minimal_subscriber]: I heard: "Hello World: 225419"
[INFO] [1691767103.383277082] [minimal_subscriber]: I heard: "Hello World: 225420"
[INFO] [1691767103.390948888] [minimal_subscriber]: I heard: "Hello World: 225421"
[INFO] [1691767103.401278160] [minimal_subscriber]: I heard: "Hello World: 225422"
[INFO] [1691767103.413802282] [minimal_subscriber]: I heard: "Hello World: 225423"
[INFO] [1691767103.427571778] [minimal_subscriber]: I heard: "Hello World: 225424"
[INFO] [1691767103.431922503] [minimal_subscriber]: I heard: "Hello World: 225425"
[INFO] [1691767103.441314595] [minimal_subscriber]: I heard: "Hello World: 225426"
[INFO] [1691767103.451798659] [minimal_subscriber]: I heard: "Hello World: 225427"
[INFO] [1691767103.462654036] [minimal_subscriber]: I heard: "Hello World: 225428"
[INFO] [1691767103.471556124] [minimal_subscriber]: I heard: "Hello World: 225429"
[INFO] [1691767103.482306454] [minimal_subscriber]: I heard: "Hello World: 225430"
[INFO] [1691767103.491220392] [minimal_subscriber]: I heard: "Hello World: 225431"
[INFO] [1691767103.501008382] [minimal_subscriber]: I heard: "Hello World: 225432"
[INFO] [1691767103.520714901] [minimal_subscriber]: I heard: "Hello World: 225433"
[INFO] [1691767103.523061310] [minimal_subscriber]: I heard: "Hello World: 225434"
[INFO] [1691767103.531049689] [minimal_subscriber]: I heard: "Hello World: 225435"
[INFO] [1691767103.541029157] [minimal_subscriber]: I heard: "Hello World: 225436"
[INFO] [1691767103.551071663] [minimal_subscriber]: I heard: "Hello World: 225437"
[INFO] [1691767103.561618128] [minimal_subscriber]: I heard: "Hello World: 225438"
[INFO] [1691767103.572102502] [minimal_subscriber]: I heard: "Hello World: 225439"
[INFO] [1691767103.585200465] [minimal_subscriber]: I heard: "Hello World: 225440"
[INFO] [1691767103.596560177] [minimal_subscriber]: I heard: "Hello World: 225441"
[INFO] [1691767103.602485474] [minimal_subscriber]: I heard: "Hello World: 225442"
[INFO] [1691767103.614550529] [minimal_subscriber]: I heard: "Hello World: 225443"
[INFO] [1691767103.647714411] [minimal_subscriber]: I heard: "Hello World: 225444"
[INFO] [1691767103.650076797] [minimal_subscriber]: I heard: "Hello World: 225445"
[INFO] [1691767103.652362909] [minimal_subscriber]: I heard: "Hello World: 225446"
[INFO] [1691767103.654355359] [minimal_subscriber]: I heard: "Hello World: 225447"
[INFO] [1691767103.660609209] [minimal_subscriber]:  toggle listener : destroy existing subscription ...
[INFO] [1691767103.661534850] [minimal_subscriber]: I heard: "Hello World: 225448"
[INFO] [1691767104.171407231] [minimal_subscriber]:  toggle listener : renew subscription ...
[INFO] [1691767104.198977663] [minimal_subscriber]: I heard: "Hello World: 225500"
[INFO] [1691767104.202179972] [minimal_subscriber]: I heard: "Hello World: 225501"
[INFO] [1691767104.204400151] [minimal_subscriber]: I heard: "Hello World: 225502"
[... cut ...]
[INFO] [1691767119.517588949] [minimal_subscriber]: I heard: "Hello World: 227033"
[INFO] [1691767119.521607065] [minimal_subscriber]: I heard: "Hello World: 227034"
[INFO] [1691767119.532779285] [minimal_subscriber]: I heard: "Hello World: 227035"
[INFO] [1691767119.545171902] [minimal_subscriber]: I heard: "Hello World: 227036"
[INFO] [1691767119.551627608] [minimal_subscriber]: I heard: "Hello World: 227037"
[INFO] [1691767119.561425756] [minimal_subscriber]: I heard: "Hello World: 227038"

</pre>

clalancette, 2023-08-11:
> ran without losing subscriptions for over 30 minutes on Fast, but "choppy"

That's in all likelihood a different problem, unrelated to the `rclpy` thing you found here.  If you'd like to pursue that, please consider opening a separate issue on https://github.com/ros2/rmw_fastrtps

PR #1150 (2023-08-28): Fix an inherent race in execution vs. destruction.
The rclpy executor collects all of the entities in one pass, then creates async tasks for each of the ready ones and attempts to "take" and execute them.  If one of those entities is destroyed after the collection but before we attempt to "take" it, then we can end up attempting to __enter__ a Destroyable-derived class that has already been destroyed. The Destroyable will then raise an InvalidHandle error.

Fix this by explicitly catching the InvalidHandle error that can be raised in all of the Destroyable-derived entities. If we do catch it,  [... cut]
closed 2023-08-28 by clalancette