ros2/rclpy #1149: Node crashes when importing Gstreamer python bindings

s-hall, 2023-08-04:
## Bug report

Importing Gstreamer python bindings seems to make rclpy segfault or abort when trying to create a node.

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - `ros-humble-ros-core/now 0.10.0-1jammy.20230623.212334 amd64`
- DDS implementation:
  - default
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Minimal example:
```
import gi
import rclpy

rclpy.init()
print("init")

# tmp_node = rclpy.create_node("fixed")
# tmp_node.destroy_node()
# del tmp_node

gi.require_version("Gst", "1.0")
from gi.repository import Gst

print("before node")
node = rclpy.create_node("gstreamer_crash")
print("after node")
rclpy.spin(node)
```
Running a container from a directory with the above script, `gst_rclpy.py`:
`docker run --rm -it -v .:/test ros:humble bash`
Inside:
```
apt update && apt install -y python3-gi python3-gst-1.0
python3 /test/gst_rclpy.py
```
Output:
```
init
before node
Segmentation fault (core dumped)
```

#### Expected behavior
No crash.
#### Actual behavior
Quickly tested on different distributions; it works on `ros:foxy` and `ros:galactic`, and crashes on `ros:humble` and onwards. Also, I happened to notice if a node is created before importing Gstreamer, it seems to avoid the crash (though this didn't seem to be a reliable fix in our actual application).

fujitatomoya, 2023-08-07:
@iuhilnehc-ynos @Barry-Xu-2018 can you check this if you have bandwidth? it is not  unlikely that application uses gstreamer plugin with ROS 2.

iuhilnehc-ynos, 2023-08-07:
I can reproduce this issue followed by https://github.com/ros2/rclpy/issues/1149#issue-1837424580, but this issue can't happen on `rmw_cyclonedds_cpp`.

After installing the debug symbols packages, got the backtrace at the point of `Segmentation fault` by running the `gst_rclpy.py` below,

```cpp
(gdb) bt
#0  0x00007ffff6702580 in ?? () from /lib/x86_64-linux-gnu/libgcc_s.so.1
#1  0x00007ffff670273f in __gcc_personality_v0 () from /lib/x86_64-linux-gnu/libgcc_s.so.1
#2  0x00007ffff4d0bfe9 in __libunwind_Unwind_Resume () from /lib/x86_64-linux-gnu/libunwind.so.8
#3  0x00007ffff56d6f8a in __gnu_cxx::new_allocator<char>::deallocate (__t=<optimized out>, __p=<optimized out>, this=<optimized out>, 
    this=<optimized out>, __p=<optimized out>, __t=<optimized out>) at /usr/include/c++/11/ext/new_allocator.h:145
#4  std::allocator_traits<std::allocator<char> >::deallocate (__n=<optimized out>, __p=<optimized out>, __a=..., __a=..., __p=<optimized out>, 
    __n=<optimized out>) at /usr/include/c++/11/bits/alloc_traits.h:496
#5  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_destroy (__size=<optimized out>, this=<optimized out>, 
    this=<optimized out>, __size=<optimized out>) at /usr/include/c++/11/bits/basic_string.h:245
#6  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_dispose (this=<optimized out>, this=<optimized out>)
    at /usr/include/c++/11/bits/basic_string.h:240
#7  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string (this=<optimized out>, this=<optimized out>)
    at /usr/include/c++/11/bits/basic_string.h:672
#8  boost::interprocess::shared_memory_object::shared_memory_object (this=0x7fffffff4cb0, name=<optimized out>, mode=<optimized out>)
    at ./thirdparty/boost/include/boost/interprocess/shared_memory_object.hpp:83
#9  0x00007ffff5b3e40d in boost::interprocess::ipcdetail::managed_open_or_create_impl<boost::interprocess::shared_memory_object, 16ul, true, false>::priv_open_or_create<boost::interprocess::ipcdetail::create_open_func<boost::interprocess::ipcdetail::basic_managed_memory_impl<char, boost::interprocess::rbtree_best_fit<boost::interprocess::mutex_family, boost::interprocess::offset_ptr<void, unsigned int, unsigned long, 0ul>, 0ul>, boost::interprocess::iset_index, 16ul> > >(boost::interprocess::ipcdetail::create_enum_t, char const* const&, unsigned long, boost::interprocess::mode_t, void const*, boost::interprocess::permissions const&, boost::interprocess::ipcdetail::create_open_func<boost::interprocess::ipcdetail::basic_managed_memory_impl<char, boost::interprocess::rbtree_best_fit<boost::interprocess::mutex_family, boost::interprocess::offset_ptr<void, unsigned int, unsigned long, 0ul>, 0ul>, boost::interprocess::iset_index, 16ul> >) [clone .constprop.0] (this=this@entry=0x555555f57408, type=type@entry=boost::interprocess::ipcdetail::DoOpen, 
    id=@0x7fffffff4dc0: 0x555555f57430 "fastrtps_port14412", size=size@entry=0, mode=mode@entry=boost::interprocess::read_write, perm=..., 
    construct_func=..., addr=0x0) at ./thirdparty/boost/include/boost/interprocess/detail/managed_open_or_create_impl.hpp:339
#10 0x00007ffff5b36a75 in boost::interprocess::ipcdetail::managed_open_or_create_impl<boost::interprocess::shared_memory_object, 16ul, true, false>::managed_open_or_create_impl<boost::interprocess::ipcdetail::create_open_func<boost::interprocess::ipcdetail::basic_managed_memory_impl<char, boost::interprocess::rbtree_best_fit<boost::interprocess::mutex_family, boost::interprocess::offset_ptr<void, unsigned int, unsigned long, 0ul>, 0ul>, boost::interprocess::iset_index, 16ul> > > (construct_func=..., construct_func=..., addr=0x0, mode=boost::interprocess::read_write, 
    id=@0x7fffffff4dc0: 0x555555f57430 "fastrtps_port14412", this=0x555555f57408)
    at ./thirdparty/boost/include/boost/interprocess/detail/managed_open_or_create_impl.hpp:202
#11 boost::interprocess::basic_managed_shared_memory<char, boost::interprocess::rbtree_best_fit<boost::interprocess::mutex_family, boost::interprocess::offset_ptr<void, unsigned int, unsigned long, 0ul>, 0ul>, boost::interprocess::iset_index>::basic_managed_shared_memory (addr=0x0, name=<optimized out>, 
    this=0x555555f57400) at ./thirdparty/boost/include/boost/interprocess/managed_shared_memory.hpp:151
#12 eprosima::fastdds::rtps::SharedSegment<boost::interprocess::basic_managed_shared_memory<char, boost::interprocess::rbtree_best_fit<boost::interprocess::mutex_family, boost::interprocess::offset_ptr<void, unsigned int, unsigned long, 0ul>, 0ul>, boost::interprocess::iset_index>, boost::interprocess::shared_memory_object>::SharedSegment(boost::interprocess::open_only_t, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) [clone .constprop.0] (this=0x555555e3c400, name=...) at ./src/cpp/utils/shared_memory/SharedMemSegment.hpp:272
--Type <RET> for more, q to quit, c to continue without paging--c
#13 0x00007ffff59eeb77 in eprosima::fastdds::rtps::SharedMemGlobal::open_port_internal (this=0x555555e89950, port_id=14412, max_buffer_descriptors=512, healthy_check_timeout_ms=1000, open_mode=eprosima::fastdds::rtps::SharedMemGlobal::Port::OpenMode::ReadExclusive, regenerating_port=std::shared_ptr<eprosima::fastdds::rtps::SharedMemGlobal::Port> (empty) = {...}) at ./src/cpp/rtps/transport/shared_mem/SharedMemGlobal.hpp:1018
[... cut ...]
#59 0x00005555557aa0ae in Py_RunMain ()
#60 0x000055555578034d in Py_BytesMain ()
#61 0x00007ffff7c82d90 in __libc_start_call_main (main=main@entry=0x555555780310, argc=argc@entry=2, argv=argv@entry=0x7fffffffe3d8) at ../sysdeps/nptl/libc_start_call_main.h:58
#62 0x00007ffff7c82e40 in __libc_start_main_impl (main=0x555555780310, argc=2, argv=0x7fffffffe3d8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffe3c8) at ../csu/libc-start.c:392
#63 0x0000555555780245 in _start ()
```

iuhilnehc-ynos, 2023-08-07:
```cpp
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /usr/bin/python3 gst_rclpy.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff4fde640 (LWP 3505394)]
[New Thread 0x7ffff25ff640 (LWP 3506068)]
init
before node
[New Thread 0x7ffff0e70640 (LWP 3506305)]
[New Thread 0x7ffff05e8640 (LWP 3506306)]
[New Thread 0x7fffefde7640 (LWP 3506307)]
[New Thread 0x7fffef5e6640 (LWP 3506308)]
[New Thread 0x7fffeede5640 (LWP 3506309)]

Thread 1 "python3" hit Breakpoint 2, boost::interprocess::shared_memory_object::shared_memory_object (this=0x7fffffff1720, name=0x55555600e410 "fastrtps_port14411", mode=boost::interprocess::read_write) at /home/chenlh/Projects/ROS2/ros2-master/src/eProsima/Fast-DDS/thirdparty/boost/include/boost/interprocess/shared_memory_object.hpp:83
83	   {  this->priv_open_or_create(ipcdetail::DoOpen, name, mode, permissions());  }
(gdb) bt
#0  boost::interprocess::shared_memory_object::shared_memory_object(boost::interprocess::open_only_t, char const*, boost::interprocess::mode_t) (this=0x7fffffff1720, name=0x55555600e410 "fastrtps_port14411", mode=boost::interprocess::read_write) at /home/chenlh/Projects/ROS2/ros2-master/src/eProsima/Fast-DDS/thirdparty/boost/include/boost/interprocess/shared_memory_object.hpp:83
#1  0x00007ffff37f1d52 in boost::interprocess::ipcdetail::managed_open_or_create_impl<boost::interprocess::shared_memory_object, 16ul, true, false>::priv_open_or_create<boost::interprocess::ipcdetail::create_open_func<boost::interprocess::ipcdetail::basic_managed_memory_impl<char, boost::interprocess::rbtree_best_fit<boost::interprocess::mutex_family, boost::interprocess::offset_ptr<void, unsigned int, unsigned long, 0ul>, 0ul>, boost::interprocess::iset_index, 16ul> > >(boost::interprocess::ipcdetail::create_enum_t, char const* const&, unsigned long, boost::interprocess::mode_t, void const*, boost::interprocess::permissions const&, boost::interprocess::ipcdetail::create_open_func<boost::interprocess::ipcdetail::basic_managed_memory_impl<char, boost::interprocess::rbtree_best_fit<boost::interprocess::mutex_family, boost::interprocess::offset_ptr<void, unsigned int, unsigned long, 0ul>, 0ul>, boost::interprocess::iset_index, 16ul> >) (this=0x55555600e3e8, type=boost::interprocess::ipcdetail::DoOpen, id=@0x7fffffff1800: 0x55555600e410 "fastrtps_port14411", size=0, mode=boost::interprocess::read_write, addr=0x0, perm=..., construct_func=...) at /home/chenlh/Projects/ROS2/ros2-master/src/eProsima/Fast-DDS/thirdparty/boost/include/boost/interprocess/detail/managed_open_or_create_impl.hpp:339
#2  0x00007ffff37f0994 in boost::interprocess::ipcdetail::managed_open_or_create_impl<boost::interprocess::shared_memory_object, 16ul, true, false>::managed_open_or_create_impl<boost::interprocess::ipcdetail::create_open_func<boost::interprocess::ipcdetail::basic_managed_memory_impl<char, boost::interprocess::rbtree_best_fit<boost::interprocess::mutex_family, boost::interprocess::offset_ptr<void, unsigned int, unsigned long, 0ul>, 0ul>, boost::interprocess::iset_index, 16ul> > >(boost::interprocess::open_only_t, char const* const&, boost::interprocess::mode_t, void const*, boost::interprocess::ipcdetail::create_open_func<boost::interprocess::ipcdetail::basic_managed_memory_impl<char, boost::interprocess::rbtree_best_fit<boost::interprocess::mutex_family, boost::interprocess::offset_ptr<void, unsigned int, unsigned long, 0ul>, 0ul>, boost::interprocess::iset_index, 16ul> > const&) (this=0x55555600e3e8, id=@0x7fffffff1800: 0x55555600e410 "fastrtps_port14411", mode=boost::interprocess::read_write, addr=0x0, construct_func=...) at /home/chenlh/Projects/ROS2/ros2-master/src/eProsima/Fast-DDS/thirdparty/boost/include/boost/interprocess/detail/managed_open_or_create_impl.hpp:202
#3  0x00007ffff38442bd in boost::interprocess::basic_managed_shared_memory<char, boost::interprocess::rbtree_best_fit<boost::interprocess::mutex_family, boost::interprocess::offset_ptr<void, unsigned int, unsigned long, 0ul>, 0ul>, boost::interprocess::iset_index>::basic_managed_shared_memory(boost::interprocess::open_only_t, char const*, void const*) (this=0x55555600e3e0, name=0x55555600e410 "fastrtps_port14411", addr=0x0) at /home/chenlh/Projects/ROS2/ros2-master/src/eProsima/Fast-DDS/thirdparty/boost/include/boost/interprocess/managed_shared_memory.hpp:151
#4  0x00007ffff3843ec2 in eprosima::fastdds::rtps::SharedSegment<boost::interprocess::basic_managed_shared_memory<char, boost::interprocess::rbtree_best_fit<boost::interprocess::mutex_family, boost::interprocess::offset_ptr<void, unsigned int, unsigned long, 0ul>, 0ul>, boost::interprocess::iset_index>, boost::interprocess::shared_memory_object>::SharedSegment(boost::interprocess::open_only_t, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (this=0x555555fddff0, name="fastrtps_port14411") at /home/chenlh/Projects/ROS2/ros2-master/src/eProsima/Fast-DDS/src/cpp/utils/shared_memory/SharedMemSegment.hpp:341
#5  0x00007ffff3dc0193 in eprosima::fastdds::rtps::SharedMemGlobal::open_port_internal(unsigned int, unsigned int, unsigned int, eprosima::fastdds::rtps::SharedMemGlobal::Port::OpenMode, std::shared_ptr<eprosima::fastdds::rtps::SharedMemGlobal::Port>) (this=0x555555c2f130, port_id=14411, max_buffer_descriptors=512, healthy_check_timeout_ms=1000, open_mode=eprosima::fastdds::rtp [... cut ...]
Run till exit from #0  boost::interprocess::shared_memory_object::shared_memory_object (this=0x7fffffff1720, name=0x55555600e410 "fastrtps_port14411", mode=boost::interprocess::read_write) at /home/chenlh/Projects/ROS2/ros2-master/src/eProsima/Fast-DDS/thirdparty/boost/include/boost/interprocess/shared_memory_object.hpp:83
terminate called after throwing an instance of 'boost::interprocess::interprocess_exception'
  what():  No such file or directory

Thread 1 "python3" received signal SIGABRT, Aborted.
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140737352327616) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.

```

Very intesting, right?

iuhilnehc-ynos, 2023-08-07:
It seems that we need to know what the following line does as it works well after commenting on it.

```
from gi.repository import Gst
```

iuhilnehc-ynos, 2023-08-08:
I am afraid that the root issue might be related to either `python3.10` or `python3-gi python3-gst-1.0`.

I have confirmed that the `gst_rclpy.py` run well with `RMW_IMPLEMENTATION=rmw_fastrtps_cpp` on `ROS 2 source galactic` which is built on `ubuntu20(python3.8)`, but it failed to run on the `ROS 2 source galactic`, which is the same source code, built on `ubuntu22(python3.10)`.

iuhilnehc-ynos, 2023-08-08:
And `gst_rclpy.py` can run well with `ROS 2 source humble` built on ubuntu20(python3.8).

clalancette, 2023-08-08:
One idea I have is that maybe the ABI between the vendored copy of boost in fastrtps, and boost from the system, are colliding.  That only makes sense if gstreamer depends on boost, which I don't know if that is the case.

@MiguelCompany Is it possible to have Fast-DDS use the system copy of boost, rather than the vendored copy?  If so, how do you do that?

fujitatomoya, 2023-08-08:
according to pmap for shared libraries, boost is not linked in the core.

```bash
(gdb) info sharedlibrary
From                To                  Syms Read   Shared Object Library
0x00007fb6c36793a0  0x00007fb6c36f48e8  No          /lib/x86_64-linux-gnu/libm.so.6
0x00007fb6c363c290  0x00007fb6c365a5a7  No          /lib/x86_64-linux-gnu/libexpat.so.1
0x00007fb6c361e280  0x00007fb6c362ec14  No          /lib/x86_64-linux-gnu/libz.so.1
0x00007fb6c341c700  0x00007fb6c35aeabd  No          /lib/x86_64-linux-gnu/libc.so.6
0x00007fb6c3773090  0x00007fb6c379c335  No          /lib64/ld-linux-x86-64.so.2
0x00007fb6c37555a0  0x00007fb6c375720b  No          /usr/lib/python3.10/lib-dynload/_bz2.cpython-310-x86_64-linux-gnu.so
0x00007fb6c2c10280  0x00007fb6c2c1c563  No          /lib/x86_64-linux-gnu/libbz2.so.1.0
0x00007fb6c2c29920  0x00007fb6c2c2c891  No          /usr/lib/python3.10/lib-dynload/_lzma.cpython-310-x86_64-linux-gnu.so
0x00007fb6c2bd53c0  0x00007fb6c2bef0de  No          /lib/x86_64-linux-gnu/liblzma.so.5
0x00007fb6c2946300  0x00007fb6c296f43b  No          /usr/lib/python3/dist-packages/gi/_gi.cpython-310-x86_64-linux-gnu.so
0x00007fb6c2817b20  0x00007fb6c28a4f22  No          /lib/x86_64-linux-gnu/libglib-2.0.so.0
0x00007fb6c27a8fd0  0x00007fb6c27dac5e  No          /lib/x86_64-linux-gnu/libgobject-2.0.so.0
0x00007fb6c277e350  0x00007fb6c278deba  No          /lib/x86_64-linux-gnu/libgirepository-1.0.so.1
0x00007fb6c2e3b460  0x00007fb6c2e41392  No          /lib/x86_64-linux-gnu/libffi.so.8
0x00007fb6c27022a0  0x00007fb6c27565b0  No          /lib/x86_64-linux-gnu/libpcre.so.3
0x00007fb6c2e344a0  0x00007fb6c2e35433  No          /lib/x86_64-linux-gnu/libgmodule-2.0.so.0
0x00007fb6c25663e0  0x00007fb6c266f0cd  No          /lib/x86_64-linux-gnu/libgio-2.0.so.0
0x00007fb6c24ec290  0x00007fb6c2516dc1  No          /lib/x86_64-linux-gnu/libmount.so.1
0x00007fb6c24bdfc0  0x00007fb6c24d667d  No          /lib/x86_64-linux-gnu/libselinux.so.1
0x00007fb6c2487d10  0x00007fb6c24a6831  No          /lib/x86_64-linux-gnu/libblkid.so.1
0x00007fb6c23eb2e0  0x00007fb6c2455adf  No          /lib/x86_64-linux-gnu/libpcre2-8.so.0
0x00007fb6c2e53140  0x00007fb6c2e5342d  No          /usr/lib/python3.10/lib-dynload/_opcode.cpython-310-x86_64-linux-gnu.so
0x00007fb6c22b2020  0x00007fb6c22de23c  No          /usr/lib/python3/dist-packages/yaml/_yaml.cpython-310-x86_64-linux-gnu.so
0x00007fb6c228c240  0x00007fb6c22a4a0d  No          /lib/x86_64-linux-gnu/libyaml-0.so.2
0x00007fb6c2029f80  0x00007fb6c21bbdee  No          /root/ros2_ws/colcon_ws/install/rclpy/lib/python3.10/site-packages/rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so
0x00007fb6c1ffcc20  0x00007fb6c2004aa8  No          /root/ros2_ws/colcon_ws/install/rcl_action/lib/librcl_action.so
0x00007fb6c2c045c0  0x00007fb6c2c08f4c  No          /root/ros2_ws/colcon_ws/install/rcl_lifecycle/lib/librcl_lifecycle.so
0x00007fb6c1a80c40  0x00007fb6c1d2f83e  No          /lib/x86_64-linux-gnu/libpython3.10.so.1.0
0x00007fb6c19d1d00  0x00007fb6c19f8d22  No          /root/ros2_ws/colcon_ws/install/rcl/lib/librcl.so
0x00007fb6c2e4e120  0x00007fb6c2e4e5a9  No          /root/ros2_ws/colcon_ws/install/rcl_logging_interface/lib/librcl_logging_interface.so
0x00007fb6c19b7820  0x00007fb6c19be8a1  No          /root/ros2_ws/colcon_ws/install/rcl_yaml_param_parser/lib/librcl_yaml_param_parser.so
0x00007fb6c19ab3e0  0x00007fb6c19afc2c  No          /root/ros2_ws/colcon_ws/install/rmw_implementation/lib/librmw_implementation.so
0x00007fb6c199e600  0x00007fb6c19a2f19  No          /root/ros2_ws/colcon_ws/install/rmw/lib/librmw.so
0x00007fb6c1995040  0x00007fb6c1995569  No          /root/ros2_ws/colcon_ws/install/lifecycle_msgs/lib/liblifecycle_msgs__ro--Type <RET> for more, q to quit, c to continue without paging--
sidl_typesupport_c.so
0x00007fb6c1972f60  0x00007fb6c1981d40  No          /root/ros2_ws/colcon_ws/install/rcutils/lib/librcutils.so
0x00007fb6c17e3420  0x00007fb6c18ebfc2  No          /lib/x86_64-linux-gnu/libstdc++.so.6
0x00007fb6c1724660  0x00007fb6c173a805  No          /lib/x86_64-linux-gnu/libgcc_s.so.1
0x00007fb6c1705080  0x00007fb6c17140f7  No          /root/ros2_ws/colcon_ws/install/rosidl_runtime_c/lib/librosidl_runtime_c.so
0x00007fb6c16dc260  0x00007fb6c16f1cef  No          /root/ros2_ws/colcon_ws/install/tracetools/lib/libtracetools.so
0x00007fb6c16b45e0  0x00007fb6c16c229c  No          /root/ros2_ws/colcon_ws/install/lifecycle_msgs/lib/liblifecycle_msgs__rosidl_generator_c.so
0x00007fb6c169db50  0x00007fb6c16a2a29  No          /root/ros2_ws/colcon_ws/install/rcl_logging_spdlog/lib/librcl_logging_spdlog.so
0x00007fb6c1682040  0x00007fb6c1682ba0  No          /root/ros2_ws/colcon_ws/install/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_c.so
0x00007fb6c1661ce0  0x00007fb6c1670571  No          /root/ros2_ws/colcon_ws/install/rosidl_dynamic_typesupport/lib/librosidl_dynamic_typesupport.so
0x00007fb6c1655040  0x00007fb6c16553dc  No          /root/ros2_ws/colcon_ws/install/type_description_interfaces/lib/libtype_description_interfaces__rosidl_typesupport_c.so
0x00007fb6c163ee00  0x00007fb6c1646f85  No          /root/ros2_ws/colcon_ws/install/type_description_interfaces/lib/libtype_description_interfaces__rosidl_generator_c.so
0x00007fb6c162ddb0  0x00007fb6c1631fff  No          /root/ros2_ws/colcon_ws/install/ament_index_cpp/lib/libament_index_cpp.so
0x00007fb6c1613a80  0x00007fb6c161afb7  No          /root/ros2_ws/colcon_ws/install/rcpputils/lib/librcpputils.so
[... cut ...]
0x00007fb6b93a3080  0x00007fb6b93a35b5  No          /lib/x86_64-linux-gnu/libbrotlicommon.so.1
0x00007fb6b939d260  0x00007fb6b939e404  No          /lib/x86_64-linux-gnu/libkeyutils.so.1
0x00007fb6b938a6a0  0x00007fb6b9393229  No          /lib/x86_64-linux-gnu/libresolv.so.2
0x00007fb6c0026b40  0x00007fb6c0029157  No          /usr/lib/python3/dist-packages/gi/overrides/_gi_gst.cpython-310-x86_64-linux-gnu.so
0x00007fb6b9182440  0x00007fb6b91831db  No          /usr/lib/python3.10/lib-dynload/_queue.cpython-310-x86_64-linux-gnu.so
```

mjcarroll, 2023-08-08:
> That only makes sense if gstreamer depends on boost, which I don't know if that is the case.

I think that it's making use of pygobject, which doesn't seem to have any boost in the build: https://gitlab.gnome.org/GNOME/pygobject

iuhilnehc-ynos, 2023-08-09:
Thanks for your reply. These help me a lot.

I tried to find out the difference of `gst_rclpy.py` run on `ROS 2 source galactic` between `ubunt20` and `ubuntu22` by debugging the path (`layout asm` in gdb step by step -> libstdc++.so.6.0.[28|30] -> libgcc_s.so.1 -> libunwind.so.8.0.1(not in ubuntu20) ).

After building the `libunwind-1.3.2`, which is needed on ubuntu22, the `gst_rclpy.py` works well with the build version of `libunwind.so.8.0.1`. (The `gst_rclpy.py` also works well on `ROS 2 rolling` with the `libunwind-1.3.2` on ubuntu22.)

Unfortunately, I have no theory to convince myself that this is a solution, it might only adjust the code mapping to update timing access for certain functions.

@jwakely, could you share some information with me? I would greatly appreciate it.

iuhilnehc-ynos, 2023-08-10:
It seems it's the problem of mixed-using the `_Unwind_RaiseException and _Unwind_Resume` between gcc and libunwind.

```cpp
Thread 1 "python3" hit Breakpoint 1, 0x00007ffff5f65c38 in _Unwind_RaiseException () from /lib/x86_64-linux-gnu/libgcc_s.so.1
(gdb) c
Continuing.

Thread 1 "python3" hit Breakpoint 2, _Unwind_Resume (exception_object=0x555555fdf8c0) at unwind/Resume.c:30
30	{
(gdb) c
Continuing.

```

<details><summary> detailed backtrace </summary>
<p>

```shell
chenlh ros2-master $ gdb /usr/bin/python3
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/python3...
Reading symbols from /usr/lib/debug/.build-id/8d/f1c7df42f40a5fa606666db1b65c8d9a4ddfe4.debug...
(gdb) b _Unwind_RaiseException
Function "_Unwind_RaiseException" not defined.
Breakpoint 1 (_Unwind_RaiseException) pending.
(gdb) b _Unwind_Resume
Function "_Unwind_Resume" not defined.
Breakpoint 2 (_Unwind_Resume) pending.
(gdb) r gst_rclpy.py 
Starting program: /usr/bin/python3 gst_rclpy.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
warning: File "/usr/lib/x86_64-linux-gnu/debug/libstdc++.so.6.0.30-gdb.py" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
To enable execution of this file add
	add-auto-load-safe-path /usr/lib/x86_64-linux-gnu/debug/libstdc++.so.6.0.30-gdb.py
line to your configuration file "/home/chenlh/.gdbinit".
To completely disable this security protection add
	set auto-load safe-path /
line to your configuration file "/home/chenlh/.gdbinit".
For more information about this security protection see the
"Auto-loading safe path" section in the GDB manual.  E.g., run from the shell:
	info "(gdb)Auto-loading safe path"
[New Thread 0x7ffff3b66640 (LWP 778809)]
[New Thread 0x7ffff15ff640 (LWP 779085)]
init
[New Thread 0x7ffff0cfe640 (LWP 779130)]
[New Thread 0x7ffff04fd640 (LWP 779131)]
[New Thread 0x7fffefcfc640 (LWP 779132)]
[New Thread 0x7fffef4fb640 (LWP 779133)]
[New Thread 0x7fffeecfa640 (LWP 779134)]

Thread 1 "python3" hit Breakpoint 1, 0x00007ffff5f65c38 in _Unwind_RaiseException () from /lib/x86_64-linux-gnu/libgcc_s.so.1
(gdb) bt
#0  0x00007ffff5f65c38 in _Unwind_RaiseException () at /lib/x86_64-linux-gnu/libgcc_s.so.1
#1  0x00007ffff5cc69bb in __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) (obj=<optimized out>, tinfo=0x7ffff2d62628 <typeinfo for boost::interprocess::interprocess_exception>, dest=0x7ffff21c5d06 <boost::interprocess::interprocess_exception::~interprocess_exception()>) at ../../../../src/libstdc++-v3/libsupc++/eh_throw.cc:93
#2  0x00007ffff21e4e83 in boost::interprocess::shared_memory_object::priv_open_or_create(boost::interprocess::ipcdetail::create_enum_t, char const*, boost::interprocess::mode_t, boost::interprocess::permissions const&) (this=0x7fffffff16a0, type=boost::interprocess::ipcdetail::DoOpen, filename=0x55555600ffa0 "fastrtps_port14411", mode=boost::interprocess::read_write, perm=...) at /home/chenlh/Projects/ROS2/ros2-master/src/eProsima/Fast-DDS/thirdparty/boost/include/boost/interprocess/shared_memory_object.hpp:363
[... 267 lines cut ...]
[New Thread 0x7fffeeb38640 (LWP 798495)]
[New Thread 0x7fffee337640 (LWP 798574)]
[New Thread 0x7fffddfff640 (LWP 799066)]
[New Thread 0x7fffdd7fe640 (LWP 799067)]
[New Thread 0x7fffdcffd640 (LWP 799068)]
[New Thread 0x7fffd3fff640 (LWP 799069)]
[New Thread 0x7fffd17fe640 (LWP 799070)]
[New Thread 0x7fffccffd640 (LWP 799071)]
[New Thread 0x7fffcc7fc640 (LWP 799072)]
[New Thread 0x7fffc9ffb640 (LWP 799073)]
[New Thread 0x7fffc77fa640 (LWP 799074)]
[New Thread 0x7fffc2ff9640 (LWP 799075)]
[New Thread 0x7fffc27f8640 (LWP 799076)]
[New Thread 0x7fffbfff7640 (LWP 799077)]
[New Thread 0x7fffbd7f6640 (LWP 799078)]
[New Thread 0x7fffbcff5640 (LWP 799079)]
[New Thread 0x7fffba7f4640 (LWP 799080)]
after node

```

</p>
</details>

What I expect is to use these unwind functions in either `gcc`([1](https://github.com/gcc-mirror/gcc/blob/7adcbafe45f8001b698967defe682687b52c0007/libgcc/unwind-seh.c#L329),[2](https://github.com/gcc-mirror/gcc/blob/7adcbafe45f8001b698967defe682687b52c0007/libgcc/unwind-seh.c#L346)) or `libunwind`([1](https://github.com/libunwind/libunwind/blob/304f9ea971be33c790e1f28f1b268f6c73458183/src/unwind/RaiseException.c#L29),[2](https://github.com/libunwind/libunwind/blob/304f9ea971be33c790e1f28f1b268f6c73458183/src/unwind/Resume.c#L29)) durning [throw](https://github.com/gcc-mirror/gcc/blob/7adcbafe45f8001b698967defe682687b52c0007/libstdc++-v3/libsupc++/eh_throw.cc#L78).

iuhilnehc-ynos, 2023-08-10:
One final solution I can figure out is to use the `LD_PRELOAD`.
```
e.g.
$ export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1
$ python3 gst_rclpy.py
```

russkel, 2024-04-24:
Still getting this with rolling. However, I do have a work around: If you initialise the Node before you import gi+Gst you do not segfault.

```python
def main(args=None):
    rclpy.init(args=args, signal_handler_options=SignalHandlerOptions.SIGTERM)
    executor = SingleThreadedExecutor()

    node = Pipeline()
    executor.add_node(node)

    # this hack is because if you import Gst before the node is initialised, it will segfault
    global Gst, Glib
    import gi
    gi.require_version('Gst', '1.0')
    from gi.repository import Gst, GLib

    Gst.init(None)
    Gst.debug_set_default_threshold(Gst.DebugLevel.DEBUG)
    node.loop = GLib.MainLoop()

    glib_mainloop_thread = threading.Thread(target=node.loop.run)
    glib_mainloop_thread.start()

    node.setup_pipeline()
    node.start_pipeline()

    try:
        executor.spin()
    except KeyboardInterrupt:
      pass
    
    node.get_logger().info('Caught interrupt signal, stopping pipeline')
    node.stop_pipeline()
    node.loop.quit()
    glib_mainloop_thread.join()

    node.destroy_node()
    # executor.shutdown()
    rclpy.shutdown()

    Gst.deinit()
```

russkel, 2024-04-24:
> If libgstreamer-1.0.so.0 includes the linunwind symbols then it should use a linker script to avoid them being exported. Otherwise you get incompatible symbols with the same name.

```bash
ros@kalesmoothie:/greenstream-packages$ nm -gD /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so | grep -i unwind
ros@kalesmoothie:/greenstream-packages$ 
```

This is on ubuntu noble.

russkel, 2024-05-02:
I had thought my original workaround was working - but I was mistaken and they have returned.

I did notice that FastDDS exports this symbol... could this be the cause of the problems? 

```
$ nm -gD /opt/ros/rolling/lib/libfastrtps.so.2.14.0  | grep -i unwind
                 U _Unwind_Resume@GCC_3.0
                 U _ZTIN10__cxxabiv115__forced_unwindE@CXXABI_1.3.2
```