ros2/rclpy #1154: Clarification about switching to branch before `colcon build`

a-mhamdi, 2023-08-25:
In the `README` file, I guess we do need to checkout to the installed _ros distro branch_ `${ROS_DISTRO}`. I did the following:

```zsh
source ~/ros2_humble/install/local_setup.zsh
git clone https://github.com/ros2/rclpy.git
git checkout humble
python -m colcon build --symlink-install
```

Am I wrong?

clalancette, 2023-08-25:
Actually, we should probably just remove the building instructions completely and just have them point to http://docs.ros.org/ instead.

a-mhamdi, 2023-08-25:
Actually, I find the instructions useful. I got an error when trying to test the `ros2` installation
>[gibberish] /install/rclpy/lib/python3.10/site-packages/rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so: file too short
The C extension '/home/mhamdi/ros2_humble/install/rclpy/lib/python3.10/site-packages/rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so' failed to be imported while being present on the system.

I got the same error namely when I tried
```python
import rclpy
```
I switched to `humble` branch, and as mentioned in the `README.md` file,  I did the `colcon build` as suggested, and moved the `rclpy` library to `site-packages`. It worked like a charm. 

Anyway, I'm closing the issue. I simply want to know if I'm missing anything or did something wrong. Thanks, @clalancette :D

closed 2023-08-25 by a-mhamdi