ros2/rclpy #1159: SingleThreadedExecutor with awaited service call doesn't process other callbacks only when rclpy.spin is passed a timeout

jpfeltracco, 2023-09-02:
## Bug report

**Required Info:**

- Operating System:
  - ros:humble dockerfile, based on ubuntu:jammy
- Installation type:
  - binaries for ros:humble
- Version or commit hash:
  - 3.3.9
- DDS implementation:
  - default
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Run the following:
```
import rclpy
from rclpy.node import Node
from rclpy.callback_groups import ReentrantCallbackGroup
from rclpy.task import Future

# Random service example.
from ros2param.api import GetParameters

class AsyncClient(Node):
    def __init__(self):
        super().__init__("async_client_node")
        callback_group = ReentrantCallbackGroup()

        self.cli = self.create_client(
            GetParameters, "test_service", callback_group=callback_group
        )
        self.service_call_timer = self.create_timer(
            1.0, self.call_service_cb, callback_group=callback_group
        )
        self.log_timer = self.create_timer(
            2.0,
            self.log_cb,
            callback_group=callback_group,
        )

    def log_cb(self):
        self.get_logger().info("Logging timer fired.")

    async def call_service_cb(self):
        self.service_call_timer.cancel()

        self.get_logger().info("Calling service")
        # I know there's no service, this is intentional and I expect this task to not complete,
        # but other tasks should still be able to complete!
        response = await self.cli.call_async(GetParameters.Request())
        self.get_logger().info(f"Service call response: {response.message}")

rclpy.init()
node = AsyncClient()

# If I run this, my log_cb is called.
# rclpy.spin_until_future_complete(node, Future())
# If I run this, my log_cb isn't called!
rclpy.spin_until_future_complete(node, Future(), timeout_sec=5.0)
```

#### Expected behavior

I get "Logging timer fired." printed every 2 seconds. The service should never complete.

#### Actual behavior

The service attempts to get called, and then from then on we're blocked.

#### Additional information

I'm wanting to run my nodes as single threaded yet asynchronous. This works generally, but I've noticed that if I include a timeout in any spin call to rclpy, this stops working.

I've dug a bit into the executor implementation. In `wait_for_ready_callbacks` I see that when a timeout is included, the generator gets recreated every single time the a spin_once is invoked, ie this print out is constant.

```
    def wait_for_ready_callbacks(self, *args, **kwargs) -> Tuple[Task, WaitableEntityType, 'Node']:
        """
        Return callbacks that are ready to be executed.

        The arguments to this function are passed to the internal method
        :meth:`_wait_for_ready_callbacks` to get a generator for ready callbacks:

        .. Including the docstring for the hidden function for reference
        .. automethod:: _wait_for_ready_callbacks
        """
        while True:
            if self._cb_iter is None or self._last_args != args or self._last_kwargs != kwargs:
                print("Creating a new generator")
                # Create a new generator
                self._last_args = args
                self._last_kwargs = kwargs
                self._cb_iter = self._wait_for_ready_callbacks(*args, **kwargs)

            try:
                return next(self._cb_iter)
            except StopIteration:
                # Generator ran out of work
                self._cb_iter = None
```

Now I think the problem is that when looking into the `_wait_for_ready_callbacks` implementation, if there is an existing not done task, that will be yielded first in the generator, without going and adding any other tasks from timers or other entities. Since we create a new generator each time, I just get the same service call task every single time?

```
            if tasks:
                for task, entity, node in reversed(tasks):
                    if (not task.executing() and not task.done() and
                            (node is None or node in nodes_to_use)):
                        yielded_work = True
                        yield task, entity, node
                with self._tasks_lock:
                    # Get rid of any tasks that are done
                    self._tasks = list(filter(lambda t_e_n: not t_e_n[0].done(), self._tasks))
```

I'm more than happy to provide a PR, if I can get a little guidance on an approach to solve this.

Barry-Xu-2018, 2023-09-04:
After checking, I share my thought about this issue. My thought is almost same as your additional information.

The root cause is about  `self._last_args != args`. While using timeout, self._last_args always not equals args.

https://github.com/ros2/rclpy/blob/03bcc32d7d530d8523f134c48fb35a96e02daa0d/rclpy/rclpy/executors.py#L745-L756

While using timeout, timeout_left parameter is changed before calling `spin_once_until_future_complete()`. The timeout parameter will pass to `wait_for_ready_callbacks()`.

https://github.com/ros2/rclpy/blob/03bcc32d7d530d8523f134c48fb35a96e02daa0d/rclpy/rclpy/executors.py#L312-L319

This leads _wait_for_ready_callbacks() is always recalled instead of next() which getting the iterator by yield in _wait_for_ready_callbacks().

https://github.com/ros2/rclpy/blob/03bcc32d7d530d8523f134c48fb35a96e02daa0d/rclpy/rclpy/executors.py#L756-L763

Since call_service_cb() cannot return, the task for call_service_cb() always exist and satisfy condition `  (not task.executing() and not task.done() and (node is None or node in nodes_to_use))` and directly return from _wait_for_ready_callbacks(). As above mention, if timeout is used,  _wait_for_ready_callbacks() is always recalled. This leads the following process below L559 never be called. That is, timer_log is never processed.

https://github.com/ros2/rclpy/blob/03bcc32d7d530d8523f134c48fb35a96e02daa0d/rclpy/rclpy/executors.py#L551-L559

A fixing method is to modify the declaration of wait_for_ready_callbacks()
from
```python
def wait_for_ready_callbacks(self, *args, **kwargs) -> Tuple[Task, WaitableEntityType, 'Node']
```
to
```python
def wait_for_ready_callbacks(self, timeout_sec, *args, **kwargs) -> Tuple[Task, WaitableEntityType, 'Node']
```

fujitatomoya, 2023-09-06:
we can reproduce this issue with `rolling` https://github.com/ros2/ros2/commit/3490c0f49487693dbaa58123504d5cc3c278099d. i believe https://github.com/ros2/rclpy/issues/1159#issuecomment-1704966783 makes sense to me. @Barry-Xu-2018 are you willing to create PR for this?

sloretz, 2023-09-06:
I agree with the analysis above on the cause being the re-creating of the generator due to the arguments to `wait_for_ready_callbacks()` changing.

I'm finding it difficult to reason about if `wait_for_ready_callbacks()` excluding `timeout_sec` from the arguments has any other side effects.

The only thing I can think of is suppose if two calls to `spin_once()` with different timeouts such that the first call has a really short timeout and the second call has a really long timeout, then the second call would reuse the generator from the first call and have the timeout timer from the first. Presumably if we're reusing the generator in the second call then it means the first call yielded work, so we're not going to wrap around to re-check the wait set, and the timeout timer won't matter.

So, it seems like an ok fix, but I can't rule out that there are other cases that might be affected.

Barry-Xu-2018, 2023-09-07:
> @Barry-Xu-2018 are you willing to create PR for this?
@fujitatomoya jpfeltracco mentioned that he wanted to submit a PR to fix this issue. 

@jpfeltracco Do you want to submit a PR ?

Barry-Xu-2018, 2023-09-07:
> The only thing I can think of is suppose if two calls to spin_once() with different timeouts such that the first call has a really short timeout and the second call has a really long timeout, then [...]

Yes, excluding timeout_sec from the arguments affects this case.

Barry-Xu-2018, 2023-09-18:
@jpfeltracco I will take over the task of fixing this bug. If you want to handle this, please let me know.

fujitatomoya, 2023-09-18:
@Barry-Xu-2018 sorry i forgot to let you know, that we had a talk about this internally. and https://github.com/ros2/rclpy/issues/1159#issuecomment-1704966783 seems to be okay as a fix, so please open the PR.
CC: @sloretz

Barry-Xu-2018, 2023-09-19:
> sorry i forgot to let you know, that we had a talk about this internally. and https://github.com/ros2/rclpy/issues/1159#issuecomment-1704966783 seems to be okay as a fix, so please open the PR.

Okay. I have made PR. But 2 of the existing tests fail, I will submit PR after tests all pass.

jpfeltracco, 2023-09-21:
Thanks folks!!

Barry-Xu-2018, 2023-10-11:
@jpfeltracco  
The fix had been merged.    
Could you try it ? If there is no further problem, please close this issue.

fujitatomoya, 2023-10-11:
@Barry-Xu-2018 @clalancette @sloretz original issue came with Humble, we should backport this fix? if so, i can handle backporting. this is unexpected behavior for user application, so probably we need to fix Iron and Humble?

Barry-Xu-2018, 2023-10-11:
@fujitatomoya 
#1165 will change interface.  
e.g.
[image]  

I am not sure if this PR can backport for rclpy.  
@clalancette @sloretz What's your opinion?

clalancette, 2023-10-11:
> I am not sure if this PR can backport for rclpy.

Given that the we didn't change the API (we only added to it), I think we can backport it.

closed 2023-11-03 by fujitatomoya