ros2/rclpy #1160: Timer documentation is missing

nstiurca, 2023-09-08:
## Bug report

**Required Info:**

- Operating System: N/A
- Installation type: N/A
- Version or commit hash: 0.6.1
- DDS implementation: N/A
- Client library (if applicable): rclpy

#### Steps to reproduce issue
Go to https://docs.ros2.org/dashing/api/rclpy/api/timers.html

#### Expected behavior
See actual documentation

#### Actual behavior
See only documentation stub

clalancette, 2023-09-08:
That documentation is very old; the modern documentation is at https://docs.ros.org/en/rolling/p/rclpy/ , and the class you are interested in particular is in https://docs.ros.org/en/rolling/p/rclpy/rclpy.timer.html#rclpy.timer.Timer.is_canceled .

That said, this is a good reminder that I need to decommission https://docs.ros2.org, so I'll make sure to put that on the TODO list.

ZeinBarhoum, 2023-09-12:
BTW, there is no documentation for `rclpy` in the `humble` version of the docs. 

there is https://docs.ros.org/en/rolling/p/rclpy/
but not https://docs.ros.org/en/humble/p/rclpy/

Why is that the case?

clalancette, 2023-09-12:
> Why is that the case?

We had to do some changes to the underlying code, configuration, and tooling, to get `rclpy` documentation generation working for Iron.  It *might* be possible to backport that to Humble, but nobody has attempted it.

ZeinBarhoum, 2023-09-12:
Ok, thanks for the explanation

nstiurca, 2023-09-20:
> That documentation is very old; the modern documentation is at https://docs.ros.org/en/rolling/p/rclpy/ , and the class you are interested in particular is in https://docs.ros.org/en/rolling/p/rclpy/rclpy.timer.html#rclpy.timer.Timer.is_canceled .

Fair, but even so the documentation is quite minimal/stubbed in the updated docs.

nstiurca, 2023-09-20:
> That said, this is a good reminder that I need to decommission https://docs.ros2.org/, so I'll make sure to put that on the TODO list.

And yeah I just went off whatever search result I got first. Frustrating to get old info in a case like this.

dghw, 2023-10-26:
The 'Node' documentation isn't showing up for me on the modern site either for Iron or Rolling. 

https://docs.ros.org/en/iron/p/rclpy/api/node.html
https://docs.ros.org/en/rolling/p/rclpy/api/node.html

The 'Subscription' page explicitly points people towards the Node documentation 

> Users should not create a subscription with this constructor, instead they should call Node.create_subscription().

so it feels like there should be something here.

closed 2025-02-18 by Yadunund

dghw, 2025-02-18:
@Yadunund On Jazzy the Node documentation also seems missing for me. Although the Timer documentation does show up. Is it the same for you?

https://docs.ros.org/en/jazzy/p/rclpy/api/node.html

Yadunund, 2025-02-18:
Looks like https://github.com/ros2/rclpy/pull/1408 needs to be backported to Jazzy.