ros2/rclpy #1166: rclpy throws an exception when it is interrupted - rclpy.executors.ExternalShutdownException

ionutnechita-intel, 2023-09-21:
## Bug report

rclpy throws an exception when it is interrupted - rclpy.executors.ExternalShutdownException

**Required Info:**

- Operating System:
  - Ubuntu 22.04 IOT LTS (Desktop Variant)
- Installation type:
  - Binary
- Version or commit hash:
  - 3.3.9-1jammy.20230822.175037
- DDS implementation:
  - default DDS
- Client library (if applicable):
  - rclpy
- ROS Distro: 
  -  Humble

#### Steps to reproduce issue
Execute command: 
1. Execute in terminal: timeout 20s ros2 topic hz /camera/color/image_raw
2. Press CTRL-C on keyboard.

#### Expected behavior
No error on output terminal.

#### Actual behavior
Messages with errors in output terminal.

```
File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.7', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/verb/hz.py", line 89, in main
    return main(args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/verb/hz.py", line 104, in main
    _rostopic_hz(node.node, topic, window_size=args.window_size, filter_expr=filter_expr,
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/verb/hz.py", line 273, in _rostopic_hz
    rclpy.spin_once(node)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 202, in spin_once
    executor.spin_once(timeout_sec=timeout_sec)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 705, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 691, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 592, in _wait_for_ready_callbacks
    raise ExternalShutdownException()
rclpy.executors.ExternalShutdownException
 ```

fujitatomoya, 2023-09-21:
with my local environment, this cannot be reproducible so far. (i think that is just racy condition)

```bash
root@tomoyafujita:~# dpkg -l | grep rclpy
ii  ros-humble-rclpy                                  3.3.9-1jammy.20230822.175037            amd64        Package containing the Python client.
root@tomoyafujita:~# ros2 topic pub -r 1 /chatter std_msgs/msg/String "data: Hello"
publisher: beginning loop
publishing #1: std_msgs.msg.String(data='Hello')

publishing #2: std_msgs.msg.String(data='Hello')

publishing #3: std_msgs.msg.String(data='Hello')

...<snip>

root@tomoyafujita:~# timeout 20s ros2 topic hz /chatter
average rate: 1.000
        min: 1.000s max: 1.000s std dev: 0.00003s window: 2
average rate: 1.000
        min: 1.000s max: 1.000s std dev: 0.00006s window: 3
^Croot@tomoyafujita:~# timeout 20s ros2 topic hz /chatter
average rate: 1.000
        min: 1.000s max: 1.000s std dev: 0.00005s window: 3
average rate: 1.000
        min: 1.000s max: 1.000s std dev: 0.00006s window: 5
average rate: 1.000
        min: 1.000s max: 1.000s std dev: 0.00007s window: 6
^Croot@tomoyafujita:~#
```

related to https://github.com/ros2/rclpy/issues/841

ionutnechita-intel, 2023-09-22:
Hi @fujitatomoya,

I reinstalled my system and this issue is reproduced again.

SIGTERM signal send on rclpy is not terminated correctly in this case?

fujitatomoya, 2023-09-22:
as far as i see, `ExternalShutdownException` is generated and `SingleThreadedExecutor` intentionally does not catch, that exception is supposed to be raised to the application to handle. (see more details forhttps://github.com/ros2/rclpy/pull/830 )
after all, i think the application needs to handle this exception, by here application means `ros2cli`.

CC: @Barry-Xu-2018 @iuhilnehc-ynos

fujitatomoya, 2023-09-22:
> SIGTERM signal send on rclpy is not terminated correctly in this case?

to be precise, if `SIGTERM` comes in, this call stack can be reproducible since `ros2cli` does not catch that exception.

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# timeout 60s ros2 topic hz /chatter
average rate: 1.000
        min: 1.000s max: 1.000s std dev: 0.00003s window: 3
average rate: 1.000
...<snip>
Traceback (most recent call last):
  File "/root/ros2_ws/colcon_ws/install/ros2cli/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli', 'console_scripts', 'ros2')())
  File "/root/ros2_ws/colcon_ws/build/ros2cli/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/root/ros2_ws/colcon_ws/build/ros2topic/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
  File "/root/ros2_ws/colcon_ws/build/ros2topic/ros2topic/verb/hz.py", line 79, in main
    return main(args)
  File "/root/ros2_ws/colcon_ws/build/ros2topic/ros2topic/verb/hz.py", line 94, in main
    _rostopic_hz(node.node, topic, window_size=args.window_size, filter_expr=filter_expr,
  File "/root/ros2_ws/colcon_ws/build/ros2topic/ros2topic/verb/hz.py", line 263, in _rostopic_hz
    rclpy.spin_once(node)
  File "/root/ros2_ws/colcon_ws/install/rclpy/lib/python3.10/site-packages/rclpy/__init__.py", line 217, in spin_once
    executor.spin_once(timeout_sec=timeout_sec)
  File "/root/ros2_ws/colcon_ws/install/rclpy/lib/python3.10/site-packages/rclpy/executors.py", line 809, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/root/ros2_ws/colcon_ws/install/rclpy/lib/python3.10/site-packages/rclpy/executors.py", line 793, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(
  File "/root/ros2_ws/colcon_ws/install/rclpy/lib/python3.10/site-packages/rclpy/executors.py", line 763, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "/root/ros2_ws/colcon_ws/install/rclpy/lib/python3.10/site-packages/rclpy/executors.py", line 671, in _wait_for_ready_callbacks
    raise ExternalShutdownException()
rclpy.executors.ExternalShutdownException

root@tomoyafujita:~/ros2_ws/colcon_ws# kill -SIGTERM 247998
```

fujitatomoya, 2023-09-22:
@Barry-Xu-2018 
CC: @iuhilnehc-ynos 

you opened https://github.com/ros2/rclpy/issues/841, and https://github.com/ros2/rclpy/issues/841#issuecomment-966530691 makes sense to me that user needs to catch the exception. something i do not figure out here is, why we keep https://github.com/ros2/rclpy/blob/827d59729f266d9449d8fcb9752707bbd052527d/rclpy/rclpy/executors.py#L862-L863 for `MultiThreadedExecutor`? It should be consistent behavior with `SingleThreadedExecutor`?

besides the question, `ros2cli` should catch `ExternalShutdownException`? (async signal hanlder can shutdown the context.)

fujitatomoya, 2023-09-22:
@Barry-Xu-2018 @iuhilnehc-ynos https://github.com/ros2/ros2cli/pull/854 does this make sense? it can avoid the problem.

Barry-Xu-2018, 2023-09-25:
@fujitatomoya 

> for MultiThreadedExecutor? It should be consistent behavior with SingleThreadedExecutor?

Agree. I checked comments in #841 and #830. No one mentioned ExternalShutdownException for MultiThreadedExecutor.  
I'm not sure if there are other reasons leading to different processing in MultiThreadedExecutor.  

> besides the question, ros2cli should catch ExternalShutdownException? (async signal hanlder can shutdown the context.)

Yes.  ros2cli should handle ExternalShutdownException.

ionutnechita-intel, 2023-11-15:
Hi @fujitatomoya,

This change is merged in Humble ROS2 branch?

fujitatomoya, 2023-11-15:
@ionutnechita-intel not yet, thanks for the ping. i will start the CI for humble and iron.

fujitatomoya, 2023-11-16:
backport to humble and iron are completed, closing.

closed 2023-11-16 by fujitatomoya

ionutnechita-intel, 2023-11-17:
Thank you @fujitatomoya for all support.