ros2/rclpy #1179: wait_for_message Accumulating CPU Load After Repeated Calls

KKSTB, 2023-10-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Iron
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
1. Setup and spin single threaded node executor in a new thread
2. Call wait_for_message in main thread or another thread:
```python3
for i in range(100):
    is_success, msg = rclpy.wait_for_message.wait_for_message(msg_type, rclpy_node, topic, time_to_wait=time_to_wait)
```

#### Expected behavior
CPU% stays the same after the for loop
#### Actual behavior
CPU% increased and does not drop
#### Additional information
`py-spy record -n -r 10 -p [PID]` indicates high CPU load at _take_subscription (rclpy/executors.py 338 https://github.com/ros2/rclpy/blob/e9bd428ed562019596e8d3964489bab90a85b203/rclpy/rclpy/executors.py#L338C25-L338C43).

KKSTB, 2023-10-03:
Fix is simple: just unsubscribe before return (tested). This should work assuming that there is no exception in between.

clalancette, 2023-10-03:
> Fix is simple: just unsubscribe before return (tested). This should work assuming that there is no exception in between.

Yes, we should definitely destroy the subscription.  I think we should put it into a `try..finally` block, so we make sure it is always destroyed.  Would you submit a pull request to do that?

closed 2023-10-10 by KKSTB

fujitatomoya, 2023-10-10:
@KKSTB i was thinking probably we want to backport this to iron and humble, since this affects user platform. what do you think? besides, the issue was originally created with iron.

CC: @Yadunund @clalancette

KKSTB, 2023-10-11:
Ah you are right. Thank you for reminding me.

FelixLittleFei, 2023-12-20:
@KKSTB @fujitatomoya @clalancette 
Dear ROS2 developers, 
Could you please backport to ROS2 Humble for this wait_for_message function?
[image]

I noticed that it already exists in Rolling and Iron.
Thank you!

KKSTB, 2023-12-21:
Hi @FelixLittleFei 

Actually I tested that the wait_for_message function is also affected by this issue:
https://github.com/ros2/rclpy/issues/1206

If wait_for_message is called many times the executor may sometimes throw exception saying it's using a destroyed subscription.

So I actually gave up in using this function in my application. I changed to permanently subscribe to topic instead.

fujitatomoya, 2024-04-19:
@tiesus currently nobody is working on that, but i think that is acceptable. CC: @clalancette

fujitatomoya, 2024-04-21:
humble backport: https://github.com/ros2/rclpy/pull/1272

tiesus, 2024-04-22:
Great.  That helps us a lot to get rid of ros1.