ros2/rclpy #1181: Service client sometimes deadlock due to 'client will not receive response' of newly launched service server node

KKSTB, 2023-10-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - Iron
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp, rclpy, controller_manager

#### Steps to reproduce issue
1. Programmatically launch a launch python containing controller_manager using method https://github.com/ros2/launch/issues/724
2. Create controller_manager/load_controller client, wait_for_service() and call the service with a valid controller
3. destroy_client()

#### Expected behavior
Execute as normal
#### Actual behavior
Client stuck at client.call(), with the following printout:
```
my program print('robot/controller_manager/load_controller before wait_for_service()') --> already waiting for service
[INFO] [ros2_control_node-1]: process started with pid [25427]
[INFO] [robot_state_publisher-2]: process started with pid [25429]
[INFO] [spawner-3]: process started with pid [25431]
[robot_state_publisher-2] [INFO] [1696905559.365691573] [robot.robot_state_publisher]: got segment robot_base_ref
[robot_state_publisher-2] [INFO] [1696905559.365703944] [robot.robot_state_publisher]: got segment robot_link1
[robot_state_publisher-2] [INFO] [1696905559.365716391] [robot.robot_state_publisher]: got segment robot_link2
[robot_state_publisher-2] [INFO] [1696905559.365727908] [robot.robot_state_publisher]: got segment robot_link3
[robot_state_publisher-2] [INFO] [1696905559.365739118] [robot.robot_state_publisher]: got segment robot_link4
[robot_state_publisher-2] [INFO] [1696905559.365751414] [robot.robot_state_publisher]: got segment robot_link5
[robot_state_publisher-2] [INFO] [1696905559.365761952] [robot.robot_state_publisher]: got segment robot_link6
[robot_state_publisher-2] [INFO] [1696905559.365773261] [robot.robot_state_publisher]: got segment robot_link_base
[robot_state_publisher-2] [INFO] [1696905559.365784672] [robot.robot_state_publisher]: got segment robot_link_eef
[ros2_control_node-1] [WARN] [1696905559.372291758] [robot.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-1] [INFO] [1696905559.372971860] [resource_manager]: Loading hardware 'robot_fake_components/GenericSystem' 
[ros2_control_node-1] [INFO] [1696905559.373770098] [resource_manager]: Initialize hardware 'robot_fake_components/GenericSystem' 
[ros2_control_node-1] [WARN] [1696905559.373822258] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-1] <state_interface name="velocity"> 
[ros2_control_node-1]   <param name="initial_value">0.0</param> 
[ros2_control_node-1] </state_interface>
[ros2_control_node-1] [INFO] [1696905559.373840623] [resource_manager]: Successful initialization of hardware 'robot_fake_components/GenericSystem'
[ros2_control_node-1] [INFO] [1696905559.374000581] [resource_manager]: 'configure' hardware 'robot_fake_components/GenericSystem' 
[ros2_control_node-1] [INFO] [1696905559.374007372] [resource_manager]: Successful 'configure' of hardware 'robot_fake_components/GenericSystem'
[ros2_control_node-1] [INFO] [1696905559.374018963] [resource_manager]: 'activate' hardware 'robot_fake_components/GenericSystem' 
[ros2_control_node-1] [INFO] [1696905559.374043514] [resource_manager]: Successful 'activate' of hardware 'robot_fake_components/GenericSystem'
[ros2_control_node-1] [INFO] [1696905559.388264078] [robot.controller_manager]: update rate is 100 Hz
[ros2_control_node-1] [INFO] [1696905559.388443970] [robot.controller_manager]: RT kernel is recommended for better performance
my program print('robot/controller_manager/load_controller before call()') --> call() is called
[ros2_control_node-1] [INFO] [1696905559.601120958] [robot.controller_manager]: Loading controller 'robot_joint_trajectory_controller'
[ros2_control_node-1] [WARN] [1696905559.626217261] [robot.robot_joint_trajectory_controller]: [Deprecated]: "allow_nonzero_velocity_at_trajectory_end" is set to true. The default behavior will change to false.
[ros2_control_node-1] [WARN] [1696905559.728814556] [robot.controller_manager.rclcpp]: failed to send response to /robot/controller_manager/load_controller (timeout): client will not receive response, at ./src/rmw_response.cpp:149, at ./src/rcl/service.c:402
[ros2_control_node-1] [INFO] [1696905559.744885882] [robot.controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-3] [INFO] [1696905559.793118712] [robot.spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-1] [INFO] [1696905559.794509173] [robot.controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-1] [INFO] [1696905559.794604989] [robot.joint_state_broadcaster]: Publishing state interfaces defined in 'joints' and 'interfaces' parameters.
[ros2_control_node-1] [WARN] [1696905559.794633095] [robot.joint_state_broadcaster]: Mapping from 'position' to interface 'position' will not be done, because 'position' is defined in 'interface' parameter.
[ros2_control_node-1] [WARN] [1696905559.794647253] [robot.joint_state_broadcaster]: Mapping from 'velocity' to interface 'velocity' will not be done, because 'velocity' is defined in 'interface' parameter.
[spawner-3] [INFO] [1696905559.819715634] [robot.spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster

```
#### Additional information
I suspect it is due to race condition when handling multiple service requests at the same time. In this case `load_controller` of my program and controller spawner.

KKSTB, 2023-10-10:
Could be related: https://github.com/ros2/rmw_fastrtps/pull/704

The default timeout of 100ms could be insufficient on my computer. Have to wait for the backport to iron to become available in order to verify it. ros-iron-rmw-fastrtps-cpp is v7.1.1 now.

fujitatomoya, 2023-10-10:
according to the problem description,

- https://github.com/ros2/ros2/issues/1253
- https://github.com/ros2/rmw_fastrtps/issues/556

those are the related problems,

as you mentioned, https://github.com/ros2/rmw_fastrtps/pull/704 can mitigate the racy condition.

> ros-iron-rmw-fastrtps-cpp is v7.1.1 now.

yeah, it just misses `v7.1.1`. what you can try right now is, using [ROS 2 Iron Source Build](https://docs.ros.org/en/iron/Installation.html#building-from-source).

KKSTB, 2023-10-11:
In rclcpp there is `async_send_request()` but there is no non-async version of service call. This leaves timeout mechanism to the like of the programmer.

I think python non-async service call should add some mechanism to escape from the deadlock, and then `remove_pending_request()` of the future to free the resources, when the service is unreliable. Im thinking how to code it.

clalancette, 2023-10-11:
> I think python non-async service call should add some mechanism to escape from the deadlock, and then `remove_pending_request()` of the future to free the resources, when the service is unreliable. Im thinking how to code it.

Personally, I think we should deprecate and remove `client.call`.  It almost always causes problems, and it is easy for the user to do it themselves in a "safe" context (by doing a wait on the returned future).

KKSTB, 2023-10-13:
I prefer trying to fix the issue. The `call()` function is much easier to use than async function, and (ironically) should be quite popular as indicated by the number of problem reports.

If a timeout is added, the original "warning: callback causes deadlock" behavior will change (https://github.com/ros2/rclpy/commit/9b70fcb7e0d25d53141f9b8f551677a91b32bf6b). Documentation needs to be updated accordingly.

closed 2023-11-17 by fujitatomoya

KKSTB, 2023-11-29:
I think this could be back-ported to iron and humble since the API is compatible, and the issue was first discovered in iron. What do you think?

fujitatomoya, 2023-11-29:
coming back to https://github.com/ros2/rclpy/pull/1188#pullrequestreview-1686030933 comment, i am not inclined to backport this to iron and humble. any thoughts? @ros2/team