ros2/rclpy #1189: Log files are given  base name `python3....` instead of node name

russkel, 2023-10-18:
## Bug report

Python node log files do not seem to be naming themselves with node name.

Related: https://github.com/ros2/rosbag2/issues/1212

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - debs
- Version or commit hash:
  - ros-iron-rclpy/jammy,now 4.1.3-1jammy.20230908.172649 amd64 [installed,automatic]
- DDS implementation:
  - 
- Client library (if applicable):
  - rclpy rclcpp ros2cli

#### Steps to reproduce issue
```
$ russ @ kalesmoothie in ~ [15:25:39]
↳ rm -rf ~/.ros/log/*
zsh: sure you want to delete more than 100 files in /home/russ/.ros/log [yn]? y

$ russ @ kalesmoothie in ~ [15:26:20]
↳ source /opt/ros/humble/setup.zsh 

$ russ @ kalesmoothie in ~ [15:26:50]
↳ ros2 bag record --all
[INFO] [1675225614.830445834] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1675225614.833532621] [rosbag2_storage]: Opened database 'rosbag2_2023_02_01-15_26_54/rosbag2_2023_02_01-15_26_54_0.db3' for READ_WRITE.
[INFO] [1675225614.833842246] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1675225614.835087643] [rosbag2_recorder]: Listening for topics...
[INFO] [1675225614.836427656] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1675225614.836970741] [rosbag2_recorder]: Subscribed to topic '/events/write_split'
[INFO] [1675225619.656144904] [rclcpp]: signal_handler(signum=2)
[INFO] [1675225619.749181886] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1675225619.851948745] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while

$ russ @ kalesmoothie in ~ [15:26:59]
↳ ls ~/.ros/log/                  
 python3_2730488_1675225614806.log

$ russ @ kalesmoothie in ~ [15:27:54]
↳ cat ~/.ros/log/python3_2730488_1675225614806.log 
[INFO] [1675225614.830445834] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1675225614.833532621] [rosbag2_storage]: Opened database 'rosbag2_2023_02_01-15_26_54/rosbag2_2023_02_01-15_26_54_0.db3' for READ_WRITE.
[INFO] [1675225614.833842246] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1675225614.835087643] [rosbag2_recorder]: Listening for topics...
[INFO] [1675225614.836427656] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1675225614.836970741] [rosbag2_recorder]: Subscribed to topic '/events/write_split'
[INFO] [1675225619.656144904] [rclcpp]: signal_handler(signum=2)
```

#### Expected behavior

Log file name would be `rosbag2_recorder_2730488_1675225614806.log`

#### Actual behavior

File name is `python3_2730488_1675225614806.log`

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

fujitatomoya, 2023-10-20:
this is unknown issue with https://github.com/ros2/ros2cli/issues/856, this is not related to `rclpy`. please see more details for https://github.com/ros2/ros2cli/issues/856#issuecomment-1736481724

russkel, 2023-10-22:
> this is unknown issue with [ros2/ros2cli#856](https://github.com/ros2/ros2cli/issues/856), this is not related to `rclpy`. please see more details for [ros2/ros2cli#856 (comment)](https://github.com/ros2/ros2cli/issues/856#issuecomment-1736481724)

Thanks for the link. I thought it was more than just `ros2cli` because a number of other nodes seem to have the similar file naming. I will have to investigate more.

sloretz, 2023-11-02:
Thank you for the report! I'm going to close it as a duplicate of https://github.com/ros2/ros2cli/issues/856

closed 2023-11-02 by sloretz