ros2/rclpy #1191: Service Client wait_for_service incorrectly returns True

Jannkar, 2023-10-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04.3 LTS
- Installation type:
  - binaries
- Version or commit hash:
  - 3.3.9
- DDS implementation:
  - rmw_cyclonedds_cpp
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

- Create a service client and service server
- Use the client to call the service
- Destroy the client and the service
- Recreate the client and call it's wait_for_service
```
#!/usr/bin/env python3
import threading
from time import sleep

# ROS
import rclpy
from example_interfaces.srv import AddTwoInts
from rclpy.executors import MultiThreadedExecutor
from rclpy.node import Node

class ServiceClientNode(Node):
    def __init__(self) -> None:
        super().__init__("service_client")
        self.result_event = threading.Event()

    def call_service_async(self, client):
        req = AddTwoInts.Request(a=0, b=1)
        service_found = client.wait_for_service(timeout_sec=1)
        if not service_found:
            return service_found
        print(f"Wait for service passed with result: {service_found}")
        future = client.call_async(req)
        future.add_done_callback(self.result_cb)
        return service_found

    def result_cb(self, _future):
        print("Result cb")
        self.result_event.set()

def service_callback(req: AddTwoInts.Request, result: AddTwoInts.Response) -> AddTwoInts.Response:
    """Service to add two ints. Execution time is determined by the sum."""
    result.sum = req.a + req.b
    return result

if __name__ == "__main__":
    rclpy.init()

    # Create the Nodes for the service server and client
    server_node = rclpy.create_node("service_server")
    service_server = server_node.create_service(AddTwoInts, "/add_two_ints", callback=service_callback)
    client_node = ServiceClientNode()

    # Spin both of the Nodes in a separate thread
    executor = MultiThreadedExecutor()
    executor.add_node(server_node)
    executor.add_node(client_node)
    thread = threading.Thread(target=executor.spin, daemon=True)
    thread.start()

    # Create a new client, call the service, wait for the result and destroy the client
    client = client_node.create_client(AddTwoInts, "/add_two_ints")
    assert client_node.call_service_async(client)
    assert client_node.result_event.wait(3)
    client_node.result_event.clear()
    client_node.destroy_client(client)

    # Destroy the service server
    server_node.destroy_service(service_server)
    # service_server.destroy()  # End result seems to be the same, no matter which way of destroying we use
    # sleep(0.0001)  # Uncommenting this will make things work as expected

    # Create a new client and call the service again. wait_for_service will return True even tho the server is
    # not available!
    client = client_node.create_client(AddTwoInts, "/add_two_ints")
    assert not client_node.call_service_async(client), "Service was found, even though it was destroyed"
    print("Finished successfully!")

    server_node.destroy_node()
    client_node.destroy_node()
    rclpy.try_shutdown()

```

#### Expected behavior
Service is not found after it has been destroyed

#### Actual behavior
Service is still found

#### Additional information
Sleeping for a very short duration after service_server is being destroyed makes things work correctly.

sloretz, 2023-10-27:
I'm able to reproduce this on `rolling`. I think this is working as intended, but I'd like to hear more about your use case. Do you need a way to tell exactly when an entity is destroyed?

The reason the service isn't destroyed right away is because something is still using it.

https://github.com/ros2/rclpy/blob/df92d9e27c3d62bb764ab512ff7e1c53d93c9af1/rclpy/src/rclpy/destroyable.cpp#L70-L72

That thing is almost certainly the executor running in the other thread

https://github.com/ros2/rclpy/blob/df92d9e27c3d62bb764ab512ff7e1c53d93c9af1/rclpy/rclpy/executors.py#L631-L632

I think why it works with sleeping is it allows the other thread to wake up

https://github.com/ros2/rclpy/blob/df92d9e27c3d62bb764ab512ff7e1c53d93c9af1/rclpy/rclpy/node.py#L1877-L1878

release it's claim on the service, and destroy it for real.

https://github.com/ros2/rclpy/blob/df92d9e27c3d62bb764ab512ff7e1c53d93c9af1/rclpy/src/rclpy/destroyable.cpp#L50-L52

Jannkar, 2023-10-31:
Thanks for checking it out. 

In my use case, I'm basically calling a service, to which I will need to eventually have some type of response: Either the result of the service call or information that the service is not available.

That's why I have a function that checks if the service is available first with `wait_for_service()` and then calls the service asynchronously if everything is ok.

I found this issue when writing tests for this function. In the test, I first successfully create the service server, call it using a client and wait for the response. Then I destroy the server and try to call it again, expecting `wait_for_service` to return False. It took me quite a while to figure out why the execution got stuck, since instead it returned True and the service was called, but without ever getting a response to this call.

So what I'm wondering is that can this cause confusion also for the others? And can this cause issues, if the service seems to be available, but is actually already being destroyed and won't handle any more requests?

And what comes to my use case, I'm now wondering if the `wait_for_service()` and immediately after it calling the service will guarantee some type of a response. Or will I need to anyway have some sort of timeout for recognizing that the service might not respond at all?

sloretz, 2023-11-01:
> I'm now wondering if the `wait_for_service()` and immediately after it calling the service will guarantee some type of a response. Or will I need to anyway have some sort of timeout for recognizing that the service might not respond at all?

I think you're correct about needing some sort of timeout on waiting for the result of the call. I'd recommend [using `client.call_async()`](https://github.com/ros2/rclpy/blob/fe8f4a19f0daf264e3f2becc1c5de0959c1560f5/rclpy/rclpy/client.py#L105) to get a future, then creating a timer [with a period](https://github.com/ros2/rclpy/blob/fe8f4a19f0daf264e3f2becc1c5de0959c1560f5/rclpy/rclpy/timer.py#L34) set to your timeout and [with a callback](https://github.com/ros2/rclpy/blob/fe8f4a19f0daf264e3f2becc1c5de0959c1560f5/rclpy/rclpy/timer.py#L32) that [calls `future.cancel()`](https://github.com/ros2/rclpy/blob/fe8f4a19f0daf264e3f2becc1c5de0959c1560f5/rclpy/rclpy/task.py#L60-L65) if the timeout is reached before the future completes. Then I'd [add a "done" callback to to the future](https://github.com/ros2/rclpy/blob/fe8f4a19f0daf264e3f2becc1c5de0959c1560f5/rclpy/rclpy/task.py#L163-L183) that [cancels the timeout timer](https://github.com/ros2/rclpy/blob/fe8f4a19f0daf264e3f2becc1c5de0959c1560f5/rclpy/rclpy/timer.py#L101-L103) if the service server responds before the timeout.

Jannkar, 2023-11-01:
Thanks for the suggestion! I'll do it like this.

closed 2023-11-03 by sloretz