ros2/rclpy #1194: Ros2 Humble is installed on the Ubuntu 22.04 but I get error for no module named rclpy._rclpy_pybind11 while the package is installed (also rclpy is installed for Python 3.10 not Python 3.11)

monajalal, 2023-11-01:
I have installed ros2 Humble on Ubuntu 22.04. However, despite having only python 3.11, I notice this specific lib is installed for Python 3.10 not 3.11. Overall, due to being new to ros and even more so ros2, I would like to know how to troubleshoot and fix this error.

## Bug report

**Required Info:**

- Operating System:
```
(base) mona@ada:~$ uname -a
Linux ada 6.2.0-35-generic #35~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Oct  6 10:23:26 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
(base) mona@ada:~$ lsb_release -a
LSB Version:	core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy

(base) mona@ada:~$ gcc --version
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

(base) mona@ada:~$ python --version
Python 3.11.5

(base) mona@ada:~/rosbag2video$ git log
commit 13b4e064eb0ad748f3d4e20eb88353bdf76acd21 (HEAD -> master, origin/master, origin/HEAD)
Merge: 44d1637 14a7d26
Author: mlaiacker <post@mlaiacker.de>
Date:   Mon Sep 18 20:11:28 2023 +0200

    Merge branch 'pipebots-humble-dev-joey'
```
- Installation type:
I followed the instructions on these two pages step by step:

    https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html
    https://github.com/IntelRealSense/realsense-ros

- Version or commit hash:
  -  (N/A?)
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
$ git clone https://github.com/mlaiacker/rosbag2video.git
(base) mona@ada:~$ cd rosbag2video/

(base) mona@ada:~/rosbag2video$ git log
commit 13b4e064eb0ad748f3d4e20eb88353bdf76acd21 (HEAD -> master, origin/master, origin/HEAD)
Merge: 44d1637 14a7d26
Author: mlaiacker <post@mlaiacker.de>
Date:   Mon Sep 18 20:11:28 2023 +0200

    Merge branch 'pipebots-humble-dev-joey'

(base) mona@ada:~/rosbag2video$ python ros2bag2video.py ~/Downloads/cheezit_1280_whole.bag 
Traceback (most recent call last):
  File "/home/mona/rosbag2video/ros2bag2video.py", line 21, in <module>
    import rclpy
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 49, in <module>
    from rclpy.signals import install_signal_handlers
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/signals.py", line 15, in <module>
    from rclpy.exceptions import InvalidHandle
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/exceptions.py", line 15, in <module>
[... 75 lines cut ...]
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f787c1a8000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f787c0c1000)
	librcl_logging_spdlog.so => /opt/ros/humble/lib/librcl_logging_spdlog.so (0x00007f787c7d7000)
	librcl_interfaces__rosidl_typesupport_c.so => /opt/ros/humble/lib/librcl_interfaces__rosidl_typesupport_c.so (0x00007f787c0b8000)
	librcl_interfaces__rosidl_generator_c.so => /opt/ros/humble/lib/librcl_interfaces__rosidl_generator_c.so (0x00007f787c09c000)
	libyaml.so => /opt/ros/humble/lib/libyaml.so (0x00007f787c078000)
	libament_index_cpp.so => /opt/ros/humble/lib/libament_index_cpp.so (0x00007f787c06d000)
	librcpputils.so => /opt/ros/humble/lib/librcpputils.so (0x00007f787c05f000)
	librosidl_typesupport_c.so => /opt/ros/humble/lib/librosidl_typesupport_c.so (0x00007f787c059000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f787c9cb000)
	libspdlog.so.1 => /lib/x86_64-linux-gnu/libspdlog.so.1 (0x00007f787bd85000)
	libbuiltin_interfaces__rosidl_generator_c.so => /opt/ros/humble/lib/libbuiltin_interfaces__rosidl_generator_c.so (0x00007f787c052000)
	libfmt.so.8 => /lib/x86_64-linux-gnu/libfmt.so.8 (0x00007f787c031000)
```

```
(base) mona@ada:~$ which -a python
/home/mona/anaconda3/bin/python

```
## Feature request

#### Feature description

#### Implementation considerations

monajalal, 2023-11-01:
also

```
(base) mona@ada:~$ which -a python
/home/mona/anaconda3/bin/python
(base) mona@ada:~$ source ~/.bashrc
(base) mona@ada:~$ mkdir -p rclpy_ws/src
(base) mona@ada:~$ cd rclpy_ws/src/
(base) mona@ada:~/rclpy_ws/src$ git clone https://github.com/ros2/rclpy.git
Cloning into 'rclpy'...
remote: Enumerating objects: 7566, done.
remote: Counting objects: 100% (515/515), done.
remote: Compressing objects: 100% (173/173), done.
remote: Total 7566 (delta 347), reused 478 (delta 327), pack-reused 7051
Receiving objects: 100% (7566/7566), 2.80 MiB | 5.05 MiB/s, done.
Resolving deltas: 100% (5350/5350), done.
(base) mona@ada:~/rclpy_ws/src$ cd ..
(base) mona@ada:~/rclpy_ws$ colcon build --symlink-install
[0.936s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'rclpy' is in: /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding rclpy

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> rclpy   
--- stderr: rclpy                               
In file included from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/type_description_service.hpp:25,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/type_description_service.cpp:17:
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/service.hpp:21:10: fatal error: rcl/service_introspection.h: No such file or directory
   21 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/client.cpp:20:10: fatal error: rcl/service_introspection.h: No such file or directory
   20 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/service.cpp:19:10: fatal error: rcl/service_introspection.h: No such file or directory
   19 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:146: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/client.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:482: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/type_description_service.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:384: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/service.cpp.o] Error 1
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/_rclpy_pybind11.cpp:18:10: fatal error: rcl/service_introspection.h: No such file or directory
   18 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:90: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/_rclpy_pybind11.cpp.o] Error 1
In file included from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/wait_set.hpp:25,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/action_client.hpp:27,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/action_client.cpp:27:
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/client.hpp:21:10: fatal error: rcl/service_introspection.h: No such file or directory
   21 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:104: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/action_client.cpp.o] Error 1
In file included from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/wait_set.hpp:25,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/action_server.hpp:28,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/action_server.cpp:31:
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/client.hpp:21:10: fatal error: rcl/service_introspection.h: No such file or directory
   21 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/wait_set.hpp:25,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/action_server.hpp:28,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/action_goal_handle.hpp:25,
[... 186 lines cut ...]
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:160: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/clock.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:440: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/subscription.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:328: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/publisher.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:342: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/qos.cpp.o] Error 1
[... cut ...]
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:314: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/node.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:356: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/event_handle.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:212: CMakeFiles/_rclpy_pybind11.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< rclpy [20.0s, exited with code 2]

Summary: 0 packages finished [20.6s]
  1 package failed: rclpy
  1 package had stderr output: rclpy

```

monajalal, 2023-11-01:
```
(base) mona@ada:~/rclpy_ws$ colcon build --symlink-install --allow-overriding rclpy
Starting >>> rclpy   
--- stderr: rclpy                             
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/service.cpp:19:10: fatal error: rcl/service_introspection.h: No such file or directory
   19 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:384: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/service.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/service_introspection.cpp:16:10: fatal error: rcl/service_introspection.h: No such file or directory
   16 | #include "rcl/service_introspection.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/_rclpy_pybind11.cpp:18:10: fatal error: rcl/service_introspection.h: No such file or directory
   18 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:412: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/service_introspection.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:90: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/_rclpy_pybind11.cpp.o] Error 1
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/client.cpp:20:10: fatal error: rcl/service_introspection.h: No such file or directory
   20 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/wait_set.hpp:25,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/action_server.hpp:28,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/action_goal_handle.hpp:25,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/action_goal_handle.cpp:24:
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/client.hpp:21:10: fatal error: rcl/service_introspection.h: No such file or directory
   21 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:146: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/client.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:118: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/action_goal_handle.cpp.o] Error 1
In file included from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/type_description_service.hpp:25,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/type_description_service.cpp:17:
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/service.hpp:21:10: fatal error: rcl/service_introspection.h: No such file or directory
   21 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:482: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/type_description_service.cpp.o] Error 1
In file included from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/wait_set.hpp:25,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/action_client.hpp:27,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/action_client.cpp:27:
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/client.hpp:21:10: fatal error: rcl/service_introspection.h: No such file or directory
   21 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/wait_set.hpp:25,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/action_server.hpp:28,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/action_server.cpp:31:
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/client.hpp:21:10: fatal error: rcl/service_introspection.h: No such file or directory
   21 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/wait_set.hpp:25,
                 from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/wait_set.cpp:27:
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/client.hpp:21:10: fatal error: rcl/service_introspection.h: No such file or directory
   21 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:132: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/action_server.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:104: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/action_client.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:510: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/wait_set.cpp.o] Error 1
In file included from /home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/lifecycle.cpp:41:
/home/mona/rclpy_ws/src/rclpy/rclpy/src/rclpy/service.hpp:21:10: fatal error: rcl/service_introspection.h: No such file or directory
   21 | #include <rcl/service_introspection.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:272: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/lifecycle.cpp.o] Error 1
[... 192 lines cut ...]
[... cut ...]
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:356: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/event_handle.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:314: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/node.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:212: CMakeFiles/_rclpy_pybind11.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< rclpy [2.73s, exited with code 2]

Summary: 0 packages finished [3.34s]
  1 package failed: rclpy
  1 package had stderr output: rclpy

```

monajalal, 2023-11-01:
comment:
I do have `python3.10` however `which -a python` didn't show it.

monajalal, 2023-11-01:
seems fixed, I'll report if not

```
(base) mona@ada:~/rosbag2video$ /usr/bin/python3.10 ros2bag2video.py ~/Downloads/cheezit_1280_whole.bag 
FPS (int) =  25
Rate (float) =  1.0
Topic (str) =  
Output File (str) =  output.mp4
Verbose (bool) =  False
Reading info from bag file: /home/mona/Downloads/cheezit_1280_whole.bag
[ERROR] [1698850707.416016253] [rosbag2_storage]: No storage id specified, and no plugin found that could open URI
No plugin detected that could open file /home/mona/Downloads/cheezit_1280_whole.bag
ROS Message name =  
Image count =  0
msgtype =  None
AJB: subscribing to msg:  None on topic:  
Traceback (most recent call last):
  File "/home/mona/rosbag2video/ros2bag2video.py", line 430, in <module>
    main(sys.argv)
  File "/home/mona/rosbag2video/ros2bag2video.py", line 420, in main
    videowriter = RosVideoWriter(args)
  File "/home/mona/rosbag2video/ros2bag2video.py", line 142, in __init__
    self.subscription = self.create_subscription(
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py", line 1356, in create_subscription
    raise ex from None
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py", line 1354, in create_subscription
    self._validate_topic_or_service_name(topic)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py", line 1183, in _validate_topic_or_service_name
    validate_topic_name(topic_or_service_name, is_service=is_service)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/validate_topic_name.py", line 43, in validate_topic_name
    raise InvalidTopicNameException(name, error_msg, invalid_index)
rclpy.exceptions.InvalidTopicNameException: Invalid topic name: topic name must not be empty string:
  ''
   ^

```

closed 2023-11-01 by monajalal

Shunying, 2023-11-10:
Hi monajalal, I got stuck on the same issue for more than one week but in my case I need a Python3.8+ROS2 Humble, how did you solve that at last? 
1. Did you run the resdep for system dependency handling, it seems that it would install couple of python3-xxx apt binary libs, which are binded to certain os version and python3 version
2. Did you recompile the pybind11 with specific python3 version?

ashfaque1329, 2023-11-27:
Hi Monajalal,

Could you please elaborate on your fix please ? Thanks.

estradavg, 2023-12-23:
I had the same problem and I solved it by simply installing the following package:

`$ sudo apt-get install python-is-python3`

Jzow, 2024-03-18:
> > 请问已经解决了吗？我也遇到过类似的问题
> 
> 由于第三方ROS2包依赖于系统Python，您可以
> 
> 1. 使用与特定 Ubuntu 版本绑定的默认 Python 版本，不要从源代码构建不同的 Python 版本（推荐）
> 2. 从源代码构建您首选的 Python 版本，但用于`python-is-python3`强制它成为默认系统 Python

Hi, I installed ROS 2 on my local Windows 11 system. What do I need to do, Thanks

Jzow, 2024-03-19:
My Python version is 3.8, but he still reports that the module cannot be found. I developed it on Windows 11 system. Do have anyone who has encountered this problem? Can share how to solve it? Thanks

```python
ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'

The C extension 'C:\ros2-windows\Lib\site-packages\_rclpy_pybind11.pyd' isn't present on the system. Please refer to 'https://docs.ros.org/en/iron/Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system' for possible solutions
```

```shell
C:\ros2-windows>python --version
Python 3.8.3
```

Oh-JongJin, 2024-04-16:
If you use Anaconda environment, deactivate Anaconda and use default Ubuntu Python.

```bash
(base) user:~$ python
>>> import rclpy
Trackback (most recent call last):
	...
ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'
The C extension ~~~ for possible solutions
# ----------------------------------------------- #
(base) user:~$ conda deactivate
user:$ python
>>> import rclpy
>>>

```