ros2/rclpy #1195: Exception when instantiating ActionServer as part of ROS package tests

sea-bass, 2023-11-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Latest binaries of Humble/Iron/Rolling
- DDS implementation:
  - Fast DDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
I have a test that is failing as part of this PR (https://github.com/RobotWebTools/rosbridge_suite/pull/886) when I try to instantiate an action server, noting that:

* This only happens on Iron and Rolling, but not on Humble
* This only happens when I run using `colcon test` (which in turn uses `ament_add_pytest_test`), but not if I run it directly using Python

Can't seem to figure out why this would be the case, so figured I'd elevate it here? Thank you!

The code:
```
class MyClass:
    def __init__(self, action_name, action_type
    self.action_server = ActionServer(
            protocol.node_handle,  # Just an rclpy.Node
            get_action_class(action_type),  # example_interfaces.action.Fibonacci
            action_name,  # /fibonacci_action_1
            self.execute_callback,
            callback_group=ReentrantCallbackGroup(),  # https://github.com/ros2/rclpy/issues/834#issuecomment-961331870
        )
```

The error:
```
          self.action_server = ActionServer(
      /opt/ros/rolling/lib/python3.10/site-packages/rclpy/action/server.py:264: in __init__
          self._logger = self._node.get_logger().get_child('action_server')
      _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
      
      self = <rclpy.impl.rcutils_logger.RcutilsLogger object at 0x7f730bfe9270>
      name = 'action_server'
      
          def get_child(self, name):
              if not name:
                  raise ValueError('Child logger name must not be empty.')
          
              if self.name:
                  # Prepend the name of this logger
      >           _rclpy.rclpy_logging_rosout_add_sublogger(self.name, name)
      E           RuntimeError: Failed to call rcl_logging_rosout_add_sublogger
```

#### Expected behavior
Should not raise an exception.

#### Actual behavior
Raises an exception.

sea-bass, 2023-11-01:
Seems the issue originates from https://github.com/ros2/rclpy/pull/982

fujitatomoya, 2023-11-03:
@iuhilnehc-ynos can you take a look at this?

closed 2023-11-15 by clalancette