ros2/rclpy #1197: CMake Error (Non-existent Path) When Building From Source

jusevitch, 2023-11-11:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04
- Installation type:
  - Source
- Version or commit hash:
  - b4b4702ecffd923548eadac6d35f3d90d2eb6f91
- DDS implementation:
  - Default for ROS2 Humble
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

I followed the instructions on the README page. More specifically:

* Source your ROS 2 installation (ROS 2 Humble, built from source)
* Build code:

```
mkdir -p rclpy_ws/src
cd rclpy_ws/src
git clone https://github.com/ros2/rclpy.git
cd ..
colcon build --symlink-install

```

#### Expected behavior

I expected the package to build.

#### Actual behavior

The following error is thrown:

```
--- stderr: rclpy                                                                                                                                                                                          
CMake Error in CMakeLists.txt:                                                                                                                                                                             
  Imported target "pybind11::embed" includes non-existent path                                                                                                                                             
                                                                                                                                                                                                           
    "/include"                                                                                                                                                                                             
                                                                                                                                                                                                           
  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:                                                                                                                                         

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "pybind11::embed" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< rclpy [23.6s, exited with code 1]

```

clalancette, 2023-11-11:
> git clone https://github.com/ros2/rclpy.git

At the very least, you'll definitely need to use the `humble` branch:

```
git clone https://github.com/ros2/rclpy.git -b humble
```

jusevitch, 2023-11-11:
Thanks for the correction. I get the same error after switching to the `humble` branch. Will edit the original post.

Incidentally, this error also arises when following the instructions for [building ROS2 Humble from source][1], which I believe automatically checks out the `humble` branch of `rclpy`.

[1]: https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html

clalancette, 2023-11-11:
My best guess is that something is off with your installation of `pybind11`.  How have you installed it?

jusevitch, 2023-11-11:
The issue turned out to be an incorrect assumption about Ubuntu 20.04 symlinks, although it's not clear what software package is responsible. A solution can be found [here][1].

In short, the problem was solved by creating a symlink `/include` pointing to `/usr/include`. After doing this, the package built without errors.

[1]: https://github.com/VowpalWabbit/vowpal_wabbit/issues/3003#issuecomment-846338259

closed 2023-11-11 by jusevitch

wjwwood, 2024-12-12:
For what it's worth, I finally found that this issue was triggered for me due to a bad PATH value. I found that if I had `/bin` before `/usr/bin` on my path (I had a rogue `.bashrc` line that was adding `/bin` to the front incorrectly) then I ran into this issue.

Symlinking `/usr/include` -> `/include` I think is just treating the symptom. Fixing the path seems better and doesn't require sudo, but ultimately I think that the issue is that the pybind11 cmake config file is not consider the possibility of symbolic links in it's install path, though it's not really pybind11's fault (this also affects boost::python) and instead I think it's in the `@PACKAGE_INIT@`, specifically a line like this (pulled from my local copy):

```
get_filename_component(PACKAGE_PREFIX_DIR "${CMAKE_CURRENT_LIST_DIR}/../../../" ABSOLUTE)
```

So, in summary, I don't know where the "right" fix is either, but I think fixing up the PATH is easier/better than putting a symlink in that requires sudo.

I'd be interested to know if this fixes things for others.