ros2/rclpy #1200: ReadtheDocs page is missing a lot of documentation for API.

guzman109, 2023-11-16:
No Node API's found or even a link where to find them.

clalancette, 2023-11-16:
Sorry, can you be more clear about what link you are looking at?

guzman109, 2023-11-16:
I went to the one for iron.
https://docs.ros.org/en/iron/p/rclpy/api/node.html

guzman109, 2023-11-16:
https://docs.ros.org/en/iron/API-Docs.html

Also, there was no easy way of finding the python docs either, since they were not posted on here.

clalancette, 2023-11-17:
> [](https://docs.ros.org/en/iron/API-Docs.html)
> 
> Also, there was no easy way of finding the python docs either, since they were not posted on here.

That page is not meant to be a listing of all documentation.  The correct way to find the API docs is through https://index.ros.org, and doing a search for the package you are interested in.  In this case, it would lead to https://index.ros.org/p/rclpy/github-ros2-rclpy/#iron, and there is a link to `API Docs` there.

sloretz, 2023-11-17:
Following the links from index.ros.org I get to the same page https://docs.ros.org/en/iron/p/rclpy/api/node.html, which seems to have no content for the Node class.

Comparing a page that does have content with one that doesn't work; the content looks about the same to me:

* https://github.com/ros2/rclpy/blob/iron/rclpy/docs/source/api/timers.rst
* https://github.com/ros2/rclpy/blob/iron/rclpy/docs/source/api/node.rst

Seems like the `rclpy.node` module (and others) fails to be imported in the doc job

https://build.ros2.org/job/Idoc__rclpy__ubuntu_jammy_amd64/11/console

```
03:53:33 WARNING: Failed to import rclpy.node.
03:53:33 Possible hints:
03:53:33 * AttributeError: module 'rclpy' has no attribute 'node'
03:53:33 * ModuleNotFoundError: No module named 'type_description_interfaces'
```

dghw, 2024-07-19:
This issue is still happening today (e.g. at http://docs.ros.org/en/jazzy/p/rclpy/ most entries are pretty much empty). 

Building the docs locally from source as described in the package readme worked fine.

fujitatomoya, 2024-07-21:
@dghw thanks for checking on this, i also confirmed that local build just works. (many warnings came up though...)

[image]

@sloretz i have no idea how often this doc CI runs, do you have any idea about this documentation build CI? CC: @clalancette

sloretz, 2024-08-09:
Linking https://github.com/ros-infrastructure/rosdoc2/issues/68