ros2/rclpy #1207: Failed to send request: cannot publish data

rty813, 2023-12-25:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from source
- Version or commit hash:
  - Foxy
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
I couldn't find a consistently reproducible path. It has only occurred once so far, where an exception occurred abruptly during the program execution, leading to an unexpected termination.

#### Expected behavior
no exception

#### Actual behavior
```
Exception in thread Thread-26:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/rclpy/executors.py", line 290, in spin
    self.spin_once()
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/rclpy/executors.py", line 770, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/rclpy/executors.py", line 767, in _spin_once_impl
    future.result()
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/rclpy/task.py", line 94, in result
    raise self.exception()
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/rclpy/task.py", line 239, in __call__
    self._handler.send(None)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/rclpy/executors.py", line 429, in handler
    await call_coroutine(entity, arg)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/rclpy/executors.py", line 384, in _execute_service
    srv.send_response(response, header)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/rclpy/service.py", line 74, in send_response
    _rclpy.rclpy_send_response(capsule, response, header)
_rclpy.RCLError: Failed to send request: cannot publish data, at /opt/ros/foxy/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:3605, at /opt/ros/foxy/src/ros2/rcl/rcl/src/rcl/service.c:356
```

fujitatomoya, 2023-12-26:
@rty813 unfortunately Foxy is already End Of Life. could you try to use Humble or later to see if this problem still happens?

note, this could be related to https://github.com/ros2/rclpy/pull/1136.

rty813, 2023-12-27:
@fujitatomoya Thanks, I'll test #1136 on foxy.

rty813, 2023-12-27:
Assuming this is the issue, why is a timeout occurring, how can it be avoided, and is it possible to increase the timeout for CycloneDDS?

fujitatomoya, 2024-01-02:
> Assuming this is the issue, why is a timeout occurring

see https://github.com/ros2/ros2/issues/1253#issuecomment-1101241234, it is well explained why this could happen.

> how can it be avoided

https://github.com/ros2/rmw_fastrtps/pull/704 is to extend the timeout for https://github.com/ros2/rmw_fastrtps. you can see how to set the configuration XML too. Note, again that https://github.com/ros2/rmw_fastrtps/pull/704 is NOT available in foxy. (Foxy is E.O.L)

> is it possible to increase the timeout for CycloneDDS?

that i do not know.

> Once a timeout occurs, what kind of result will the synchronous calling client receive, or will it remain blocked indefinitely?

if the timeout is not set, it will wait forever. so synchronous calling client is not recommended w/o timeout.
since client does not know what is going on for service server side, i would do

1. check service is available or waits for the service online.
2. call async request.
3. wait for certain timeout. (depends on application)
4. if timeouts, call `remove_pending_request`. back to sequence 1.

fujitatomoya, 2024-01-02:
@rty813 as i mentioned before, Foxy is already E.O.L. if we do not have specific requirements to fix the `rclpy` in mainline, i would like to close this issue for now. if you have any further questions, you can reach out to https://robotics.stackexchange.com/, that is the central QA site.

closed 2024-01-02 by fujitatomoya