ros2/rclpy #1221: xmlrpc.client.Fault: <Fault 1: "<class 'RuntimeError'>:!rclpy.ok()">

ionutnechita-intel, 2024-02-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 LTS IOT Desktop Variant
- Installation type:
  - binaries
- Version or commit hash:
  -  3.3.11-1jammy.20240125.223153
- DDS implementation:
  - default
- Client library (if applicable):
  - rclpy - daemon

#### Steps to reproduce issue
# ros2 topic list - executed command in terminal

#### Expected behavior
View all ROS2 topic in terminal

#### Actual behavior
I have issue with ROS2 command:

```
Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.8', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 91, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/verb/list.py", line 56, in main
    topic_names_and_types = get_topic_names_and_types(
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2topic/api/__init__.py", line 41, in get_topic_names_and_types
    topic_names_and_types = node.get_topic_names_and_types()
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1122, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1464, in __request
    response = self.__transport.request(
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1166, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1182, in single_request
    return self.parse_response(resp)
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1354, in parse_response
    return u.close()
  File "/usr/lib/python3.10/xmlrpc/client.py", line 668, in close
    raise Fault(**self._stack[0])
xmlrpc.client.Fault: <Fault 1: "<class 'RuntimeError'>:!rclpy.ok()">

```

```
dpkg -l | grep humble | grep rcl
ii  ros-humble-parameter-traits                       0.3.7-1jammy.20240125.233218                          amd64        Functions and types for rclcpp::Parameter
ii  ros-humble-rcl                                    5.3.7-1jammy.20240125.221432                          amd64        The ROS client library common implementation.
ii  ros-humble-rcl-action                             5.3.7-1jammy.20240125.222649                          amd64        Package containing a C-based ROS action implementation
ii  ros-humble-rcl-interfaces                         1.2.1-1jammy.20240125.214810                          amd64        The ROS client library common interfaces.
ii  ros-humble-rcl-lifecycle                          5.3.7-1jammy.20240125.222651                          amd64        Package containing a C-based lifecycle implementation
ii  ros-humble-rcl-logging-interface                  2.3.1-1jammy.20240125.211226                          amd64        Interface that rcl_logging backends needs to implement.
ii  ros-humble-rcl-logging-spdlog                     2.3.1-1jammy.20240125.211547                          amd64        Implementation of rcl_logging API for an spdlog backend.
ii  ros-humble-rcl-yaml-param-parser                  5.3.7-1jammy.20240125.212054                          amd64        Parse a YAML parameter file and populate the C data structure.
ii  ros-humble-rclcpp                                 16.0.8-1jammy.20240125.223158                         amd64        The ROS client library in C++.
ii  ros-humble-rclcpp-action                          16.0.8-1jammy.20240125.234425                         amd64        Adds action APIs for C++.
ii  ros-humble-rclcpp-components                      16.0.8-1jammy.20240125.234442                         amd64        Package containing tools for dynamically loadable components
ii  ros-humble-rclcpp-lifecycle                       16.0.8-1jammy.20240125.234555                         amd64        Package containing a prototype for lifecycle implementation
ii  ros-humble-rclpy                                  3.3.11-1jammy.20240125.223153                         amd64        Package containing the Python client.

```

aravind-reddy, 2024-02-20:
have you sourced the setup.bash file, if not please source it using source /opt/ros/humble/setup.bash

ionutnechita-intel, 2024-02-20:
I have source file loaded.

fujitatomoya, 2024-02-20:
according to the stack, it looks like client receives the Fault from the server. can you restart the daemon to see if anything changes?

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 daemon stop
The daemon is not running
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 daemon start
The daemon has been started
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 daemon status
The daemon is running
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic list
/parameter_events
/rosout
```

ionutnechita-intel, 2024-02-21:
Hi @fujitatomoya ,

With restart daemon, this issue is not reproduced now.

fujitatomoya, 2024-02-25:
@ionutnechita-intel thanks for checking. i will go ahead to close this for now. please reopen or create another if you meet the problem. i think we need to keep the reproducible environment to debug the ros2 daemon process in next time.

closed 2024-02-25 by fujitatomoya