ros2/rclpy #1223: Timer hanging and high CPU load when using MultiThreadedExecutor

JasperTan97, 2024-02-22:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04

- Installation type: Binaries

- Version or commit hash: Iron

- DDS implementation: eProsima’s Fast DDS (the default)

- Client library (if applicable): rclpy

#### Steps to reproduce issue

My publisher:
``` python
import rclpy
from rclpy.node import Node
from std_msgs.msg import String
import time

class MyPublisher(Node):
    def __init__(self):
        super().__init__('my_publisher')
        self.publisher_ = self.create_publisher(String, 'my_topic',  10 ) 
        timer_period =  1 / 500  
        self.timer = self.create_timer(timer_period, self.timer_callback)

    def timer_callback(self):
        msg = String()
        # self.get_logger().info('Publishing: "%s"' % msg.data)
        self.publisher_.publish(msg)
```
my subscriber:
``` python
import rclpy
from rclpy.node import Node
from std_msgs.msg import String

class MySubscriber(Node):
    def __init__(self):
        super().__init__('my_subscriber')
        self.subscription = self.create_subscription(
            String,
            'my_topic',
            self.listener_callback,
            10)
        self.subscription  # prevent unused variable warning

    def listener_callback(self, msg):
        pass
```
And my main function:
``` python

import rclpy
from my_multithreaded_package.publisher import MyPublisher
from my_multithreaded_package.subscriber import MySubscriber
import time

def main(args=None):
    rclpy.init(args=args)

    publisher = MyPublisher()
    subscriber = MySubscriber()

    executor = rclpy.executors.MultiThreadedExecutor(4)
    executor.add_node(publisher)
    executor.add_node(subscriber)
    executor.spin()

    # Shutdown the ROS client library
    rclpy.shutdown()

if __name__ == '__main__':
    main()
 ```

#### Expected behavior (which is what I get when using the SingleThreadedExecutor)
``` console
ros2 topic hz /my_topic
average rate: 499.945
        min: 0.002s max: 0.002s std dev: 0.00007s window: 501
average rate: 499.965
        min: 0.001s max: 0.002s std dev: 0.00008s window: 1001
average rate: 499.995
        min: 0.001s max: 0.003s std dev: 0.00009s window: 1502
average rate: 499.981
        min: 0.001s max: 0.003s std dev: 0.00009s window: 2002
average rate: 499.978
        min: 0.001s max: 0.003s std dev: 0.00009s window: 2502
average rate: 500.004
        min: 0.001s max: 0.003s std dev: 0.00009s window: 3003
average rate: 499.989
        min: 0.001s max: 0.003s std dev: 0.00010s window: 3503
  ```

#### Actual behavior
``` console
ros2 topic hz /my_topic
average rate: 25.933
        min: 0.002s max: 0.919s std dev: 0.16741s window: 29
average rate: 14.580
        min: 0.000s max: 0.937s std dev: 0.21933s window: 41
average rate: 13.928
        min: 0.000s max: 0.937s std dev: 0.21447s window: 62
average rate: 12.983
        min: 0.000s max: 0.937s std dev: 0.22058s window: 81
average rate: 10.575
        min: 0.000s max: 0.937s std dev: 0.24748s window: 85
average rate: 9.774
        min: 0.000s max: 0.937s std dev: 0.25675s window: 89
average rate: 9.845
        min: 0.000s max: 0.937s std dev: 0.25056s window: 102
average rate: 9.399
```

#### Additional information

So similar issues have been brought up with rclcpp, but I have not seen any comments made about rclpy. I found an issue here:https://github.com/ros2/rclcpp/issues/1487, with other people also reporting something like: https://github.com/ros2/rclcpp/issues/1618 and the fix is https://github.com/ros2/rclcpp/pull/1516 and then https://github.com/ros2/rclcpp/pull/1692.

Aside from the timer callback hanging (I assume it to be with after following some tic toc), my CPU load becomes really high using the MultiThreadedExecutor, while the SingleThreadedExecutor does not cause any noticeable CPU load. I have also tried using both the `MutuallyExclusiveCallbackGroup` and `ReentrantCallbackGroup` with no change in behaviour.

I am not sure if my QOS settings are the problem, or this is an issue intrinsic to python (because of GIL or etc.) but either a more suitable example for how to use the MultiThreadedExecutor could be provided (if my usage is wrong), or the ROS wiki pages should reflect that this significant problem exists (if no fix is possible).

Thank you for helping!

wjwwood, 2024-02-29:
From our weekly issue triage meeting:

Assigning @clalancette so he can make a high level tracking issue about known issues with Python's executors.

Also, we think this issue will not make progress without staffing of an engineer, which we don't currently have, or a dedicated community member to investigate and make a good suggestion on what to change and why. This is because executor related issues tend to be very nuanced and complicated to work on. So we're also assigning this the "help wanted" label with that in mind.

KKSTB, 2024-04-16:
Hi @JasperTan97 I recommend you could run py-spy to see where the high CPU load comes from. This helps immensely. Thanks

JasperTan97, 2024-04-22:
[image]

@KKSTB, it looks like the waiting for ready callbacks are the issue. Do you get the same on your end?

KKSTB, 2024-04-23:
@JasperTan97 i ran py-spy on my ubuntu ROS iron machine and got this result:
[image]

And `py-spy dump` to see what threads they are in the flame graph:
```
Thread 11386 (idle): "MainThread"
    handle (rclpy/guard_condition.py:38)
    _wait_for_ready_callbacks (rclpy/executors.py:582)
    wait_for_ready_callbacks (rclpy/executors.py:723)
    _spin_once_impl (rclpy/executors.py:794)
    spin_once (rclpy/executors.py:813)
    spin (rclpy/executors.py:295)
    main (xavest_gui/main.py:44)
    <module> (xavest_gui:33)
Thread 11398 (idle): "ThreadPoolExecutor-0_0"
    _worker (concurrent/futures/thread.py:81)
    run (threading.py:953)
    _bootstrap_inner (threading.py:1016)
    _bootstrap (threading.py:973)
Thread 11399 (idle): "ThreadPoolExecutor-0_1"
    _worker (concurrent/futures/thread.py:81)
    run (threading.py:953)
    _bootstrap_inner (threading.py:1016)
    _bootstrap (threading.py:973)
Thread 11400 (idle): "ThreadPoolExecutor-0_2"
    _worker (concurrent/futures/thread.py:81)
    run (threading.py:953)
    _bootstrap_inner (threading.py:1016)
    _bootstrap (threading.py:973)
Thread 11401 (idle): "ThreadPoolExecutor-0_3"
    _worker (concurrent/futures/thread.py:81)
    run (threading.py:953)
    _bootstrap_inner (threading.py:1016)
    _bootstrap (threading.py:973)
```

This shows the 4 worker threads were mostly idle, which makes sense because there is basically nothing to do inside subscriber and timer node.

The main thread was instead very very busy retrieving tasks for the 4 worker threads to do. It seems such workload of retrieving and distributing tasks and gathering results at high frequency (500Hz) is marginal for one core. Therefore the rate slows down considerably. Although I have no clue why your single core is much slower (I can achieve 3XX-4XX Hz on my i7-9750H).

As for single threaded executor, I can achieve 500Hz. The CPU utilization was half of multi threaded case. I have to push to 2500Hz before the rate starts to drop.

I believe the problem has to do with the efficiency of transferring tasks from main thread to worker threads, relative to the actual useful tasks done in the worker thread.

MatthijsBurgh, 2024-05-16:
I experience the same issue, +1

MatthijsBurgh, 2024-05-17:
SVG: https://gist.githubusercontent.com/MatthijsBurgh/a5ee1d710e0ff4b6524b279a2688e8be/raw/15da19a2ca8ca0d1e28bfeb46e9726205357a855/profile.svg

But it doesn't show the files and line numbers correct. Therefore I have included another screenshot, which does show this. You should check the linenumbers in this [revision of `executors.py`](https://github.com/ros2/rclpy/blob/7.0.1/rclpy/rclpy/executors.py)

[image]

clalancette, 2024-05-17:
> _update_: I have looked at the code a bit, `_wait_for_ready_callbacks` just looks very inefficient. It constructs many objects and lists. The lists are appended and then iterate over again, etc. The [`rclcpp::Executor`](https://github.com/ros2/rclcpp/blob/rolling/rclcpp/src/rclcpp/executor.cpp), looks more efficient.
> (It is complex code, so if you feel I am wrong, let me know.)

Yep, you are correct.  And this flame graph is the kind of thing I was waiting to see to confirm.  So thanks for spending the time to do that.

We spent a bunch of time over the last year optimizing the `rclcpp` executor to make things faster; the `rclpy` one needs a similar effort to become more performant.

MatthijsBurgh, 2024-05-17:
@clalancette just let me know whether you need a more detailed graph, etc.

clalancette, 2024-05-17:
> @clalancette just let me know whether you need a more detailed graph, etc.

To be totally transparent; we don't (currently) have plans to work on this.  While that could change in the future, if it is something you are interested in, I'd suggest trying to make some improvements and opening PRs.  We are more than happy to review them.

MatthijsBurgh, 2024-05-19:
@clalancette I don't have all the knowledge to fix this. But I do think this does require to be fixed. As it is baked into many ROS applications, i.e. rqt. But it works much worse than the SingleThreadedExecutor.

I want to help, though I am not able to figure out the design of it. It is very complex. So it would be helpful of someone could explain with enough detail what should be happening. Then me and other people could check the code.

KKSTB, 2024-05-19:
@MatthijsBurgh can you tell what frequency you can achieve? I can achieve 2xx-3xx hz on i7 9750H (9th gen Intel). But @JasperTan97 only managed 2x hz on i7 1355U (13th gen Intel). I think the problem lies on why there's such a big difference.

MatthijsBurgh, 2024-05-20:
SingleThreadedExecutor: 500Hz
MultiThreadedExecutor(4): 60Hz
MultiThreadedExecutor(20): 50Hz
MultiThreadedExecutor(): ~5Hz
i7-13700H

It is really weird newer hardware is much slower. But still, the MultiThreadedExectuor, should only be slower with very few threads.

KKSTB, 2024-05-23:
I have no idea. Maybe try these:
1. I suspect its network problem causing a lot of traffic errors. Try to disconnect all networks, or use cyclone DDS and set domain 0 to local only
2. Test on a fresh unmodified Ubuntu machine (no realtime kernel) and see if there is any improvement

Commands to use CycloneDDS:
```
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
export CYCLONEDDS_URI=file:///home/user/ros2_ws/ddsconfig.xml
```
and ddsconfig.xml:
```
<?xml version="1.0" encoding="UTF-8" ?>
<CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
  <Domain id="0">
    <General>
      <Interfaces>
        <NetworkInterface name="lo"/>
      </Interfaces>
    </General>
  </Domain>
</CycloneDDS>
```

MatthijsBurgh, 2024-05-23:
@KKSTB Using cyclone did fix the issue for me. All MultiThreaded configurations now are able to reach 500Hz.

Now I am still wondering how a bad network/RMW results in the distributing of task to the worker threads being very slow.

KKSTB, 2024-05-24:
My plain guess of reasons:
1. Nodes in SingleThreadedExecutor doesn't need to communicate across threads, so there is no network traffic (I am not sure if this is true at all)
2. When running in MultiThreadedExecutor, network traffic is generated to communicate across threads (I am not sure as well). If there is network traffic collision then a lot of errors are generated, causing high CPU load. Traffic can collide when there is other application using the same port (https://docs.ros.org/en/rolling/Concepts/Intermediate/About-Domain-ID.html) or there is an incompatible ROS2 node

MatthijsBurgh, 2024-05-24:
I have no clue, I can only observe there is a big difference in distribution of the calls. I have collected them with for `cyclone` and `fastrtps`. Both with the `ReentrantCallbackGroup` and the `MutuallyExclusiveCallbackGroup`, these have the `_mutually` suffix. Both in a flamegraph SVG but also in speedscope JSON format. Which you need in import on [https://www.speedscope.app](https://www.speedscope.app/).

The SVG is not a result of the same execution as the speedscope. I have run all combination such that have around 30.000 samples.

The results can be found at https://gist.github.com/MatthijsBurgh/cd7871c6597c4cdc196526b658693a18

KKSTB, 2024-05-28:
@MatthijsBurgh thank you for your analysis. In fact py-spy has an -n option to also sample non-python libraries to get a better look of where takes the most CPU time, similar to the record in https://github.com/ros2/rclpy/issues/1223#issuecomment-2071293729 where rcl_wait and rmw_wait can be seen in the graph. But this option is very slow so the py-spy sampling frequency should be lowered too. 

But I think even if you rerun py-spy with this option, it is difficult to look into a C++ problem from py-spy. Probably you can try setting ROS2 log level to debug, i.e. `--ros-args --log-level debug`, and see what happen. Thanks.

astroseger, 2024-06-25:
I seem to have a related issue with heavy CPU usage (100%), even when running the sender and receiver in separate processes. In one program, I run a sender that sends strings at a frequency of 10 Hz. In another program, I run a receiver with a `MultiThreadedExecutor`, which manages a single node.

Initially, I used the `MultiThreadedExecutor` to spin a single node due to this issue #1133  . However, it appears that the issue #1133  can be avoided by using the `SingleThreadedExecutor`, which does not cause any performance problems.

KKSTB, 2024-06-27:
@astroseger are you using fastDDS? Would you also try `MultiThreadedExecutor` with cycloneDDS to see if there is any difference in CPU load? Thanks.

astroseger, 2024-06-27:
> @astroseger are you using fastDDS? Would you also try `MultiThreadedExecutor` with cycloneDDS to see if there is any difference in CPU load? Thanks.

yes. I was using fastrtps (default in ros:jazzy docker). With cycloneDDS I don't have this issue with 100% CPU usage. (I'm referring to the test where I run a sender in a separate process, which sends short strings at a frequency of 10Hz. Meanwhile, I run a receiver in another separate process using `MultiThreadedExecutor`)

KKSTB, 2024-07-08:
Thanks @astroseger. It seems im able to repeat the CPU high load issue when using fastrtps and separating pub and sub nodes. The sub node has unusually high CPU load. I ran the sub node with `--ros-args --log-level debug` and got something:

For comparison, cycloneDDS (normal):
```
[DEBUG] [1720407705.546314475] [rcl]: Initializing wait set with '1' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407705.645685839] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407705.645856729] [rcl]: Subscription taking message
[DEBUG] [1720407705.645882954] [rcl]: Subscription take succeeded: true
1720407705645147284
[DEBUG] [1720407705.646196664] [rcl]: Initializing wait set with '1' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407705.745888051] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407705.746095989] [rcl]: Subscription taking message
[DEBUG] [1720407705.746131258] [rcl]: Subscription take succeeded: true
1720407705745290717
[DEBUG] [1720407705.746658603] [rcl]: Initializing wait set with '1' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407705.746959277] [rcl]: Initializing wait set with '1' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407705.845609822] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407705.845773664] [rcl]: Subscription taking message
[DEBUG] [1720407705.845796741] [rcl]: Subscription take succeeded: true
1720407705845086190
[DEBUG] [1720407705.846099624] [rcl]: Initializing wait set with '1' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407705.945733864] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407705.945893749] [rcl]: Subscription taking message
[DEBUG] [1720407705.945918029] [rcl]: Subscription take succeeded: true
1720407705945149903
[DEBUG] [1720407705.946252046] [rcl]: Initializing wait set with '1' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407706.045476776] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407706.045610969] [rcl]: Subscription taking message
[DEBUG] [1720407706.045631707] [rcl]: Subscription take succeeded: true
1720407706045105652
```

FastRTPS multithreaded executor (high CPU load):
```
[DEBUG] [1720407503.914894042] [rcl]: Initializing wait set with '1' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.914984317] [rcl]: Initializing wait set with '1' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.949298022] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.949436931] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.949545225] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.949648604] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.949749716] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.949848268] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.949973682] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.950072082] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.950171214] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.950267957] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.950364975] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.950461963] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.950559496] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.950657420] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.950754699] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.950881678] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720407503.950980185] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[... cut ...]
[DEBUG] [1720408494.103437831] [rcl]: Initializing wait set with '1' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720408494.103626654] [rcl]: Initializing wait set with '1' subscriptions, '2' guard conditions, '0' timers, '0' clients, '7' services
[DEBUG] [1720408494.202752911] [rcl]: Subscription taking message
[DEBUG] [1720408494.202790116] [rcl]: Subscription take succeeded: true
1720408494202448064
```

So the wait on the wait set fails when it is fastrtps and in thread pool executor. Very strange.

fujitatomoya, 2025-10-28:
@pattylo no i do not think so. there has been some minor fixes in handling the task in MultiThreadedExecutor as it is being developed, but i do not think those are addressing the root cause of the performance issue reported here.

LostPatrol, 2026-10-21:
Jazzy still uses `NumberOfEntities.__add__` for each waitable during wait-set rebuilds, creating temporary count objects. Rolling already avoids these allocations through #1285.

I have a Jazzy candidate that accumulates the six fields directly in the executor-local counter, leaving public arithmetic methods unchanged. Would you prefer a backport of #1285 or this local adaptation?

The source-built candidate passed 59/59 CTest targets (527 Python and 4 C++ tests; ament's cppcheck 2.13 skip applies). Baseline tests passed after an official XML schema cache resolved a download failure.

A standalone single-threaded workload with 400 Hz input, 40/100 Hz timers and 16 event waitables averaged 22.83% → 20.24% of one core across eight ABBA samples, with approximately unchanged rates. This does not evaluate the multithreaded timer starvation issue. The PR can include the individual measurements and standalone reproducer.

OpenAI gpt-6.1-sol assisted with the patch, reproducer, and gpt-6-astra assisted with the review.

Candidate patch: https://github.com/LostPatrol/rclpy/commit/2f215ee7b467b1ef1a89cade4545d51fdc7a48df

nsherif-fugro, 2026-10-29:
Hello! Does ros2/rmw_fastrtps#930 fix this? I noticed the MultiThreadedExecutor hogging the GIL until python forced it to switch to a worker thread and traced it down to this issue.

In short, fastRTPS did not have a separate mechanism to distinguish incoming data events from subscription QoS events. As a result rmw_wait never waited but with nothing else to process the MultiThreadedExecutor kept rebuilding the same waitset over and over again... which in turn stopped the worker thread from ever picking up the GIL.

Disabling use_default_callbacks in SubscriptionEventCallbacks should also resolve the issue by disabling the event callbacks entirely, though this is just a bandaid or a quick test rather than a fix.

dlanov, 2026-10-29:
Thanks for connecting this issue to ros2/rmw_fastrtps#930.

The fix was merged into Rolling on October 26. We confirmed the underlying spurious QoS wakeup problem with regression tests, but haven't reproduced this specific rclpy MultiThreadedExecutor scenario.

Could someone experiencing the issue test against the updated Rolling Fast DDS implementation and confirm whether CPU usage and timer performance improve?

That would also help establish whether the fix should be backported to Jazzy.

nsherif-fugro, 2026-11-03:
I have created backports for https://github.com/ros2/rmw_fastrtps/pull/930 and https://github.com/ros2/rmw_fastrtps/pull/890 for jazzy and tested this using the same timer/publisher/subscriber system described above. I used both 500 Hz and 2000 Hz timer frequencies.

## Before the changes:

### MultiThreaded @ 500Hz

[subscriber]: rx_rate=30.4 Hz  process_cpu=105%
[subscriber]: rx_rate=27.3 Hz  process_cpu=105%
[subscriber]: rx_rate=31.6 Hz  process_cpu=105%
[subscriber]: rx_rate=29.0 Hz  process_cpu=105%

### MultiThreaded @ 2000Hz

[subscriber]: rx_rate=28.9 Hz  process_cpu=106%
[subscriber]: rx_rate=27.1 Hz  process_cpu=106%
[subscriber]: rx_rate=27.7 Hz  process_cpu=106%
[subscriber]: rx_rate=28.0 Hz  process_cpu=106%

### SingleThreaded @ 500Hz
[subscriber]: rx_rate=497.8 Hz  process_cpu=41%
[subscriber]: rx_rate=500.0 Hz  process_cpu=39%
[subscriber]: rx_rate=500.0 Hz  process_cpu=42%
[subscriber]: rx_rate=500.0 Hz  process_cpu=43%

### SingleThreaded @ 2000Hz
[subscriber]: rx_rate=1901.7 Hz  process_cpu=96%
[subscriber]: rx_rate=1969.6 Hz  process_cpu=96%
[subscriber]: rx_rate=1974.0 Hz  process_cpu=96%
[subscriber]: rx_rate=1980.2 Hz  process_cpu=95%

## After ros2/rmw_fastrtps/930 and ros2/rmw_fastrtps/890 backports:

### MultiThreaded @ 500Hz

[subscriber]: rx_rate=497.5 Hz  process_cpu=68%
[subscriber]: rx_rate=500.0 Hz  process_cpu=78%
[subscriber]: rx_rate=500.0 Hz  process_cpu=78%
[subscriber]: rx_rate=500.0 Hz  process_cpu=80%

### MultiThreaded @ 2000Hz

[subscriber]: rx_rate=1134.6 Hz  process_cpu=104%
[subscriber]: rx_rate=1139.0 Hz  process_cpu=104%
[subscriber]: rx_rate=1139.3 Hz  process_cpu=104%
[subscriber]: rx_rate=1147.2 Hz  process_cpu=104%

### SingleThreaded @ 500Hz
[subscriber]: rx_rate=497.8 Hz  process_cpu=42%
[subscriber]: rx_rate=500.0 Hz  process_cpu=46%
[subscriber]: rx_rate=500.0 Hz  process_cpu=41%
[subscriber]: rx_rate=500.0 Hz  process_cpu=44%

### SingleThreaded @ 2000Hz
[subscriber]: rx_rate=1977.3 Hz  process_cpu=95%
[subscriber]: rx_rate=1994.8 Hz  process_cpu=95%
[subscriber]: rx_rate=1985.2 Hz  process_cpu=95%
[subscriber]: rx_rate=1993.2 Hz  process_cpu=95%

## Conclusions

The patches certainly helped going from ~30 hz to ~1000 Hz a x33 improvement. 

The CPU utilization also improved hovering at ~80% but was almost double that of the SingleThreadedExecutor (~40%). I have not profiled this yet but potentially multi-thread overhead at high frequencies?

Overall a step in the right direction but there is room for investigation.

I have opened a PR for the backports on jazzy. For those who want to test this the changes are in: https://github.com/nsherif-fugro/rmw_fastrtps/tree/jazzy

Ufuk-a, 2026-11-10:
Independent confirmation on Jazzy, since @dlanov asked for a test from a system that hits this. Short version: the backport of ros2/rmw_fastrtps#930 removes the busy loop completely here, in a minimal case and in a 45-process system.

**Setup.** Ubuntu 24.04 container on x86_64 (16 threads), Jazzy binaries (rclpy 7.1.11, rmw_fastrtps_cpp 8.4.4). "Patched" is @nsherif-fugro's `jazzy` branch (backports of ros2/rmw_fastrtps#930 and ros2/rmw_fastrtps#890) rebased onto the current `jazzy` (8.4.5); `rmw_fastrtps_shared_cpp` and `rmw_fastrtps_cpp` built Release as an overlay, everything else stock.

### 1. Minimal case

One process publishes `std_msgs/Int64` at a fixed rate; another has a single subscription with a trivial callback (script below). "Empty waits" counts returns of `wait_set.wait()` with no subscription, guard condition, timer, client or service ready. CPU is the subscriber process, 100% = one core, 10 s windows.

| Publisher | Executor | Stock: received | Stock: CPU | Stock: empty waits/s | Patched: received | Patched: CPU | Patched: empty waits/s |
|---|---|---:|---:|---:|---:|---:|---:|
| 100 Hz | Multi | **60.5 Hz** | 105.5% | 32147 | 100.0 Hz | 3.1% | 0 |
| 500 Hz | Multi | **75.5 Hz** | 110.7% | 33701 | 500.0 Hz | 9.1% | 0 |
| 2000 Hz | Multi | **34.2 Hz** | 111.6% | 33957 | 2000.0 Hz | 26.8% | 0 |
| 100 Hz | Single | 100.0 Hz | 2.3% | 0 | 100.0 Hz | 2.3% | 0 |
| 500 Hz | Single | 500.0 Hz | 8.9% | 0 | 500.0 Hz | 9.4% | 0 |
| 2000 Hz | Single | 2000.0 Hz | 21.8% | 0 | 2000.0 Hz | 23.0% | 0 |

On stock, `SubscriptionEventCallbacks(use_default_callbacks=False)` gives the same picture as the patch for the `MultiThreadedExecutor` (2.8% / 8.4% / 24.8% CPU, full rate, no empty waits), which matches the diagnosis in ros2/rmw_fastrtps#929.

With the patch the `MultiThreadedExecutor` costs about the same as the `SingleThreadedExecutor` in this test (within 15%), so I do not see the residual 2x that was reported above; that may depend on what the callback does.

### 2. A whole system

A robot software stack in simulation: 25 rclpy processes (each a `MultiThreadedExecutor`, on simulation time from a 100 Hz `/clock`), 20 C++ nodes and Gazebo. CPU summed over the Python processes, three 20 s windows each (spread under 2%):

| | Python processes, CPU | Simulation real-time factor |
|---|---:|---:|
| Stock | 790% | 0.53 |
| Stock, default event callbacks disabled | 142% | 0.72 |
| Patched | 171% | 0.71 |
| Patched, default event callbacks disabled | 141% | 0.71 |

One consequence worth spelling out for people who land here from "high idle CPU in simulation": with `use_sim_time` every rclpy node has a `/clock` subscription, so on stock Jazzy even a node that does nothing pays this on every clock message when it runs on a `MultiThreadedExecutor`.

So +1 for getting the backport into Jazzy, and happy to rerun this against the split PRs when they are up.

<details>
<summary>Script for the minimal case</summary>

```python
#!/usr/bin/env python3
"""rclpy executor cost per message. usage: repro.py pub HZ | repro.py sub multi|single [noev]
sub prints receive rate, CPU of the process and how many wait-set waits returned with nothing ready."""
import os, resource, sys, threading, time
import rclpy, rclpy.executors as ex_mod
from rclpy.executors import MultiThreadedExecutor, SingleThreadedExecutor
from std_msgs.msg import Int64

SECS = float(os.environ.get("SECS", 10))
rclpy.init()
if sys.argv[1] == "pub":
    node = rclpy.create_node("repro_pub")
    pub = node.create_publisher(Int64, "repro", 10)
    node.create_timer(1.0 / float(sys.argv[2]), lambda: pub.publish(Int64()))
    rclpy.spin(node)

waits = empty = 0
Orig = ex_mod._rclpy.WaitSet
class Counting:
    def __init__(self, *a): self.w = Orig(*a)
    def __getattr__(self, k): return getattr(self.w, k)
    def wait(self, t):
        global waits, empty
        r = self.w.wait(t)
        waits += 1
        empty += not any(self.w.get_ready_entities(k) for k in ("subscription", "guard_condition", "timer", "client", "service"))
        return r
ex_mod._rclpy.WaitSet = Counting

if "noev" in sys.argv:
    from rclpy.event_handler import SubscriptionEventCallbacks
    events = SubscriptionEventCallbacks(use_default_callbacks=False)
else:
    events = None
got = 0
def cb(msg):
    global got
    got += 1
node = rclpy.create_node("repro_sub")
node.create_subscription(Int64, "repro", cb, 10, event_callbacks=events)
ex = MultiThreadedExecutor() if sys.argv[2] == "multi" else SingleThreadedExecutor()
ex.add_node(node)
threading.Thread(target=ex.spin, daemon=True).start()
while not got: time.sleep(0.05)
time.sleep(1)
g0, w0, e0, t0, c0 = got, waits, empty, time.time(), resource.getrusage(resource.RUSAGE_SELF)
time.sleep(SECS)
dt, c1 = time.time() - t0, resource.getrusage(resource.RUSAGE_SELF)
cpu = c1.ru_utime + c1.ru_stime - c0.ru_utime - c0.ru_stime
print(f"{' '.join(sys.argv[2:]):12s} rx {(got-g0)/dt:7.1f} Hz  cpu {100*cpu/dt:5.1f}%  waits/s {(waits-w0)/dt:8.0f}  empty waits/s {(empty-e0)/dt:8.0f}", flush=True)
os._exit(0)
```

`python3 repro.py pub 500` in one terminal, then `python3 repro.py sub multi`, `sub single`, `sub multi noev` in another.

</details>

Disclosure: the investigation and this write-up were done with Claude (AI assistant) working on my machine.

dlanov, 2026-11-11:
Thanks for the detailed independent confirmation. The zero empty-wait rate with the patch and the whole-system CPU results make the connection to the spurious QoS-event wakeups in ros2/rmw_fastrtps#930 quite convincing.

This also gives strong justification for a Jazzy backport. The initial combined backport PR was closed to address API/ABI compatibility and split #930 and #890 into separate backports, so I think the next step is to validate the revised #930 backport once it is available.

nsherif-fugro, 2026-11-11:
Thank you for confirming!

Regarding the higher CPU usage in my test, I ran my tests from inside a Jazzy container running on an Ubuntu VM, so that could explain the difference vs native Ubuntu.

The new PRs are up now including the rmw_implementation backports (ros2/rmw_implementation#314 & ros2/rmw_implementation#315). I still have ros2/rmw_fastrtps#972 as a draft while ros2/rmw_fastrtps#971 gets merged.

I have repeated the original test and yours (publisher and subcriber on separate processes) using the update backport and got this result:

| Publisher | Executor | Stock: received | Stock: CPU | Stock: empty waits/s | Patched: received | Patched: CPU | Patched: empty waits/s |
|---|---|---:|---:|---:|---:|---:|---:|
| 500 Hz | Multi | **63.2 Hz** | 116.5% | 7919 | 499.1 Hz | 40.8% | 0 |
| 2000 Hz | Multi | **27.1 Hz** | 123.4% | 7954 | 1948.8 Hz | 119.3% | 0 |
| 500 Hz | Single | 499.9 Hz | 23.9% | 0 | 500.0 Hz | 23.8% | 0 |
| 2000 Hz | Single | 1999.1 Hz | 82.9% | 0 | 1998.9 Hz | 83.2% | 0 |

So having a separate subscriber process does significantly reduce CPU usage for me, but the MultiThreadedExecutor still has higher CPU usage compared to the SingleThreadedExecutor.

If you do validate the revised backport I would be curious to know if having the timer/publisher/subscriber in 1 Executor shows a singificantly higher CPU %.