ros2/rclpy #1234: async wait/sleep implementation

russkel, 2024-02-29:
## Feature request

#### Feature description
I have been writing some hardware "test" scripts used for checking the integration of hardware in real time (not using any test framework or assertions but simply stimulating a system with inputs and manually confirming the hardware is performing as expected). I have found writing this using async has made things much simpler and nicer:

e.g.

```python
    async def test_1(self):
        self.get_logger().info("Setting Rudder to Center")
        await self.set_and_wait_until_rudder_angle(0.0, 5.0)

        self.get_logger().info("Setting Rudder to Full Port")
        await self.set_and_wait_until_rudder_angle(-1.0, 5.0)
```

To which I kick off like so:

```python
def main(args=None):
    rclpy.init(args=args)
    executor = SingleThreadedExecutor()
    node = TestNode()
    executor.add_node(node)

    try:
        node.get_logger().info("Starting Test 1")
        task = executor.create_task(node.test_1())
        executor.spin_until_future_complete(task)

        node.get_logger().info("Testing complete!")
    except KeyboardInterrupt:
        pass

    node.destroy_node()
    executor.shutdown()
```

Now there is pretty much no examples of this `create_task` being used like this so I have been experimenting and reading a bit of the rclpy code. I wasn't sure how to use the executor to do a `await asyncio.sleep` after looking at the code in depth. I had a quick look at how `asyncio` did it and I came up with this hack using a `Timer`:

```python
    async def async_wait_for(self, rel_time: float):
        fut = Future()
        timer = None

        def done_waiting():
            fut.set_result(None)
            timer.cancel()

        timer = self.create_timer(rel_time, done_waiting)
        await fut
```

This was the only missing piece and I was able to asynchronously "wait" and write the co-routines as expected.

My request is something like this is added to the `Executor` class, perhaps using a more elegant interface than abusing a Timer like this.

#279 might be related.

arjo129, 2024-02-29:
I like this idea. I recently ran into a similar need. Perhaps `spin_until_future_complete` has a timeout option which I abused with a default empty future that never gets set.

russkel, 2024-02-29:
> I like this idea. I recently ran into a similar need. My solution was `spin_until_future_complete` has a timeout option which I abused with a default empty future that never gets set. It probably is not great but 🤷.

If I am not mistaken that will block the thread you're in? How do you use that to await in a co-routine?

arjo129, 2024-03-01:
Yeah it does. I wasnt using async await so... If an async/await option were available I'd probably have prefered that.

russkel, 2024-03-01:
> Yeah it does. I wasnt using async await so... If an async/await option were available I'd probably have prefered that. Just echoing the sentiment that an API like this would be useful for people.

Yep I understood what you meant, I was just wondering if I had missed something.

jmblixt3, 2024-09-15:
> If I am not mistaken that will block the thread you're in? How do you use that to await in a co-routine?

I was also looking into this, and looks like since #1316  `spin_until_future_complete` now works in callbacks. I'm still not a fan of relying on an empty future never finishing though

russkel, 2025-06-18:
I still would like this to be formalised. @clalancette does this make sense to be added?

sungpyo9053, 2026-10-28:
On the last question (whether the new async executor makes this unnecessary): I ran the Timer + Future helper from this thread against the stock Docker images, 50 sleeps of 0.02 s with a 20 Hz timer running on the same node.

| distro | rclpy | total | worst overshoot | timers left on node after 50 sleeps |
|---|---|---|---|---|
| Humble | 3.3.22 | 1.02-1.03 s | 1.6 ms | 51 with `timer.cancel()`, 1 with `node.destroy_timer(timer)` |
| Jazzy | 7.1.12 | 1.02 s | 0.6 ms | 51 / 1 |
| Lyrical | 10.0.11 | 1.03 s | 1.4 ms | 51 / 1 |

So the helper works on all three, but if it only calls `cancel()` every sleep leaks a timer on the node; it needs `destroy_timer()` in a `finally`.

On Lyrical, `rclpy.experimental` already covers this: `await node.get_clock().sleep(0.02)` under the asyncio-based executor gave 50 x 0.02 s = 1.02 s, worst overshoot 1.5 ms, with the 20 Hz timer still ticking 20 times (Python 3.14). It is experimental and needs Python >= 3.12, so for Humble/Jazzy the Timer + Future helper (with `destroy_timer`) is still the way.

Scripts and raw output are in a write-up on my blog: https://huntlab.app/en/rclpy-asyncio-no-running-event-loop-call-async-hangs/ (disclosure: it's my blog; I used an AI assistant to help run the measurements and draft this comment, and checked the numbers myself).