ros2/rclpy #1257: [Feature] Static Type Checking With ament_mypy

InvincibleRMC, 2024-03-22:
## Feature request

#### Feature description

Using `ament_mypy` to assure future code has proper typing annotations. First and foremost this change would need to wait on many pull requests as there are over 100 mypy errors still remaining on rolling. This change would also likely require any upstream dependencies to adopt `ament_mypy` to avoid `[no-untyped-call]` error and other similar errors being raised from a new upstream function being added and called. I bring up this question now so if this feature is desired I can start getting those upstream dependencies checked with `ament_mypy` earlier to avoid future waiting. For CI time considerations `ament_mypy` runs in about 2.4s. The benefits of this would be improved code readability, helping catch unexpected edge cases, as well remove the need for downstream dependencies who type check to suppress `[no-untyped-call]` and other similar errors whenever using rclpy. 

#### Implementation considerations

For implementation currently only `ament_mypy` exists. The alternative would be perhaps creating an `ament_pyright` and using that instead for static type checking.

If this is a feature the maintainers are willing to support I will gladly help accomplish this goal. And if anyone else wants to help test with `ament_mypy` or add type annotations to `rclpy` or any other ROS2 python file that would be greatly appreciated.

fujitatomoya, 2024-03-22:
@InvincibleRMC 1st of all, really appreciate all the contribution you have been and are making for `rclpy` type check.

we expected that this feature request would come up.
i think mostly your consideration describes pretty much everything like dependencies, benefits and concerns so on. (IMO, this should be integrated into CI/CD pipeline, so that we can sustain the type checking.)

i think that this will be better consistent and readable for sure, but that is gonna be the work and a long run. i also would like to hear from more opinion here.

let's keep this open to get more feedback!

InvincibleRMC, 2024-09-13:
@fujitatomoya The current default for ament_mypy doesn't include `no-untyped-def` (more info found [here](no-untyped-def)). I think this should be updated either in ament_mypy's default for every ROS package or at a minimum for `rclpy` since it gets new features and modifications quite often. Otherwise in a few years a bunch of code will acrew no type annotations and we will be back where we started.

clalancette, 2024-09-13:
> I think this should be updated either in ament_mypy's default for every ROS package 

We can discuss this, but we'll need to discuss with the whole community.  Changing this default will affect all packages (though looking through the packages released into Rolling, there are only maybe a dozen that use `ament_mypy`).

> or at a minimum for `rclpy` since it gets new features and modifications quite often.

This is pretty straightforward, so I would suggest starting there.

InvincibleRMC, 2025-01-16:
Will need this merged https://github.com/ament/ament_lint/pull/516.

closed 2026-08-31 by fujitatomoya