ros2/rclpy #1266: :farmer: `test_guard_condition`

Crola1702, 2024-04-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Jammy
- Installation type:
  - Source
- Version or commit hash:
  - Rolling
- DDS implementation:
  - Cyclonedds
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

1. Run a build in [Rolling cyclonedds](http://build.ros2.org/view/Rci/job/Rci__nightly-cyclonedds_ubuntu_jammy_amd64/)
2. See test regressions appear

#### Additional information

Reference build: https://build.ros2.org/view/Rci/job/Rci__nightly-cyclonedds_ubuntu_jammy_amd64/765/#showFailuresLink

Test regressions:
* [projectroot.test_guard_condition](https://build.ros2.org/view/Rci/job/Rci__nightly-cyclonedds_ubuntu_jammy_amd64/765/testReport/junit/(root)/projectroot/test_guard_condition/)
* [rclpy.rclpy.test.test_guard_condition.TestGuardCondition.test_double_trigger](https://build.ros2.org/view/Rci/job/Rci__nightly-cyclonedds_ubuntu_jammy_amd64/765/testReport/junit/rclpy.rclpy.test.test_guard_condition/TestGuardCondition/test_double_trigger/)
* [rclpy.rclpy.test.test_guard_condition.TestGuardCondition.test_trigger](https://build.ros2.org/view/Rci/job/Rci__nightly-cyclonedds_ubuntu_jammy_amd64/765/testReport/junit/rclpy.rclpy.test.test_guard_condition/TestGuardCondition/test_trigger/)

Log output:

<details>
<summary>
projectroot.test_guard_condition (and TestGuardCondition.test_double_trigger and TestGuardCondition.test_trigger)
</summary>

```
============================= test session starts ==============================
platform linux -- Python 3.10.12, pytest-6.2.5, py-1.10.0, pluggy-0.13.0
cachedir: /tmp/ws/build_isolated/rclpy/ament_cmake_pytest/test_guard_condition/.cache
rootdir: /tmp/ws/src/ros2/rclpy, configfile: pytest.ini
plugins: ament-xmllint-0.16.4, ament-copyright-0.16.4, ament-pep257-0.16.4, ament-flake8-0.16.4, ament-lint-0.16.4, mock-3.6.1, timeout-2.1.0, cov-3.0.0, colcon-core-0.16.0, rerunfailures-10.2
collected 2 items

../../src/ros2/rclpy/rclpy/test/test_guard_condition.py FF               [100%]

=================================== FAILURES ===================================
____________________ TestGuardCondition.test_double_trigger ____________________

self = <test.test_guard_condition.TestGuardCondition testMethod=test_double_trigger>

    def test_double_trigger(self):
        called1 = False
        called2 = False
    
        def func1():
            nonlocal called1
            called1 = True
    
        def func2():
            nonlocal called2
            called2 = True
    
        gc1 = self.node.create_guard_condition(func1)
        gc2 = self.node.create_guard_condition(func2)
    
        self.executor.spin_once(timeout_sec=0)
        self.assertFalse(called1)
        self.assertFalse(called2)
    
        gc1.trigger()
        gc2.trigger()
        self.executor.spin_once(timeout_sec=0)
        self.executor.spin_once(timeout_sec=0)
        self.assertTrue(called1)
>       self.assertTrue(called2)
E       AssertionError: False is not true

../../src/ros2/rclpy/rclpy/test/test_guard_condition.py:80: AssertionError
_______________________ TestGuardCondition.test_trigger ________________________

self = <test.test_guard_condition.TestGuardCondition testMethod=test_trigger>

    def test_trigger(self):
        called = False
    
        def func():
            nonlocal called
            called = True
    
        gc = self.node.create_guard_condition(func)
    
        self.executor.spin_once(timeout_sec=0)
        self.assertFalse(called)
    
        gc.trigger()
        self.executor.spin_once(timeout_sec=0)
>       self.assertTrue(called)
E       AssertionError: False is not true

../../src/ros2/rclpy/rclpy/test/test_guard_condition.py:52: AssertionError
- generated xml file: /tmp/ws/test_results/rclpy/test_guard_condition.xunit.xml -
=========================== short test summary info ============================
FAILED ../../src/ros2/rclpy/rclpy/test/test_guard_condition.py::TestGuardCondition::test_double_trigger
FAILED ../../src/ros2/rclpy/rclpy/test/test_guard_condition.py::TestGuardCondition::test_trigger
============================== 2 failed in 0.26s ===============================
-- run_test.py: return code 1
-- run_test.py: verify result file '/tmp/ws/test_results/rclpy/test_guard_condition.xunit.xml'
```

</details>

---

This is a consistent issue for Rolling CycloneDDS

Crola1702, 2024-06-18:
This is still a consistent issue in nightly-cyclonedds jobs ([Rci](https://build.ros2.org/view/Rci/job/Rci__nightly-cyclonedds_ubuntu_noble_amd64/60/#showFailuresLink), [Jci](https://build.ros2.org/view/Jci/job/Jci__nightly-cyclonedds_ubuntu_noble_amd64/23/#showFailuresLink))

claraberendsen, 2024-07-23:
This are back green.

closed 2024-07-23 by claraberendsen