ros2/rclpy #1269: Add utility function to get datetime.datetime from Timn

tonynajjar, 2024-04-03:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

gordon-n-stevenson, 2024-04-23:
This is an excellent idea, as I think it is a common issue when trying to timestamp out things and the management of the seconds/nanoseconds into a correct datetime can be challenging for new users.

It wouldn't be far off something like this SO answer - https://stackoverflow.com/questions/15649942/how-to-convert-epoch-time-with-nanoseconds-to-human-readable - but with the use of rclpy.Time right?

JeanPaulSB, 2025-04-04:
Hi everyone, I'm a new contributor, can I work on this as my first issue? thx

sloretz, 2025-04-04:
@JeanPaulSB Welcome!

I think you could take this as a first issue. However, time is always more complicated than one would think it should be. There might be something subtle we're not thinking of. I recommend [reading the ROS Contributing guide](https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing.html), but here's some more targeted guidance:

* Target the `rolling` branch with your PR
* Add a new function or method to do the conversion. I would recommend `Time.to_datetime()` to match [the existing API `Time.to_msg()`](https://github.com/ros2/rclpy/blob/c489ab80c3ad375102dba9d28af4853e6f680e30/rclpy/rclpy/time.py#L157)
* Add a new test to [test_time.py](https://github.com/ros2/rclpy/blob/rolling/rclpy/test/test_time.py)

Something subtle about time in ROS is we support different kinds of Clocks. `ROS_TIME` and `SYSTEM_TIME` both make sense to me to be convertible to a `datetime` object, but `STEADY_TIME` ([or `UNINITIALIZED`](https://github.com/ros2/rclpy/blob/c489ab80c3ad375102dba9d28af4853e6f680e30/rclpy/src/rclpy/_rclpy_pybind11.cpp#L66-L69)) does not. I recommend making `Time.to_datetime()` raise an exception if the time object's `clock_type`  is not one of `ClockType.ROS_TIME` or `ClockType.SYSTEM_TIME`.

Feel free to ask me questions here as you go. I'm happy to help!

JeanPaulSB, 2025-04-05:
@sloretz Hi! hope you are doing well. I'm trying to build **rclpy** from source with my forked repo, however, it is throwing me this error when building with `colcon build --symlink-install`
```
Starting >>> rclpy   
--- stderr: rclpy                             
/app/rclpy/rclpy/src/rclpy/action_server.cpp: In member function ‘void rclpy::ActionServer::configure_introspection(rclpy::Clock&, pybind11::object, rcl_service_introspection_state_t)’:
/app/rclpy/rclpy/src/rclpy/action_server.cpp:380:19: error: ‘rcl_action_server_configure_action_introspection’ was not declared in this scope; did you mean ‘rcl_service_configure_service_introspection’?
  380 |   rcl_ret_t ret = rcl_action_server_configure_action_introspection(
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                   rcl_service_configure_service_introspection
/app/rclpy/rclpy/src/rclpy/action_client.cpp: In member function ‘void rclpy::ActionClient::configure_introspection(rclpy::Clock&, pybind11::object, rcl_service_introspection_state_t)’:
/app/rclpy/rclpy/src/rclpy/action_client.cpp:282:19: error: ‘rcl_action_client_configure_action_introspection’ was not declared in this scope; did you mean ‘rcl_client_configure_service_introspection’?
  282 |   rcl_ret_t ret = rcl_action_client_configure_action_introspection(
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                   rcl_client_configure_service_introspection
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:132: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/action_server.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:104: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/action_client.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:212: CMakeFiles/_rclpy_pybind11.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
```
I'm using _Ubuntu 24.04 noble_.

Should I build ROS2 from source to make it work? Thank you so much for your help.

christophebedard, 2025-04-05:
Yeah, we sometimes make changes to one repository/package that requires changes in another package. In this case, you're building `rclpy` @ `rolling` from source (I assume), and it uses a recent change to `rcl` (https://github.com/ros2/rcl/pull/1207). If you just build all of ROS 2 from source, it should work. (Otherwise, you can try to figure out which exact packages need to be built from source, but that's a bit annoying and it can change at any time.)

JeanPaulSB, 2025-04-05:
@christophebedard I was mixing binaries and building other packages from source but It ended up being a nightmare because I had to figure out which ones need to be built from source as you pointed out so I finally moved on by building everything from source. Thanks for your response. I have already raised the implementation on #1443.