ros2/rclpy #1279: Update the executor to not throw ExternalShutdownException

clalancette, 2024-05-07:
This is an issue split out of https://github.com/ros2/examples/pull/379

Currently the rclpy executor can throw an `ExternalShutdownException`: https://github.com/ros2/rclpy/blob/99f1ce933ddd1677fda8fa602af3684b93f814eb/rclpy/rclpy/executors.py#L689

However, there is really no need to throw an exception at this point that callers of `rclpy.spin` would need to handle; we could instead just return a value indicating why the spin stopped.  This would make it so users need less error-handling code, make this act more like rclcpp (which doesn't throw an exception here), and still provide information in case somebody wants to know why the spin stopped.

Further, we can easily stop throwing an exception here with no impact to downstream consumers.  If they aren't handling the exception today, this will just make it a bit more robust.  If they are handling `ExternalShutdownException` today, then that effectively becomes dead code but nothing really breaks.

sbarber-dsi, 2024-10-07:
No.
This exception was useful because the multi-threaded executor does NOT shutdown cleanly so we can catch this exception and ignore it.
Now it throws a generic RuntimeError so ignoring it is unadvised.
Now we're **checking strings** to decide if we report an exception or not.

`  File ".../ros/node.py", line 433, in spin
    self._executor.spin()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 294, in spin
    self.spin_once()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 794, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 786, in _spin_once_impl
    self._executor.submit(handler)
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 167, in submit
    raise RuntimeError('cannot schedule new futures after shutdown')
RuntimeError: cannot schedule new futures after shutdown`

clalancette, 2024-10-07:
To be clear; this was an idea we had, never implemented.  Whatever you are running into is a different issue.