ros2/rclpy #1282: aarch64 cross compiling

XQSwx, 2024-05-11:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 22.04
- Installation type:
  Source code compilation
- Version or commit hash:
 humble
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue
"When cross-compiling with aarch64, it was found that the generated .so file is for the X86 platform instead of aarch64. Other ROS2-generated files are for the aarch64 platform."
```

```

#### Expected behavior

#### Actual behavior

#### Additional information

+ ROS_DISTRO=
+ export LD_LIBRARY_PATH=/buildroot/host/lib/
+ LD_LIBRARY_PATH=/buildroot/host/lib/
+ export SYSROOT=/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/
+ SYSROOT=/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/
+ export CC=/buildroot/host/bin/aarch64-buildroot-linux-gnu-gcc
+ CC=/buildroot/host/bin/aarch64-buildroot-linux-gnu-gcc
+ export CXX=/buildroot/host/bin/aarch64-buildroot-linux-gnu-g++
+ CXX=/buildroot/host/bin/aarch64-buildroot-linux-gnu-g++
+ export CROSS_COMPILE=/buildroot/host/bin/aarch64-buildroot-linux-gnu-
+ CROSS_COMPILE=/buildroot/host/bin/aarch64-buildroot-linux-gnu-
+ export TARGET_ARCH=aarch64
+ TARGET_ARCH=aarch64
+ export PYTHON_SOABI=cpython-38-aarch64-linux-gnu
+ PYTHON_SOABI=cpython-38-aarch64-linux-gnu
+ export ROS2_INSTALL_PATH=/buildroot/target/opt/ros/
+ ROS2_INSTALL_PATH=/buildroot/target/opt/ros/
+ export TARGET_TRIPLE=/buildroot/host/bin/aarch64-buildroot-linux-gnu
+ TARGET_TRIPLE=/buildroot/host/bin/aarch64-buildroot-linux-gnu
+ export ARCH=aarch64
+ ARCH=aarch64
+ trap cleanup EXIT
+ prepare_source
+ sed -i '/^.*ExternalProject.*$/i \ \ list(APPEND cmake_configure_args "-DCMAKE_SYSTEM_NAME=${CMAKE_SYSTEM_NAME}") \n\ \ list(APPEND cmake_configure_args "-DCMAKE_SYSTEM_PROCESSOR=${CMAKE_SYSTEM_PROCESSOR}")' src/ros2/mimick_vendor/CMakeLists.txt
+ mkdir -p /buildroot/build/ros /buildroot/target/opt/ros
+ pushd /buildroot/host/aarch64-buildroot-linux-gnu/sysroot/
/buildroot/host/aarch64-buildroot-linux-gnu/sysroot /opt/ros/humble
+ rm -rf opt
+ ln -sf ../../../target/opt/ros/ opt
+ popd
/opt/ros/humble
+ mkdir -p /etc/qemu-binfmt
+ ln -sf /buildroot/target/ /etc/qemu-binfmt/aarch64
+ colcon mixin remove default
+ colcon mixin add default file:///root/index.yaml
+ colcon mixin update default
fetching default: file:///root/index.yaml ...
  fetching file:///root/cross-compile.mixin ...
  . /root/.colcon/mixin/default/cross-compile.mixin
+ LDFLAGS='-L/buildroot/host/lib -Wl,-rpath,/buildroot/host/lib'
+ CFLAGS='-O2 -I/buildroot/host/include/'
+ colcon build --mixin aarch64-linux --merge-install --build-base /buildroot/build/ros/ --install-base /buildroot/target/opt/ros --event-handlers console_direct+ --cmake-force-configure --packages-select rclpy
Starting >>> rclpy
[... 117 lines cut ...]
-- Installing: /buildroot/target/opt/ros/share/rclpy/environment/library_path.dsv
-- Up-to-date: /buildroot/target/opt/ros/share/ament_index/resource_index/package_run_dependencies/rclpy
-- Up-to-date: /buildroot/target/opt/ros/share/ament_index/resource_index/parent_prefix_path/rclpy
-- Up-to-date: /buildroot/target/opt/ros/share/rclpy/environment/ament_prefix_path.sh
-- Installing: /buildroot/target/opt/ros/share/rclpy/environment/ament_prefix_path.dsv
-- Up-to-date: /buildroot/target/opt/ros/share/rclpy/environment/path.sh
-- Installing: /buildroot/target/opt/ros/share/rclpy/environment/path.dsv
-- Up-to-date: /buildroot/target/opt/ros/share/rclpy/local_setup.bash
-- Up-to-date: /buildroot/target/opt/ros/share/rclpy/local_setup.sh
-- Up-to-date: /buildroot/target/opt/ros/share/rclpy/local_setup.zsh
-- Installing: /buildroot/target/opt/ros/share/rclpy/local_setup.dsv
-- Installing: /buildroot/target/opt/ros/share/rclpy/package.dsv
-- Up-to-date: /buildroot/target/opt/ros/share/ament_index/resource_index/packages/rclpy
-- Up-to-date: /buildroot/target/opt/ros/share/rclpy/cmake/rclpyConfig.cmake
-- Up-to-date: /buildroot/target/opt/ros/share/rclpy/cmake/rclpyConfig-version.cmake
-- Up-to-date: /buildroot/target/opt/ros/share/rclpy/package.xml
Finished <<< rclpy [21.9s]

Summary: 1 package finished [22.6s]
----
## Feature request

#### Feature description

#### Implementation considerations

XQSwx, 2024-05-11:
Can you help me? thank you

fujitatomoya, 2024-05-11:
@XQSwx thanks for creating issue.

unfortunately, cross-compile is not supported anymore? https://docs.ros.org/en/humble/How-To-Guides/Cross-compilation.html, https://github.com/ros-tooling/cross_compile repo has been archived.

XQSwx, 2024-05-13:
"I know that I compiled this using the Buildroot from Rockchip, so I can only work based on this compiled base."

XQSwx, 2024-05-13:
Ultimately, there's a bug here, unable to generate cross-compilation chain's link libraries. Only x86 ones can be generated.

XQSwx, 2024-05-13:
Starting >>> rclpy
--- output: rclpy
Not searching for unused variables given on the command line.
-- Found ament_cmake: 1.3.8 (/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/share/ament_cmake/cmake)
-- Found Python3: /usr/bin/python3 (found version "3.10.12") found components: Interpreter
-- Found lifecycle_msgs: 1.2.1 (/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/share/lifecycle_msgs/cmake)
-- Found rosidl_generator_c: 3.1.5 (/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.5 (/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.5 (/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found rcl: 5.3.7 (/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/share/rcl/cmake)
-- Found rmw_implementation_cmake: 6.1.1 (/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 6.2.6 (/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/share/rmw_fastrtps_cpp/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rcl_action: 5.3.7 (/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/share/rcl_action/cmake)
-- Found rcl_lifecycle: 5.3.7 (/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/share/rcl_lifecycle/cmake)
-- Found Python3: /usr/bin/python3 (found version "3.10.12") found components: Interpreter Development Development.Module Development.Embed
-- Found pybind11_vendor: 2.4.2 (/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/share/pybind11_vendor/cmake)
-- Found pybind11: /buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/include/pybind11_vendor (found version "2.9.1")
-- Found ament_lint_auto: 0.12.10 (/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/share/ament_lint_auto/cmake)
-- Found gtest sources under '/buildroot/host/aarch64-buildroot-linux-gnu/sysroot/opt/src/gtest_vendor': C++ tests using 'Google Test' will be built
-- Added test 'copyright' to check source files copyright and LICENSE
-- Added test 'cpplint' to check C / C++ code against the Google style
-- Configured cpplint exclude dirs and/or files:
-- Added test 'flake8' to check Python code syntax and style conventions
-- Added test 'lint_cmake' to check CMake code style
-- Added test 'pep257' to check Python code against some of the docstring style conventions in PEP 257
-- Added test 'uncrustify' to check C / C++ code style
-- Configured uncrustify additional arguments:
-- Added test 'xmllint' to check XML markup files
-- Configuring done
-- Generating done
-- Build files have been written to: /buildroot/build/ros/rclpy
Consolidate compiler generated dependencies of target gtest_main
Consolidate compiler generated dependencies of target gtest
Consolidate compiler generated dependencies of target _rclpy_pybind11
[  5%] Built target gtest_main
[ 10%] Built target gtest
[ 10%] Built target ament_cmake_python_copy_rclpy
Consolidate compiler generated dependencies of target test_python_allocator
[ 16%] Built target test_python_allocator
[ 18%] Linking CXX shared library test_rclpy/_rclpy_pybind11.cpython-38-aarch64-linux-gnu.so
running egg_info
writing rclpy.egg-info/PKG-INFO
writing dependency_links to rclpy.egg-info/dependency_links.txt
writing top-level names to rclpy.egg-info/top_level.txt
reading manifest file 'rclpy.egg-info/SOURCES.txt'
writing manifest file 'rclpy.egg-info/SOURCES.txt'
[ 18%] Built target ament_cmake_python_build_rclpy_egg
[100%] Built target _rclpy_pybind11
-- Install configuration: ""
-- Up-to-date: /buildroot/target/opt/ros/share/rclpy/environment/pythonpath.sh
-- Installing: /buildroot/target/opt/ros/share/rclpy/environment/pythonpath.dsv
-- Up-to-date: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy-3.3.12-py3.10.egg-info
-- Installing: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy-3.3.12-py3.10.egg-info/SOURCES.txt
-- Installing: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy-3.3.12-py3.10.egg-info/PKG-INFO
-- Installing: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy-3.3.12-py3.10.egg-info/top_level.txt
-- Installing: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy-3.3.12-py3.10.egg-info/dependency_links.txt
-- Up-to-date: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy
-- Up-to-date: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy/node.py
-- Up-to-date: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy/validate_topic_name.py
-- Up-to-date: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy/publisher.py
-- Up-to-date: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy/topic_or_service_is_hidden.py
-- Up-to-date: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy/validate_namespace.py
-- Up-to-date: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy/constants.py
-- Up-to-date: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy/logging.py
-- Up-to-date: /buildroot/target/opt/ros/local/lib/python3.10/dist-packages/rclpy/parameter_service.py
[... cut ...]
-- Up-to-date: /buildroot/target/opt/ros/share/rclpy/local_setup.zsh
-- Installing: /buildroot/target/opt/ros/share/rclpy/local_setup.dsv
-- Installing: /buildroot/target/opt/ros/share/rclpy/package.dsv
-- Up-to-date: /buildroot/target/opt/ros/share/ament_index/resource_index/packages/rclpy
-- Up-to-date: /buildroot/target/opt/ros/share/rclpy/cmake/rclpyConfig.cmake
-- Up-to-date: /buildroot/target/opt/ros/share/rclpy/cmake/rclpyConfig-version.cmake
-- Up-to-date: /buildroot/target/opt/ros/share/rclpy/package.xml
---
Finished <<< rclpy [21.8s]

XQSwx, 2024-05-13:
Maybe my understanding is incorrect, I suspect that the CMakeLists file does not match the generation of linked libraries for cross-compilation.

clalancette, 2024-05-13:
> unfortunately, cross-compile is not supported anymore? https://docs.ros.org/en/humble/How-To-Guides/Cross-compilation.html, https://github.com/ros-tooling/cross_compile repo has been archived.

You are right that it is currently not a supported configuration.  Because of that, there may indeed be a bug here.

That said, if @XQSwx can find and fix a bug here, we'd be very happy to review a PR to make fixes.

XQSwx, 2024-05-13:
Although cross compile is not supported, every compilation passes the cmake parameter. Can we obtain these parameters to generate a linked library. I have compiled all packages for ROS2, and only this package and ROSBAG2 have encountered this issue. Although I used some clever methods to compile it. But if others use this cmakelist, they may still encounter these issues
    cmake-args:
      - "-DCMAKE_SYSTEM_NAME=Linux"
      - "-DCMAKE_SYSTEM_VERSION=1"
      - "-DCMAKE_SYSTEM_PROCESSOR=aarch64"
      - "-DCMAKE_C_COMPILER=/usr/bin/aarch64-linux-gnu-gcc"
      - "-DCMAKE_CXX_COMPILER=/usr/bin/aarch64-linux-gnu-g++"
      - "-DCMAKE_SYSROOT=$CC_ROOT/sysroot"
      - "-DCMAKE_TRY_COMPILE_TARGET_TYPE=STATIC_LIBRARY"
      - "-DCMAKE_FIND_ROOT_PATH=$CC_ROOT/sysroot/root_path:$CC_ROOT/install"
      - "-DCMAKE_INSTALL_RPATH=$CC_ROOT/sysroot/opt/ros/$ROS_DISTRO/lib"
      - "-DCMAKE_INSTALL_RPATH_USE_LINK_PATH=TRUE"
      - "-DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=NEVER"
      - "-DCMAKE_FIND_ROOT_PATH_MODE_PACKAGE=ONLY"
      - "-DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY"
      - "-DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY"
      - "-DPYTHON_SOABI=cpython-36m-aarch64-linux-gnu"
      - "-DTHREADS_PTHREAD_ARG=0"
      - "--no-warn-unused-cli"

XQSwx, 2024-05-13:
@clalancette 
Others should use the following two parameters passed by colcon to define the generated file
"-DCMAKE_SYSTEM_PROCESSOR=aarch64"
 “-DPYTHON_SOABI=cpython-36m-aarch64-linux-gnu”