ros2/rclpy #1284: Executors mishandle invalid waitables

mhidalgo-bdai, 2024-05-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- Version or commit hash:
  - `humble`
- Client library (if applicable):
  -  `rclpy`

#### Steps to reproduce issue

Take the following snippet:
```python
from rclpy.task import Future

def wait_for_message_async(node, msg_type, topic_name, qos_profile): 
    future = Future()

    def callback(msg) -> None:
        if not future.done():
            future.set_result(msg)

    sub = node.create_subscription(msg_type, topic_name, callback, qos_profile)
    future.add_done_callback(lambda future: node.destroy_subscription(sub))
    return future
```

And call it repeatedly in a thread other than that of the executor associated to the `node`.

#### Expected behavior

No issues whatsoever.

#### Actual behavior

Eventually, an `InvalidHandle` exception is raised in the executor thread from within subscription event handlers.

#### Additional information

This issue is much harder to reproduce than it is to explain. When the [executor collects node entities](https://github.com/ros2/rclpy/blob/99f1ce933ddd1677fda8fa602af3684b93f814eb/rclpy/rclpy/executors.py#L588-L594), it does not exclude [waitables with invalid handles](https://github.com/ros2/rclpy/blob/99f1ce933ddd1677fda8fa602af3684b93f814eb/rclpy/rclpy/executors.py#L652-L657) from the wait set like it does for every other entity. So if you are unlucky enough to have a waitable destroyed in between the time it is collected and the time it is added to the wait set, you will get an `InvalidHandle` exception. The fix is equally simple: prune the list of waitables, excluding those that raised an `InvalidHandle` when pushed to the context stack.

Barry-Xu-2018, 2024-05-17:
> When the [executor collects node entities](https://github.com/ros2/rclpy/blob/99f1ce933ddd1677fda8fa602af3684b93f814eb/rclpy/rclpy/executors.py#L588-L594), it does not exclude [waitables with [...]

Yes. The invalid waitable should be removed here.

```python
for waitable in list(waitables):
    try:
        context_stack.enter_context(waitable)
        entity_count += waitable.get_num_entities()
    except InvalidHandle:
        waitables.remove(waitable)
```
From what I understand, calling list.remove() while iterating through the list is unsafe.
How about  
```python
valid_waitables = []
for waitable in list(waitables):
    try:
        context_stack.enter_context(waitable)
        entity_count += waitable.get_num_entities()
        valid_waitables.append(waitable)
    except InvalidHandle:
        pass
waitables = valid_waitables
```

MatthijsBurgh, 2024-05-17:
First solution already creates a new list, so it is not removing items from the list is iterating on.
So either go for:
```py
for waitable in list(waitables):
    try:
        context_stack.enter_context(waitable)
        entity_count += waitable.get_num_entities()
    except InvalidHandle:
        waitables.remove(waitable)
```
or
```py
valid_waitables = []
for waitable in waitables:
    try:
        context_stack.enter_context(waitable)
        entity_count += waitable.get_num_entities()
        valid_waitables.append(waitable)
    except InvalidHandle:
        pass
#  Use valid_waitables later on in the code
```

I have no clue, which one has the best performance.

Barry-Xu-2018, 2024-05-17:
> First solution already creates a new list, so it is not removing items from the list is iterating on.

Yes. I was mistaken. list(waitables) created a new a list.

mhidalgo-bdai, 2024-05-17:
> I have no clue, which one has the best performance.

Probably the latter, though the difference is likely negligible for the usual number of waitables a single executor may deal with (<100 I'd think). It's three O(N) operations (recreate, iterate, remove) plus (maybe) garbage collection vs one O(N) operation (iterate).

audrow, 2025-02-20:
@mhidalgo-bdai, why was this reopened?

nishantg96, 2025-07-26:
This error still exists - like everyone said it is hard to reproduce though

> [rosbridge_websocket-18] Exception in callback <function main.<locals>.<lambda> at 0x7f73cd18f400>
[rosbridge_websocket-18] Traceback (most recent call last):
[rosbridge_websocket-18]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/tornado/ioloop.py", line 945, in _run
[rosbridge_websocket-18]     val = self.callback()
[rosbridge_websocket-18]   File "/home/rosbridge/install/rosbridge_server/lib/rosbridge_server/rosbridge_websocket", line 340, in <lambda>
[rosbridge_websocket-18]     spin_callback = PeriodicCallback(lambda: executor.spin_once(timeout_sec=0.01), 1)
[rosbridge_websocket-18]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 751, in spin_once
[rosbridge_websocket-18]     self._spin_once_impl(timeout_sec)
[rosbridge_websocket-18]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 740, in _spin_once_impl
[rosbridge_websocket-18]     handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
[rosbridge_websocket-18]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 723, in wait_for_ready_callbacks
[rosbridge_websocket-18]     return next(self._cb_iter)
[rosbridge_websocket-18]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 617, in _wait_for_ready_callbacks
[rosbridge_websocket-18]     waitable.add_to_wait_set(wait_set)
[rosbridge_websocket-18]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/action/client.py", line 361, in add_to_wait_set
[rosbridge_websocket-18]     self._client_handle.add_to_waitset(wait_set)
[rosbridge_websocket-18] rclpy._rclpy_pybind11.RCLError: Failed to add 'rcl_action_client_t' to wait set: action client pointer is invalid, at ./src/rcl_action/action_client.c:470

closed 2026-02-02 by ahcorde