ros2/rclpy #1290: First Time Contributors Documentation (CONTRIBUTING.md)

KristyWaters, 2024-05-25:
I would love it if we could expand upon the instructions for contributing to this repository. 

#TLDR
I have drafted some potential changes to include based on the consistent advice that I have read within the comments of various issues and pull requests that I have reviewed.  That said, I am hoping to get feedback from major contributors to ensure that the instructions are indeed following your preferred best practices for contributing. 

# Details
The CONTRIBUTING.md document needs to be updated to reflect some of the directions included in the ROS2 documentation webpage.  As a new (potential) contributor I found myself confused when looking to contribute to the rclpy repository, and the Issues and Pull Requests pages directed me to the CONTRIBUTING.md document.  However, the only contents of the document are the requirements for the Apache 2 license statements for the files.  It would be beneficial to new contributors to outline the process for contributing to this repository directly within the files rather than only within the ROS2 documentation pages.  This would reduce the time that experienced contributors spend re-explaining the same directions to first-time contributors. 

# Sections to Include
 - Current File Contents: Current License Information and Citation Statement
 - Section for Reporting an Issue
        - Issue Postings
        - Pull Requests
 - Section for First-Time Contributors
 - ?? What else?  Please let me know in the comments below so I can add it to a draft and submit a pull request.

clalancette, 2024-05-25:
While I agree with you that the rules for first time contributions aren't well laid out, I think we should approach this differently.

In particular, the same rules apply to (most of) the 150 repositories that are in the ROS 2 core.  Thus, I think the correct thing to do here is to update the documentation at http://docs.ros.org/en/rolling/The-ROS2-Project/Contributing.html , and then just add links from this repository (and others) to that documentation.

Would you be willing to open a PR over at https://github.com/ros2/ros2_documentation to start that process?

KristyWaters, 2024-05-25:
Absolutely!  Thank you for the feedback. 

Is there anything else that you think should be added to the document before I open the PR? 
(I will start with a DRAFT PR per the guidelines for now.)