ros2/rclpy #1315: Logging with `rclpy` to file misses messages unless accompanied by a consistent stream

alberthli, 2024-07-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries

#### Steps to reproduce issue
I'm trying to dump logs into a file. However, I've found in my project that when we use one-off log messages, nothing gets logged to file, but if that node is also continuously logging a different message on a timer, then the messages which previously did not appear now do get logged to file. This issue only seems to affect `python` nodes.

Here's a minimum example:
```
import rclpy
from rclpy.node import Node
from std_msgs.msg import String

class DummyPublisher(Node):
    def __init__(self):
        super().__init__('dummy_publisher')
        self.publisher_ = self.create_publisher(String, 'dummy_topic', 10)
        self.timer = self.create_timer(0.001, self.publish_dummy_string)
        self.get_logger().info('Dummy Publisher Node has been started.')  # want this to be logged in file

    def publish_dummy_string(self):
        msg = String()
        msg.data = 'Hello, this is a dummy string!'
        self.publisher_.publish(msg)
        # self.get_logger().info('Publishing: "%s"' % msg.data)  # the above only gets logged if this is uncommented

def main(args=None):
    rclpy.init(args=args)
    node = DummyPublisher()
    rclpy.spin(node)
    node.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()

```

When running the above node with the command
```
ROS_LOG_DIR=/path/to/log/dir ros2 run pkg mwe
```
I see an empty file with no messages in it. However, if I uncomment the line `self.get_logger().info('Publishing: "%s"' % msg.data)`, then the message now appears at the top of the file. I'd like to know whether this is somehow expected behavior, as well as any obvious workarounds in Humble.

Barry-Xu-2018, 2024-07-16:
It is related to the below code
https://github.com/ros2/rcl_logging/blob/ecb3ab62168e873ef78201c8fda65b622b9f2707/rcl_logging_spdlog/src/rcl_logging_spdlog.cpp#L203-L204  

By default, you need to wait 5 seconds to ensure the content is flushed into the log file. Alternatively, you can change the log type to error. Currently, it seems that configuring this flush time is not supported. 

FYI, I tried to set `RCL_LOGGING_SPDLOG_EXPERIMENTAL_OLD_FLUSHING_BEHAVIOR=1`. But I've found that it still doesn't write immediately to log file.

alberthli, 2024-07-16:
Thanks for the response. Waiting 5 seconds doesn't do anything, but the `try`/`except`/`finally` block does successfully flush the log messages, so I'll consider that a workable solution for my application.

closed 2024-07-16 by alberthli