ros2/rclpy #1320: Flaky `get_node_names_and_namespaces` shows it's own node name but duplicated

saikishor, 2024-07-25:
We are trying to add some tests to our libraries, and when adding some launch testing we saw some flakiness, and upon more introspection, it seems that the `get_node_names_and_namespaces` method sometimes is giving correct number of nodes, but all of them having the same node name as its own, but other nodes in the same launch testing print them properly.

The correct output `get_node_names_and_namespaces`  printed by a different Node  
```
1: [spawner-4] [WARN] [1721940021.065275625] [spawner_fts_broadcaster]: Node: test_node, Namespace: /
1: [spawner-3] [WARN] [1721940021.065367724] [spawner_joint_state_broadcaster]: Node: spawner_joint_state_broadcaster, Namespace: /
1: [spawner-4] [WARN] [1721940021.065715004] [spawner_fts_broadcaster]: Node: _test_node_I56B2MZ2EL, Namespace: /
1: [spawner-4] [WARN] [1721940021.066066336] [spawner_fts_broadcaster]: Node: controller_manager, Namespace: /
1: [spawner-4] [WARN] [1721940021.066395060] [spawner_fts_broadcaster]: Node: robot_state_publisher, Namespace: /
1: [spawner-4] [WARN] [1721940021.066764435] [spawner_fts_broadcaster]: Node: spawner_joint_state_broadcaster, Namespace: /
1: [spawner-4] [WARN] [1721940021.067098941] [spawner_fts_broadcaster]: Node: spawner_fts_broadcaster, Namespace: /
```

The output of the node few times is like the following:
```
1: [spawner-5] [WARN] [1721939183.798181067] [spawner_forward_position_controller]: Node: spawner_forward_position_controller, Namespace: /
1: [spawner-5] [WARN] [1721939184.000523775] [spawner_forward_position_controller]: Node: spawner_forward_position_controller, Namespace: /
1: [spawner-5] [WARN] [1721939184.202215019] [spawner_forward_position_controller]: Node: spawner_forward_position_controller, Namespace: /
1: [spawner-5] [WARN] [1721939184.403937840] [spawner_forward_position_controller]: Node: spawner_forward_position_controller, Namespace: /
1: [spawner-5] [WARN] [1721939184.606189584] [spawner_forward_position_controller]: Node: spawner_forward_position_controller, Namespace: /
1: [spawner-5] [WARN] [1721939184.807338552] [spawner_forward_position_controller]: Node: spawner_forward_position_controller, Namespace: /
```

## Bug report

**Required Info:**

- Operating System:
  - 22.04 and 24.04
- Installation type:
  - Binaries
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Expected behavior

Always be able to give the information of all the nodes and their namespace

#### Actual behavior

Gives a duplicated info of it's own node name and namespace

fujitatomoya, 2024-07-26:
@saikishor thanks for posting issue.

could you provide us the minimal reproducible example for this?

saikishor, 2024-07-26:
> @saikishor thanks for posting issue.
> 
> could you provide us the minimal reproducible example for this?

Hello @fujitatomoya!

Thank you for responding. I don't have a minimal example as of now. I have a testcase in ros2_control framework that is reproducible. I think you want something simpler and minimal, if that is the case, I can try to prepare one in the following days and get back to you here.

Thank you.