ros2/rclpy #1352: `rcl_shutdown` is not thread-safe when used in both `signal_handler` and `Context.__exit__`.

YuanYuYuan, 2024-09-02:
## Bug report

We found this sporadic failure with rclpy (rolling) due to the race condition while calling `rcl_shutdown`.
In this [issue](https://github.com/ros2/rmw_zenoh/issues/275), the conflict happens if `rmw_shutdown` is slow so that `rcutils_atomic_store` on Thread 1 is set after the check `rcl_context_is_valid` on Thread 2.

## Analysis
- Thread 1
  - [rclpy::shutdown_contexts](https://github.com/ros2/rclpy/blob/8a2e2d3de5a96eaa50b9c9336d97396f8489bda4/rclpy/src/rclpy/signal_handler.cpp#L144)
  - [rcl_shutdown](https://github.com/ros2/rclpy/blob/8a2e2d3de5a96eaa50b9c9336d97396f8489bda4/rclpy/src/rclpy/context.cpp#L49)
  - [check rcl_context_is_valid](https://github.com/ros2/rcl/blob/522f3476c76d911d72922584b18df1fa1761f608/rcl/src/rcl/init.c#L289)
  - [rmw_shutdown](https://github.com/ros2/rcl/blob/522f3476c76d911d72922584b18df1fa1761f608/rcl/src/rcl/init.c#L294)  // Slow
  - [rcutils_atomic_store](https://github.com/ros2/rcl/blob/522f3476c76d911d72922584b18df1fa1761f608/rcl/src/rcl/init.c#L301)

- Thread 2
  - [InitContextManger.\_\_exit\_\_](https://github.com/ros2/rclpy/blob/8a2e2d3de5a96eaa50b9c9336d97396f8489bda4/rclpy/rclpy/__init__.py#L112)
  - [_try_shutdown](https://github.com/ros2/rclpy/blob/8a2e2d3de5a96eaa50b9c9336d97396f8489bda4/rclpy/rclpy/__init__.py#L200)
  - [g_default_context.try_shutdown()](https://github.com/ros2/rclpy/blob/8a2e2d3de5a96eaa50b9c9336d97396f8489bda4/rclpy/rclpy/utilities.py#L100)
  - [if self.__context.ok()](https://github.com/ros2/rclpy/blob/8a2e2d3de5a96eaa50b9c9336d97396f8489bda4/rclpy/rclpy/context.py#L172)
  - [Context::ok() -> rcl_context_is_valid](https://github.com/ros2/rclpy/blob/8a2e2d3de5a96eaa50b9c9336d97396f8489bda4/rclpy/src/rclpy/context.cpp#L147)  // If this passes
  - [Context::shutdown() -> rcl_shutdown](https://github.com/ros2/rclpy/blob/8a2e2d3de5a96eaa50b9c9336d97396f8489bda4/rclpy/src/rclpy/context.cpp#L151) // , then this will run

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- Installation type:
  - Docker
- Version or commit hash:
  - ros-rolling
- DDS implementation:
  - This bug is RMW independent. Any RMW implementation with a slow `rmw_shutdown` could lead to [this issue](https://github.com/ros2/rmw_zenoh/issues/275).
- Client library (if applicable):
  - Both rcl and rclpy are rolling. 
- Relevant issue: #1081

Barry-Xu-2018, 2024-09-03:
After checking code, there is a issue.

Thread 1  
Before calling rcl_shutdown(),   `g_contexts_mutex` is locked.

https://github.com/ros2/rclpy/blob/464a357a7d9e39a12aed966fa3ccb44d50bb6816/rclpy/src/rclpy/context.cpp#L42-L53

Thread 2

Before calling rcl_shutdown(), `g_contexts_mutex` isn't used.

https://github.com/ros2/rclpy/blob/464a357a7d9e39a12aed966fa3ccb44d50bb6816/rclpy/src/rclpy/context.cpp#L150-L165

If thread 2 executes rcl_shutdown first, thread 1 can still call rcl_shutdown again.  
So rcl_shutdown should be safeguarded by g_contexts_mutex in Context::shutdown().  

I create a fix https://github.com/ros2/rclpy/pull/1353. 
Could you test in your environment to see if the issue no longer occurs with this PR ?

YuanYuYuan, 2024-09-03:
Hi @Barry-Xu-2018! Thanks for your attempt. But it seemed not to work...

Barry-Xu-2018, 2024-09-03:
My fixing only prevented simultaneous calls to rcl_shutdown(). 
Another problem is that repeated calls to shutdown() cause an exception.  

I have updated fixing. Please try again.  

Currently, rclpy is designed to throw an exception if rcl_shutdown() is called multiple times on the same context (there's a specific test case for this). So, the error is expected behavior. However, in your situation, it shouldn't throw an error, but rather ignore the second shutdown call.   
I will discuss with other members on how to fix this problem.

YuanYuYuan, 2024-09-03:
Hi @Barry-Xu-2018, I have confirmed your latest fix resolves the issue (no more duplicated `rcl_shutdown`). Thanks!

closed 2024-11-08 by fujitatomoya