ros2/rclpy #1368: No way to set rosout QoS for a Node

rg-gravis, 2024-10-04:
## Feature request

#### Feature description
The `rclcpp` `NodeOptions` class has a `rosout_qos` field that allows you to set the QoS policy for when the node publishes to `/rosout` as part of ROS logging. The `rclpy` Node constructor has no such option, so python users are stuck with the default rosout_qos, which is unsuitable for some applications (we're experiencing dropped log messages that can be resolved for cpp nodes by adjusting the `rosout_qos`). 

#### Implementation considerations
An additional parameter in the `Node` constructor would work perfectly well, and would just need to get passed to `rcl` [here](https://github.com/ros2/rclpy/blob/1dd3b210dbe520f06146c47d74af63cafda5f9fa/rclpy/src/rclpy/node.cpp#L501)

cottsay, 2024-11-01:
Hi there, this seems like a reasonable feature to add. Please consider opening a pull request to implement it!

fujitatomoya, 2024-11-03:
@rg-gravis you can try https://github.com/ros2/rclpy/pull/1376.

closed 2025-01-30 by ahcorde