ros2/rclpy #1378: High CPU load when setting `use_sim_time` to `True` and subscribing to `/clock`

Tacha-S, 2024-11-08:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 24.04
- Installation type:
  -  binaries
- Version or commit hash:
  -   7.1.2
- DDS implementation:
  -  Fast-RTPS, Cyclone DDS
- Client library (if applicable):
  -  rclpy

#### Steps to reproduce issue

example subscriber
```python
import rclpy
from rclpy.node import Node
from std_msgs.msg import String

class TestSubscriber(Node):

    def __init__(self) -> None:
        super().__init__('test_subscriber')

        self.create_subscription(String, 'topic', self.callback, 1)

    def callback(self, msg: String) -> None:
        self.get_logger().info(f'I heard: {msg.data}')

def main() -> None:
    rclpy.init()
    node = TestSubscriber()
    rclpy.spin(node)

if __name__ == '__main__':
    main()
```

terminal 1
```bash
ros2 run test_package test_sub --ros-args -p use_sim_time:=true
```

terminal 2
Assuming gz sim is running with `empty.sdf` and publishing the clock.
```bash
ros2 topic pub /clock rosgraph_msgs/msg/Clock -r 1000
```

#### Expected behavior

Ideally, the load should remain below 1% and only reach a few percent at most unless `/topic` is being published.

#### Actual behavior

Depending on the CPU and DDS, the CPU load reaches around 20% to 35%.

#### Additional information

----
## Feature request

Is it possible to reduce CPU load without changing the clock frequency?

#### Feature description

#### Implementation considerations

fujitatomoya, 2024-11-08:
> Is it possible to reduce CPU load without changing the clock frequency?

I believe that it is really hard to achieve the requirement... 😓
when the simulation clock is enabled, internally the node subscribes on `/clock` topic to update the simulation clock, which is 1kHz in your example.

Tacha-S, 2024-11-08:
Thank you for the response.

It seems the only options are either to allow an increase in CPU load,
to use `rclpy` only for nodes where the simulation clock doesn't need to be considered and replace nodes that need consideration with `rclcpp`, 
or to reduce the update rate on the simulator side.

I thought I might be handling this incorrectly, but it appears to be working as expected, so I'll close this.

closed 2024-11-08 by Tacha-S