ros2/rclpy #1388: Why are tf updates slower with ReentrantCallbackGroup in rclpy compared to the default callback group?

Tacha-S, 2024-12-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu24.04
- Installation type:
  - binaries
- Version or commit hash:
  - 7.1.2
- DDS implementation:
  - Cyclone dds
- Client library (if applicable):
  - rclpy

I am experimenting with parallelizing callbacks in **rclpy** using the **MultiThreadedExecutor** and **ReentrantCallbackGroup** based on the official documentation. 

For example, I want to parallelize TF updates and always refer to the latest TF. To test this, I created a large number of dummy callbacks (where `dummy_topic` publishes at 100Hz) to intentionally clog the callback queue and expect the benefits of parallelization. 

test.py
```python
import time

import rclpy
from rclpy.callback_groups import ReentrantCallbackGroup
from rclpy.executors import MultiThreadedExecutor
from rclpy.node import Node
from rclpy.time import Time
from std_msgs.msg import Empty
import tf2_ros

class TestCallbackGroup(Node):

    def __init__(self, node_name):
        super().__init__(node_name)
        # node = Node('tf_listener_01')
        self.tf_buffer = tf2_ros.Buffer()
        self.tf_listener = tf2_ros.TransformListener(self.tf_buffer, self)

        for _ in range(100):
            self.create_subscription(Empty,
                                     'dummy_topic',
                                     self.sub_callback,
                                     10,
                                     callback_group=ReentrantCallbackGroup())

        self.timer = self.create_timer(0.03, self.timer_callback, callback_group=ReentrantCallbackGroup())

    def sub_callback(self, msg):
        time.sleep(0.01)

    def timer_callback(self):
        try:
            transform = self.tf_buffer.lookup_transform('base_link', 'map', Time())
            self.get_logger().info(
                f'Latest transform stamp: {transform.header.stamp.sec + transform.header.stamp.nanosec * 1e-9}')
        except Exception as e:
            self.get_logger().error(str(e))

if __name__ == '__main__':
    rclpy.init()
    node = TestCallbackGroup('test_callback_group')
    executor = MultiThreadedExecutor()
    executor.add_node(node)
    executor.spin()
    rclpy.shutdown()
```

However, in practice, the TF updates are slow. Moreover, when I use the **default callback group** (instead of explicitly specifying a **ReentrantCallbackGroup**), the TF updates appear to be faster. This behavior persists even when I set the **spin_thread** parameter of `TransformListener` to `True` or move the `TransformListener` to a separate node.

Why does specifying a **ReentrantCallbackGroup**, which allows parallelization, result in seemingly less efficient TF updates compared to the default callback group?

#### Steps to reproduce issue

Publish transform from map to base_link.

run
```bash
python3 test.py
```

run
```bash
ros2 topic pub /dummy_topic std_msgs/msg/Empty -r 100
```

#### Expected behavior

#### Actual behavior

#### Additional information

Tacha-S, 2024-12-17:
It was my misunderstanding.
I resolved it myself, so I will close this.

closed 2024-12-17 by Tacha-S

apockill, 2024-12-17:
Could you tell us what you found? I'd be curious as to the root cause.

Tacha-S, 2024-12-18:
I’m not entirely sure if this is correct,
but according to the documentation and code, it seems that timers are processed with the highest priority in the callback execution order.
https://docs.ros.org/en/foxy/Concepts/About-Executors.html#scheduling-semantics
https://github.com/ros2/rclpy/blob/8f1f16f16090184062f1993728d063ff2680f958/rclpy/rclpy/executors.py#L746-L783

Therefore, when using a `MutuallyExclusiveCallbackGroup`,
the TF updates appear to proceed without being overwhelmed by the callbacks from the dummy topic.

However, I don’t know why the timer isn’t prioritized in the case of a `ReentrantCallbackGroup`.
This is just speculation, but I think `MutuallyExclusiveCallbackGroup` determines the next callback to execute each time something becomes executable based on priority, whereas `ReentrantCallbackGroup` might decide all callbacks to be executed at a certain point in time, including those from the dummy topic that are already in the queue.