ros2/rclpy #1390: Canceling an action goal returns "Exception while canceling goal: None"

sea-bass, 2024-12-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- Installation type:
  - Binaries
- Version or commit hash:
  - 7.7.0
- DDS implementation:
  - FastDDS or CycloneDDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
I'm not entirely sure this is a bug, but it was pointed out to me in an issue on one of my own repos. I could then reproduce it with the simple `rclpy` examples:

If I start an example action server and then cancel it:

```
ros2 run examples_rclpy_minimal_action_server server_single_goal 

ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci "{order: 100}"

Ctrl + C
```

The output is the following

```
Waiting for an action server to become available...
Sending goal:
     order: 100

Goal accepted with ID: 8deec6ba2ecb430e81c476963f65b6dc

Canceling goal...
Exception while canceling goal: None
```

#### Expected behavior

It should say something like

```
Canceling goal...
Goal canceled
```

#### Actual behavior

```
Canceling goal...
Exception while canceling goal: None
```

fujitatomoya, 2024-12-20:
This can be reproduced with rolling source code build.

fujitatomoya, 2024-12-20:
I thin the root cause of this is SIGINT handler in `rclpy`,

https://github.com/ros2/rclpy/blob/ca59a7f05b6712ba535a78ec617627d401668268/rclpy/src/rclpy/signal_handler.cpp#L144

`rclpy::shutdown_context` will be called in another deferred thread once the signal comes in. precisely this thread is waiting on the semaphore from the signal handler.

and in the `final` statement, it tries to send the cancel request. i confirmed that this request is delivered to the action server and action server sends the response to the client. but client never gets this response message, because at this time context is already shut down by deferred thread. that makes cancel result is None, and generate the `RuntimeError` exception by ros2action.

i think we need to have the signal handler, and expects that is the cancel request from the terminal.

fujitatomoya, 2024-12-20:
@sea-bass thanks for posting issue.

can you try https://github.com/ros2/ros2cli/pull/956 see if that fixes your problem too? that would be really appreciated.

sea-bass, 2024-12-21:
> ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci "{order: 100}"

Just tested and can confirm this fixes the issue -- thanks @fujitatomoya!

fujitatomoya, 2024-12-23:
@sea-bass thanks for testing.

closed 2025-01-16 by fujitatomoya