ros2/rclpy #1393: Race condition in `MultiThreadedExecutor`

felixdivo, 2025-01-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04.1
- Installation type:
  - Binaries
- Version or commit hash:
  - `rolling` via `ros-tooling/action-ros-ci@v0.3` GitHub action
- DDS implementation:
  - unknown
- Client library (if applicable):
  - rclpy

#### Steps to reproduce the issue

Documented here: https://github.com/felixdivo/ros2-easy-test/issues/45. It is not 100% reproducible, yet it occurs frequently in my repo.

#### Expected behavior

No exception is raised.

#### Actual behavior

```
self = <rclpy.executors.MultiThreadedExecutor object at 0x7fb80c2f32f0>
timeout_sec = <rclpy.executors.TimeoutObject object at 0x7fb80c41cb90>
wait_condition = <bound method Future.done of <rclpy.task.Task object at 0x7fb80c2f30b0>>

    def _spin_once_impl(
        self,
        timeout_sec: Optional[Union[float, TimeoutObject]] = None,
        wait_condition: Callable[[], bool] = lambda: False
    ) -> None:
        try:
            handler, entity, node = self.wait_for_ready_callbacks(
                timeout_sec, None, wait_condition)
        except ExternalShutdownException:
            pass
        except ShutdownException:
            pass
        except TimeoutException:
            pass
        except ConditionReachedException:
            pass
        else:
            self._executor.submit(handler)
            self._futures.append(handler)
            # make a copy of the list that we iterate over while modifying it
            # (https://stackoverflow.com/q/1207406/3753684)
            for future in self._futures[:]:
                if future.done():
>                   self._futures.remove(future)
E                   ValueError: list.remove(x): x not in list

/opt/ros/rolling/lib/python3.12/site-packages/rclpy/executors.py:895: ValueError
```

#### Additional information

Previously, I had similar issues caused by race conditions reported in https://github.com/ros2/rclpy/pull/1129. This again appears to be caused by multiple threads simultaneously acting on the executor. Is there some unintended usage by me?

#### Possible solution

If not, I suggest going with the [it is "Easier to ask for forgiveness than permission"](https://docs.python.org/3/glossary.html#term-EAFP) and just wrap the `self._futures.remove(future)` in a try-except block catching and `ValueError`s.

Barry-Xu-2018, 2025-01-06:
> This again appears to be caused by multiple threads simultaneously acting on the executor. Is there some unintended usage by me?

In your test, is it possible to call  _spin_once_impl() simultaneously from different threads for the same executor ?

felixdivo, 2025-01-06:
I thought not; that's why I was skeptical. But upon further inspection, there is a case!

1. On place is here, where the executor spins until the test case is complete [here](https://github.com/felixdivo/ros2-easy-test/blob/37ed7d1acf78ca06350d2441bec2294b2157654b/ros2_easy_test/ros2_easy_test/decorators.py#L128-L131).
2. The other one is in the test case, [where waiting for the action to conclude](https://github.com/felixdivo/ros2-easy-test/blob/37ed7d1acf78ca06350d2441bec2294b2157654b/ros2_easy_test/tests/test_interactions.py#L115-L117) eventually [calls `spin_until_future_complete()` too](https://github.com/felixdivo/ros2-easy-test/blob/37ed7d1acf78ca06350d2441bec2294b2157654b/ros2_easy_test/ros2_easy_test/env.py#L321).

They necessarily run in parallel. Is that intended usage of a `rclpy.executors.MultiThreadedExecutor`? The documentation around its thread safety is a bit ... [thin](https://github.com/ros2/rclpy/blob/4e8b071127228d5dace5aebf61d02260ecb91253/rclpy/rclpy/executors.py#L878-L887).

Barry-Xu-2018, 2025-01-14:
> They necessarily run in parallel. Is that intended usage of a rclpy.executors.MultiThreadedExecutor?

For the same executor, spin_until_future_complete() should not be run in parallel. So I think this isn't the correct way to use it.

felixdivo, 2025-01-14:
Hmm, is there some documentation on what is thread-safe and what isn't? I must've missed it if it exists.

I'm honestly a bit lost on properly doing this kind of thing if not that way. Is there some other way in which one can wait for a `Future` to be available that is not busy-waiting?

felixdivo, 2025-01-27:
Hey @sloretz, thank you for your message! The second option works like a charm. I implemented it in https://github.com/felixdivo/ros2-easy-test/pull/52 and it solved the issue wonderfully. Thank you so much!

PR #1477 (2025-10-02): Fix issue with MultiThreadedExecutor by adding lock to protect _futures
## Description

The MultiThreadedExecutor contains a possible race condition. If _spin_once_impl is called more than once, it is possible that multiple threads may end up passing the condition on line 1008 and try to remove the same future on line 1009.

This solution proposes locking self._futures. This also has the benefit of removing the need to make a shallow copy of self._futures (#1129). 

Replaces #1129
Fixes #1393

### Is this user-facing behavior change?

No

### Did you use Generative AI?

No

### Additional Info [... cut]
closed 2025-01-27 by felixdivo